gnubee-git/GnuBee_Docs #2: Missing Documentation

ulaker, 2017-09-09:
Board received with zero documentation and no hint where it might be found. Please provide info where documentation is located for booting and attaching the gnubee to a network.

pinney, 2017-09-09:
Non sequitur

ulaker, 2017-09-09:
So can you help or provide alink where help is available to start this nice toy?

Reagards,
ulaker
From: L. D. Pinney 
Sent: Saturday, September 9, 2017 7:55 PM
To: gnubee-git/GnuBee_Docs 
Cc: ulaker ; Author 
Subject: Re: [gnubee-git/GnuBee_Docs] Missing Documentation (#2)

Non sequitur

pinney, 2017-09-09:
Sure ... I'll admit it's early in the game.
But all one needs to do is follow the Assembly
https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/Assembly
And connect to the device with the USB to UART cable ... or to your router.
You can and should update the firmware.
It's been accepted to both "mainline" lede-project and libreCMC.
The GnuBee version is here 
https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware

If you are using the GnuBee firmware you can connect the device to your router with the Blue RJ-45

ulaker, 2017-09-09:
Ok, thanks for the fast reply despite inappropriate time to ask these questions. I just received the board and am quite keen to see it doing something. I did follow the assembly instructions but that only addresses the four screws/brackets. The power up etc. pp. I did not manage to locate yet. The connection to the router shows up as libreCMC with it accepting a dhcp allocated ip 4 an 6 identity.
On port 80 no response and on port 22 rejected request.
 
I can understand if you terminate this session and we continue tomorrow. I have prepared microsd card with mips debian but have no clue on how to cranc up the set up or installation of firmware or operating system.

pinney, 2017-09-09:
No I understand you want your new toy up and running
been there done that... lets keep going

ulaker, 2017-09-09:
Ok, thanks 1e6
how do I get to see any interaction on the LAN or WAN port? tried both now and no response on port 80, reject on ssh, so what can I do to get a login or some sort of interaction?

pinney, 2017-09-09:
Please do this the new firmware is better... and will improve :)
download and place the new firmware on a USB stick (msdos format)
you can choose either LEDE or libreCMC
https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware
RENAME the firmware gnubee.bin

then reboot the device ....It's like magic ...the boot-loader reads the file updates the firmware

ulaker, 2017-09-09:
done as instructed with libreCMC. the green led flashed fast than slow for a while now steady green. Do I now power down remove usb stick and reboot?

pinney, 2017-09-09:
Yes ... great
Remove the USB stick and reboot.

pinney, 2017-09-09:
Please connect your device to your router via the blue RJ-45

ulaker, 2017-09-09:
done, getting ip 4/6 dhcp positive identification libreCMC and mac.

pinney, 2017-09-09:
Rock on .... have a nice day :)

ulaker, 2017-09-09:
Thanks so far, I am at quare one. neither port 80 nor 22 let me interact. So please one more hint than I let you get to have a good nights sleep. pretty please:)

pinney, 2017-09-09:
If you are using libreCMC :

use 192.168.10.1

also ... do "whatever" it takes to login to a so called unsecure site .... it's hardly unsecure ...it's YOUR device :)

pinney, 2017-09-09:
I'm ...still here ...let's get you into the device

ulaker, 2017-09-09:
Is this libreCMC 192.168.10.1 hardcoded?
 my router  assigned ip to the gnubee is 192.168.178.35,

pinney, 2017-09-09:
This could be correct .... it's set for DHCP
Try to login @ 192.168.178.35

ulaker, 2017-09-09:
tried ssh to 192.168.178.35 but get reject. Maybe need to somehow start a sshd on the gnubee?

pinney, 2017-09-09:
192.168.10.1 is the default IP

ulaker, 2017-09-09:
So I would need a machine with a ethernet port here in the 192.168.10.xxx network for the initial setup?

pinney, 2017-09-09:
I should ask... whta OS do you use?

pinney, 2017-09-09:
what...damn GFW lag

ulaker, 2017-09-09:
windows10 and ubuntu both over wifi to router. the router is wired to gnubee directly.

ulaker, 2017-09-09:
Maybe I can wire the gnubee to the notebook ethernet?

pinney, 2017-09-09:
You should be able to connect your PC to the BLACK RJ-45 and login to the webinterface

pinney, 2017-09-09:
We are getting close ....
Please try the simple way
make an ethernet from your PC to the device (DHCP)...your PC to the black port on the PC1

ulaker, 2017-09-09:
BINGO, the cable from the black ethernet port to the notebook provided a lua login and root/root got me to the promised land. God Bless thanks and have a good sleep.
Best wishes for all of you for the weekend,

pinney, 2017-09-09:
Happy Dayz :)

pinney, 2017-09-09:
I'm kinda sorry ... because what I do everyday seems SO SIMPLE to me.
But of course ... I know it really not simple :)

Checkout this PHat one 
https://github.com/lede-project/source/pull/1316/commits/6a93c5288c0394dda7cb3eb400f6603d7a5cea29

pinney, 2017-09-09:
Thanks ulaker ....closing

closed 2017-09-09 by pinney

ulaker, 2017-09-11:
Dis do some exercises with the GnuBee and tried both LEDE and libre CMC. Both seem to behave like identical twins with regard to functionality or rather the lack of it.

Both show no inclination to allow network shares and nfs does not install due to dependancy problems.

Installing nfs-kernel-server (2.1.1-1) to root...
Downloading https://librecmc.org/librecmc/downloads/snapshots/v1.4/packages/mipsel_24kc/packages/nfs-kernel-server_2.1.1-1_mipsel_24kc.ipk

Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for nfs-kernel-server:
 * 	kmod-fs-nfsd * 	kmod-fs-nfs * 
 * opkg_install_cmd: Cannot install package nfs-kernel-server.

 Any thoughts on how to persue this  a NAS to behave like one?