gnubee-git/GnuBee_Docs #4: GnuBee NAS functionality?

ulaker, 2017-09-11:
Did do some exercises with the GnuBee and tried both LEDE and libreCMC. Both seem to behave like identical twins with regard to functionality or rather the lack of it.
Mounted both SATA as well as USB drives and SDcard ok
Both show no inclination to allow network shares and nfs does not install due to dependancy problems.
I inderstand that the USB port near the SD card will let me flash firmware. It worked with LEDE and libre CMC. How do I get Debian or openmediavault to install?

Here I failed to achieve what I thought should maybe give access to drives over the network:
Installing nfs-kernel-server (2.1.1-1) to root...
Downloading https://librecmc.org/librecmc/downloads/snapshots/v1.4/packages/mipsel_24kc/packages/nfs-kernel-server_2.1.1-1_mipsel_24kc.ipk

Collected errors:

    satisfy_dependencies_for: Cannot satisfy the following dependencies for nfs-kernel-server:
    kmod-fs-nfsd * kmod-fs-nfs *
    opkg_install_cmd: Cannot install package nfs-kernel-server.

Any thoughts on how to persue this NAS to behave like one?

pinney, 2017-09-11:
Looks like a kernel dependency problem.
I'm building new firmware with these packages (and many more) built-in.

pinney, 2017-09-11:
To install Debian and then optionally OMV please see:
https://github.com/gnubee-git/gnubee-git.github.io

If you like I can send you new libreCMC/LEDE to try...just need an email

ulaker, 2017-09-11:
Would love to test the bewborn brainchildren for the GnuBee, send them and I will try to comply.

Thanks in advance,
ulaker

From: L. D. Pinney 
Sent: Monday, September 11, 2017 3:20 PM
To: gnubee-git/GnuBee_Docs 
Cc: ulaker ; Author 
Subject: Re: [gnubee-git/GnuBee_Docs] GnuBee NAS functionality? (#4)

To install Debian and then optionally OMV please see:
https://github.com/gnubee-git/gnubee-git.github.io

If you like I can send you new libreCMC/LEDE to try...just need an email

ulaker, 2017-09-11:
Thank you, I will try to follow these instructions, I am just embarrassed that I did not find them myself whilst strolling over the github site.

ulaker, 2017-09-11:
ulaker58@gmail.com if you can not see it on the reply I done from gmail.

ulaker, 2017-09-11:
installation script for debian failed:
Choose partition to install (will be formatted !!!): /dev/sda1

/dev/sda1: LABEL="Seagate Expansion Drive" UUID="0EB0EBABB0EB980B" TYPE="ntfs" PARTUUID="22be084d-01"
Partition will be formatted, please type CONFIRM if you agree: CONFIRM
Formatting /dev/sda1, please wait...
mke2fs 1.43.3 (04-Sep-2016)
Creating filesystem with 488377856 4k blocks and 122101760 inodes
Filesystem UUID: f5746acf-5709-482a-af76-9066d4923075
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
	4096000, 7962624, 11239424, 20480000, 23887872, 71663616, 78675968, 
	102400000, 214990848

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (262144 blocks): done
Writing superblocks and filesystem accounting information: done       

./debian-jessie-install: line 80: debootstrap: not found
Error: Command debootstrap failed.
root@libreCMC:~#

sluzynsk, 2017-09-11:
opkg install debootstrap

That will get it one step further. However:
I: Extracting util-linux...
I: Extracting zlib1g...
W: Failure trying to run: chroot /tmp/newroot dpkg-deb -f /var/cache/apt/archives/dpkg_1.17.27_mipsel.deb Version
W: See /tmp/newroot/debootstrap/debootstrap.log for details
W: Failure trying to run: chroot /tmp/newroot mount -t proc proc /proc
W: See /tmp/newroot/debootstrap/debootstrap.log for details
Error: Command debootstrap failed.
root@libreCMC:~#

The log file mentioned contains two lines that state "Illegal Instruction". I get that same error if I try to execute either of the failed commands manually.

sluzynsk, 2017-09-11:
Breaking form by replying to myself, but... more info:

root@libreCMC:~# file /bin/busybox
/bin/busybox: ELF 32-bit LSB executable, MIPS, MIPS32 rel2 version 1, dynamically linked, interpreter /lib/ld-musl-mipsel-sf.so.1, corrupted section header size

root@libreCMC:~# file /tmp/run/mountd/mmcblk0p1/usr/bin/passwd
/tmp/run/mountd/mmcblk0p1/usr/bin/passwd: setuid ELF 32-bit LSB executable, MIPS, MIPS-II version 1 (SYSV), dynamically linked, interpreter /lib/ld.so.1, for GNU/Linux 2.6.32, BuildID[sha1]=fd2483ffcbbf12e8852ddf1f27299e0d1c2236c0, stripped

So the LEDE/libreCMC images are using the MIPS32R2 ISA; Debian Jessie is MIPSII. The installed image can't run the tools inside Jessie since they are incompatible.

Presumably this could be fixed by installing a MIPSII version of LEDE or whatever first?

pinney, 2017-09-11:
Hi Steve 

To install Debian / OMV please use the firmware / info from 
https://github.com/gnubee-git/gnubee-git.github.io

sluzynsk, 2017-09-12:
That got me a LOT closer. The installation script runs and all looks well. It reboots, it finds the partition, and tries to boot it. Then this happens:

[   10.670000] Waiting disk spinup and searching for partition GNUBEE-ROOT...done.
[   17.540000] EXT4-fs (mmcblk0p1): couldn't mount as ext3 due to feature incompatibilities
[   17.560000] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
[   17.570000] Partition GNUBEE-ROOT found. Starting...
[   21.680000] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00000004
[   21.680000] 
[   21.690000] Rebooting in 1 seconds..

I don't know how to troubleshoot it from there.

pinney, 2017-09-12:
Hi Steve :
Please try using a USB drive or SATA drive for troubleshooting.
Thanks!

sluzynsk, 2017-09-12:
using a SATA drive worked - apparently it does not work using the SD card.

Last problem now - what in the heck is the root password set in the Debian environment by your script?

sluzynsk, 2017-09-12:
It worked installing on a SATA disk rather than the SD card. I also managed to guess that pre-set root password. :)

pinney, 2017-09-12:
Hi Steve 
I've added the user/pass to the debian readme
Great ... it's working.
Let's close this issue ... 
Please open a new one if needed.

Thanks!

closed 2017-09-12 by pinney