gnubee-git/GnuBee_Docs #7: instruction for invoking failsafe mode?

FHulot, 2017-09-13:
So I stupidly didn't think to look at closed issues here and have been banging my head against the shipped firmware. Made some changes in the settings (did not upgrade firmware) and eventually the GnuBee is not on speaking terms with me. 

I am trying the failsafe mode according to https://gogs.librecmc.org/libreCMC/libreCMC/wiki/Router+Reset+Instructions but I don't seem to be having any luck. The lights don't respond as described to my pressing the reset button  (the little black one, right?) during the initial slow blink. No luck with ssh or ping. 

Could you let me know the proper way, if it's different of invoking  failsafe mode is different on GnuBee? 

TIA!

PS The last thing I did was to follow this advice:
https://lede-project.org/docs/user-guide/lede_as_clientdevice

pinney, 2017-09-13:
Hello FHulot : 
I think the OpenWrt page is still a good reference for failsafe mode
https://wiki.openwrt.org/doc/howto/generic.failsafe

However it may be easier just to use the bootloader to update the firmware.
One can easily update the firmware using the bootloader :
Place the new firmware on a USB stick (msdos format)
RENAME the firmware gnubee.bin
Then reboot the device ... the boot-loader reads the file then updates the firmware.
While loading the new firmware the LEDs will give an indication or reading/writing.
Lower LED fast blink = read, slow blink = write, solid = done.
When the upper LED changes from off to on solid it's time to remove the USB Stick and reboot into the new firmware.

Updated firmware is here :
https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware
and here :
https://github.com/gnubee-git/gnubee-git.github.io

Thanks!
Larry

ngiger, 2017-09-14:
I have a question

Do I have to download all the files or eg just one of the following

* https://github.com/gnubee-git/GnuBee_Docs/blob/master/GB-PC1/firmware/LEDE/GnuBee/GnuBee-lede-17.01-full-0911.bin
* https://github.com/gnubee-git/GnuBee_Docs/blob/master/GB-PC1/firmware/uBoot/115200/GB_PCx_uboot_115200-900MHz-1066RAM.bin

FHulot, 2017-09-14:
Thanks, Larry. That worked!

ngiger, I was able to flash https://github.com/gnubee-git/GnuBee_Docs/blob/master/GB-PC1/firmware/libreCMC/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade.bin

pinney, 2017-09-14:
@FHulot You are welcome :)

@ngiger Only the firmware should be needed ... 
Please don't change u-boot...unless you know what you are doing.

closed 2017-09-14 by pinney

twotwenty, 2018-04-02:
I cannot find firmware anymore... many dead links.

jpgey, 2018-08-20:
@ghost, try this:
https://github.com/gnubee-git/GnuBee_Docs/blob/master/GB-PCx/firmware/GnuBee-LEDE-17.01/lede-ramips-mt7621-gb-pc1-squashfs-sysupgrade.bin

or this full