gnubee-git/GnuBee_Docs #8: Proposal for a GnuBee home page

ngiger, 2017-09-14:
As I would like to have a similar informative page for the GnuBee PC as Martin Michlmayers page on installling Debian on a QNAP device (see http://www.cyrius.com/debian/kirkwood/qnap/ts-119/), I added a gh-pages branch in my repository https://github.com/ngiger/GnuBee_Docs/tree/gh-pages. The result is shown under

https://github.com/gnubee-git/GnuBee_Docs

Feell free to submit suggestion for imrovements/corrections. If you want to add it to your repository the following steps should work
  
    git remote add ngiger https://github.com/ngiger/GnuBee_Docs.git
    git fetch ngiger
    git checkout gh-pages

https://help.github.com/articles/using-a-custom-domain-with-github-pages/ documents how it would be easy to use the web page under another domain name (eg. gnubee.org).

There are some word on how to setup a webserver to test changes in the readme.txt.

The whole pages is inspired by a project a work with (http://elexis.info), where I copied a lot of ideas from https://buildr.apache.org/

pinney, 2017-09-14:
@ngiger  When it comes to web pages ... I get lost easily :)
I would say go right ahead ... I'm happy to have help.

FHulot, 2017-09-14:
This is great, ngiger. Thank you.

Honos2014, 2017-09-15:
It would be great, indeed.

ngiger, 2017-09-21:
http://gnubee.org/ is up and running and documents firmware and debian installation.

closed 2017-09-21 by ngiger