gnubee-git/GnuBee_Docs #10: Failed installing new firmware. Error is "Linux image size 32845 too small"

ngiger, 2017-09-15:
I get the following from u-boot when trying to download the  firmware (both LEDE and libreCMC)

    U-Boot 1.1.3 (Jul  9 2017 - 14:14:05)
    Board: MediaTek APSoC DRAM: 512 MB

    Config XHCI 40M PLL 
    ******************************
    Software System Reset Occurred
    ******************************
    MediaTek SPI flash driver, SPI clock: 32MHz
    spi device id: ef 40 19 0
    find flash: W25Q256FV
    ============================================ 
    MediaTek U-Boot Version: 5.0.1.0-6
    -------------------------------------------- 
    ASIC MT7621A DualCore (MAC to MT7530 Mode)
    DRAM_CONF_FROM: Auto-Detection 
    DRAM_TYPE: DDR3 
    DRAM bus: 16 bit
    Xtal Mode=3 OCP Ratio=1/3
    Flash component: SPI Flash
    Date:Jul  9 2017  Time:14:14:05
    ============================================ 
    icache: sets:256, ways:4, linesz:32, total:32768
    dcache: sets:256, ways:4, linesz:32, total:32768 

    #### The CPU freq = 900 MHZ #### 
    estimate memory size = 512 Mbytes

    Reset MT7530
    set LAN/WAN WLLLL
    (Re)start USB...
    USB0:   mtk-xhci: init hccr be1c0000 and hcor be1c0020 hc_length 32
    Register 300010f NbrPorts 3
    Starting the controller
    USB XHCI 0.96
    scanning bus 0 for devices... 3 USB Device(s) found
          scanning bus for storage devices... 1 Storage Device(s) found
    reading gnubee.bin

    32845 bytes read
    Linux image size 32845 too small!

under libreCMC I got the following md5sum

    root@libreCMC:/# md5sum /mnt/sdc1/gnubee.bin 
    32a5687d9a6db9b36fc4661a84589cb0  /mnt/sdc1/gnubee.bin

Which correspond nicely to my downlaoded images

    32a5687d9a6db9b36fc4661a84589cb0 GnuBee-lede-17.01-sysupgrade-030917.bin
    92089de2069b9df1c017ec1576abf9ae librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade.bin

Any ideas about what is going wrong? BTW, libreCMC version is r7+10-2e31b23 Version: v1.4-core-RC1

pinney, 2017-09-15:
@ngiger 

That seems strange ... is there a device in one of the USB ports?
with a file named gnubee.bin ?

If one boots with a msdos formatted device in any one of the USB ports
and there is a file named gnubee.bin (case insensitive) the PC1 will write (or try) to write this to the flash memory.
 
Today I updated : 
https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/LEDE/GnuBee
Please try one of those :)

ngiger, 2017-09-15:
It was a stupid errror of my part.

If I download using the correct link https://github.com/gnubee-git/GnuBee_Docs/raw/master/GB-PC1/firmware/LEDE/GnuBee/GnuBee-lede-17.01-lite-0915.bin I get a 15 MB big file.
I had however downloaded https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/LEDE/LEDE-17.01 which contains only a HTML page with the link to to above file.

Discovered this because I found out for the first time howto use the embedded WebServer of ibreCMC, which correctly sayed `The uploaded image file does not contain a supported format. Make sure that you choose the generic image format for your platform. ` when I used the "Backup / Flash Firmware" button under "System".

Once this was done and GnuBee rebooted, I could continue with the following commands on the console
opkg update
opkg install debootstrap
opkg install fish bash
wget https://raw.githubusercontent.com/gnubee-git/GnuBee_Docs/master/GB-PC1/scripts/jessie_3.10.14/debian-jessie-install
bash debian-jessie-install

This command is still runing.

Will document this setup tomorrow. Thanks for your help.

pinney, 2017-09-15:
@ngiger 
If you are trying to setup Debian 
Please use the firmware found on
https://github.com/gnubee-git/gnubee-git.github.io

closed 2017-09-16 by ngiger