gnubee-git/GnuBee_Docs #9: wget in LibreCMC lacks SSL ?

FHulot, 2017-09-15:
Sorry to open an "issue" about gnubee when most likely the issue is with the user.

Using libreCMC Frivolous Fred v1.4 r31-2ee88be / LuCI Master (git-17.231.19234-2ee88be)

I can't get updating lists does not work. From the commandline, it responds
`Downloading https://librecmc.org/librecmc/downloads/snapshots/v1.4/packages/mipsel_24kc/packages/Packages.gz
wget: not an http or ftp url: https://librecmc.org/librecmc/downloads/snapshots/v1.4/packages/mipsel_24kc/packages/Packages.gz
*** Failed to download the package list from https://librecmc.org/librecmc/downloads/snapshots/v1.4/packages/mipsel_24kc/packages/Packages.gz
`

Doing it on luci gives some additional info:
`wget: SSL support not available, please install one of the libustream-ssl-* libraries as well as the ca-bundle and ca-certificates packages.
`

ca-certificates are already installed.
ca-bundle is not listed in the available packages nor is  anything starting with libustream-ssl

Help?

Thanks!

pinney, 2017-09-15:
@FHulot : 
It seems to missing the libustream-openssl package.
If you like I can email you the package to install ... However
I'm building a new firmware now and I'll uploaded when it's done.

This will help others ... thanks for opening an issue!

FHulot, 2017-09-15:
Hi Larry

Thank you very much. I can definitely wait for the new build. Probably good for me to work on other things for a while!

FHulot

pinney, 2017-09-15:
@FHulot :
Turns out it's not so simple ... ssl / https isn't cooperating.
So I recommend that you try one of the GnuBee-lede-17.01-*-0915.bin firmwares in :
https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/LEDE/GnuBee

lede-project doesn't use ssl / https ... but it also doesn't have a libre gnu kernel.

I'll keep working on/testing libreCMC and uploading when it's ready.

FHulot, 2017-09-15:
@pinney Thanks very much for this.

So the L in LEDE is Linux but it has nonfree parts so not Gnu?  If I go on and install Debian from LEDE will the kernel be replaced and the GB will be Absolutely Free?

Not critical to answer that as I will probably go back later and start clean with the working LibreCMC but I'd be happy to know if you know. 

FH

pinney, 2017-09-15:
If you are trying to setup Debian
Please use the firmware found on
https://github.com/gnubee-git/gnubee-git.github.io

FHulot, 2017-09-18:
OK thanks! That got me a lot further!

debian and debian modules installed fine.

I went on to run omv-jessie-install and it bailed out with

`Setting up libgtk-3-0:mipsel (3.14.5-1+deb8u1) ...`
`Bus error`
`dpkg: unrecoverable fatal error, aborting:`
` unable to flush updated status of ` `libgtk-3-0:mipsel': Read-only file system`
`./omv-jessie-install: line 12:   861 Bus error               apt-get install openmediavault -y --force-yes`
`./omv-jessie-install: error reading input file: Input/output error`
`root@gnubee-pc1:~# `

I was still able to type commands but almost none--including `ls`--would work. So I rebooted and things seemed more normal, but when I hopefully tried

`root@gnubee-pc1:~# omv-initsystem`
`/usr/sbin/omv-initsystem: 24: .: Can't open /etc/default/openmediavault`
`root@gnubee-pc1:~# `

Just for fun, I pointed a browser at the GnuBee and got:

`Welcome to nginx on Debian!
If you see this page, the nginx web server is successfully installed and working on Debian. Further configuration is required. 
etc`

I'm installing to a 16GB USB drive, which is plugged into the single port at the end with the red button.

Thanks!

pinney, 2017-09-18:
@FHulot 
It looks like a problem reading/writing to the drive.
Can you try a SATA drive? or maybe another USB drive?

FHulot, 2017-09-18:
Okay. Some progress! Used a SATA drive in slot 1. Gets all the way through omv-jessie-install, but then freezes. I reinstalled and before running omv-jessie-install commented out omv-initsystem  just to make sure that's where it freezes. And yes it is.  All previous finished normally, then when issuing omv-initsystem from the prompt, sure enough it froze again.  I don't have a serial cable so maybe stuff is happening???  I've been doing this over ssh and I understand omv by default turns off ssh. 

nmap reports from the WAN side:
`Starting Nmap 6.47 ( http://nmap.org ) at 2017-09-18 19:08 EDT`
`Nmap scan report for 192.168.10.4`
`Host is up (0.031s latency).
`All 1000 scanned ports on 192.168.10.4 are closed`

and finds nothing other than my laptop on the LAN side. 

I've searched the installation outputs for signs of error and have found only one. This was during omv-jessie-install:
`Setting up monit (1:5.9-1+deb8u1) ...`
`newaliases: fatal: open /etc/postfix/main.cf: No such file or directory`

Is that the problem?

Just ordered the serial cable.

FH

FHulot, 2017-09-19:
Hold the phone.  I had chosen "no configuration"  in the omv email options. Will try again and accept default.

pinney, 2017-09-19:
@FHulot 
One can use any of the popular USB to UART or USB to TTL devices available.
I'm still not sure what the problem is ... but if possible please use the defaults.

FHulot, 2017-09-19:
Okay yes of course. Why am I fooling around? I accepted all defaults.  I think it works. No errors in the install. Just executes omv-initsystem and clams up.  nmap reports all ports closed,  but I think rather than a problem that's a feature.  OMV probably locks down all the ports initially and expects you to configure on the webgui via the NAS itself. OMV forums and guides seem to assume that you have a keyboard and monitor hooked directly.  I've been expecting to be able to get to it using a browser on a computer on the LAN or WAN.

If this is true, will I be able to configure using the UART/USB cable? Using a text-based browser or something?

pinney, 2017-09-20:
@FHulot 
Yes one can configure OMV during the install using the serial console.
I think it's best just to accept the defaults ... 
Once OMV is running then use the web interface.

optlink, 2017-09-25:
@pinney How is the SSL fix for LibreCMC coming along? I just got my GnuBee today.

pinney, 2017-09-25:
@optlink 

Please upgrade your firmware the latest libreCMC can be found here :
https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/libreCMC

You will need to set the Distribution feeds to :

src/gz frivolous_fred_core https://librecmc.org/librecmc/downloads/snapshots/v1.4-core/ramips/mt7621/packages

Just add -core to v1.4 as shown above.

Thanks!

optlink, 2017-09-25:
@pinney Awesome, thank you. Turns out I had to make sure that "Keep my settings" was unchecked when flashing to solve the SSL problems.

FHulot, 2017-10-04:
Closing this thread. The SSL issue seems to be solved. The OMV issue is discussed [here](https://github.com/gnubee-git/GnuBee_Docs/issues/18).

closed 2017-10-04 by FHulot