gnubee-git/GnuBee_Docs #11: Please enable the Wiki

ngiger, 2017-09-16:
I think it might be a good idea to enable persons to participate via the Wiki in this github repository (which you can enable easily via the project settings).
It would easen to the task to build up a documentation, because I imagine that the following workflow could be very efficient.

1. User opens a issue (and get  hopefully always a response as quickly as in the past)
1. Once the issue is resolved, a first condensed answer to the question is posted on the wiki (this takes a minute or two)
1. Important wiki entries will eventually (but may be a few days/weeks later) added into the documentation 

I volunteer to make task 2 and 3 happen for the next few week/months.

pinney, 2017-09-16:
@ngiger I think it's enabled now please try it.
Thanks!

ngiger, 2017-09-16:
No. It is not yet enabled (at least not for non collaborators).

In the  https://github.com/gnubee-git/GnuBee_Docs/settings dialog the
**Features...Wiki** scheckbot must be **enabled**
The next checkbox **Restrict editing to collaborators only** must  be**DISABLED**.

You can test it by logging out of github and going to https://github.com/gnubee-git/GnuBee_Docs/wiki to check whether you can add a wiki page or not.

pinney, 2017-09-16:
Seems sort of backwards ... try it now

ngiger, 2017-09-16:
Now it works.
I had be logged in github to add a .
Added https://github.com/gnubee-git/GnuBee_Docs/wiki
Now you can see (but not edit) the wiki also when your are not logged in.

Thanks!

closed 2017-09-16 by ngiger