gnubee-git/GnuBee_Docs #12: Illegal instruction while runnint debootstrap (from LEDE)

ngiger, 2017-09-16:
I am still stuggling to install Debian

As from a previous failed debootstrap I already had a

    mkdir: can't create directory 'rom': Read-only file system
    pivot_root: No such file or directory

Therefore I would recommend to label the installed partition as GNUBEE-ROOT once the debootstrap completed successuflly. Pull request will follow soon.

I tried installing LEDE(GnuBee-lede-17.01-full-0915.bin) and could start debbootstrap again using

    opkg update
    opkg install deboostrap curl
    curl  --insecure -O https://raw.githubusercontent.com/gnubee-git/GnuBee_Docs/master/GB-PC1/scripts/jessie_3.10.14/debian-jessie-install
    sh debian-jessie-install
    <..>
    W: Failure trying to run: chroot /tmp/newroot dpkg-deb -f /var/cache/apt/archives/dpkg_1.17.27_mipsel.deb Version
    W: See /tmp/newroot/debootstrap/debootstrap.log for details
    W: Failure trying to run: chroot /tmp/newroot mount -t proc proc /proc
    W: See /tmp/newroot/debootstrap/debootstrap.log for details
    cat /tmp/newroot/debootstrap/debootstrap.log 
    illegal instruction

Any hints?

pinney, 2017-09-16:
@ngiger 
Please use the firmware found at
https://github.com/gnubee-git/gnubee-git.github.io

This firmware serves as the kernel/recovery system for Debian and OMV

ngiger, 2017-09-17:
I used librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-07-22.bin.
Then I get (because it looks that I still have GNUBEE-ROOT partition laying around the following error

    [   10.070000] md: Waiting for all devices to be available before autodetect
    [   10.070000] md: If you don't use raid, use raid=noautodetect
    [   10.080000] md: Autodetecting RAID arrays.
    [   10.080000] md: Scanned 0 and added 0 devices.
    [   10.080000] md: autorun ...
    [   10.090000] md: ... autorun DONE.
    [   10.090000] VFS: Mounted root (squashfs filesystem) readonly on device 31:5.
    [   10.100000] devtmpfs: mounted
    [   10.110000] Freeing unused kernel memory: 232K (806d6000 - 80710000)
    [   10.140000] random: nonblocking pool is initialized
    [   10.540000] Waiting disk spinup and searching for partition GNUBEE-ROOT...
    [   11.040000] mtk_soc_eth 1e100000.ethernet eth0: port 4 link up
    [   13.550000] done.
    [   16.980000] EXT4-fs (sdb1): couldn't mount as ext3 due to feature incompatibilities
    [   17.100000] EXT4-fs (sdb1): mounted filesystem with ordered data mode. Opts: (null)
    [   17.110000] Partition GNUBEE-ROOT found. Starting...
    mkdir: can't create directory 'rom': Read-only file system
    pivot_root: No such file or directory
    chroot: failed to run command ‘/sbin/init’: No such file or directory
    [   21.330000] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00
    [   21.330000] 
    [   21.330000] Rebooting in 1 seconds..

I think the initrd should handle this situation, too. Will try againg after deleting the partition table on the hard disk

pinney, 2017-09-17:
Are you trying to install Debian to a Hard Drive?
Or do you have the SDCard with Debian installed?

It seems easier to me to only install the target drive (and no others)
then it is quite easy to choose the partition where you want to install Debian.

dgazineu, 2017-09-20:
Please try to run the debian-jessie-install from stratch after switching to the recommended firmwares:

- https://github.com/gnubee-git/gnubee-git.github.io/blob/master/debian/openwrt-ramips-mt7621-gb-n1-squashfs-sysupgrade.bin (older "official" kernel from SoC vendor, very stable and have support for keeping the Debian root on the mmc slot)
or
- https://github.com/gnubee-git/gnubee-git.github.io/blob/master/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-07-22.bin (newer kernel, with some support and improvements being worked out actively by us and the LibreCMC community)

ngiger, 2017-09-21:
I have documented the installation of Debian under http://gnubee.org/install_debian.html.

The full log of my installation can be found under http://gnubee.org/logs/install_debian.log

And now Debian works fine for me.

closed 2017-09-21 by ngiger

smurfix, 2018-02-05:
@dgazineu if I use the "official" GnuBee libreCMC repository to rebuild the firmware as per instructions, I get "Illegal instruction" whenever I try to run a Debian command (via chroot, obviously).

Thus, please tell us what's special in that repository. After all, GnuBee is supposed to use only free software (as defined by the FSF), and that includes being able to rebuild firmware. Thank you.

leocrawford, 2018-02-21:
Agreed. I have rebuilt the libreCMC kernel and get illegal instruction. I must admit to adding in some kernel modules to the build so it isn't entirely clean, but I find it hard to imagine they're making the difference. I'll try again with a completely clean repo but it does seem hard to make a repeatable build from source.

pinney, 2018-02-21:
> After all, GnuBee is supposed to use only free software (as defined by the FSF), and that includes being able to rebuild firmware.

Note : Please rmember that the GnuBee devices shipped with uBoot and libreCMC on the Flash Chip.
Thus only those sources 'must' be made available. (as defined by the FSF)