gnubee-git/GnuBee_Docs #15: SATA Slot 5 not active

azwirko, 2017-09-19:
Good Day,
  I've got an issue with one of the slots in the GnuBee PC1.  Slot 5 cannot be accessed.  I'm running 6 identical Samsung 850 EVO SSD's.  All work and format fine under Windows.  I've tried swapping drives within the GnuBee slots and I still cannot get Slot 5 to recognize the drive.  I've loaded and tried both the latest LEDE and libreCMC bin and the problem still persists.  

  Attached is a copy of the System Log after booting, where I see reference to "ata5: SATA link down" and some error codes.  Please assist !

Thanks
andyz

[GnuBee-syslog.txt](https://github.com/gnubee-git/GnuBee_Docs/files/1315071/GnuBee-syslog.txt)

pinney, 2017-09-19:
Hi AndyZ

Based on your log ... This does look like a hardware problem.

Can you try booting up with only one drive (in slot5 of course) installed ?

azwirko, 2017-09-21:
Pinney, 

Hi and thanks for the quick response.  I performed two tests one with SATA5 slot with HD installed and second with NO HD's installed.  Attached are the two system logs, one for each case.  Please review and advise.  Thanks.

[SATA5-installed.txt](https://github.com/gnubee-git/GnuBee_Docs/files/1319617/SATA5-installed.txt)
[NO-HD-installed.txt](https://github.com/gnubee-git/GnuBee_Docs/files/1319618/NO-HD-installed.txt)

pinney, 2017-09-21:
Hi AndyZ : 
I'm sorry say ...  I think it's a hardware problem.
Can you send a photo of the suspect SATA connector?

azwirko, 2017-09-21:
Pinney,
  I think u hit the problem right on the head! I took out my magnifying glass to look at the data connector portion of SATA5 and I can swear it looks like some of the pins were never or barely soldered to the board.  Here's a pix from my cell phone with zoom x4, best I could get.  SATA5 is on the left and SATA6 is on the right.  SATA6 pins look nice and flat on the board whereas SATA5 data pins appear bent upwards slightly and the very back pin 7 may not even be connected!  Please review and advise  Thanks.

https://photos.app.goo.gl/TOZCOa6ihJoYFIyf2 

andyz

pinney, 2017-09-21:
Hi AndyZ : 

I'm in China and behind the firewall...so I can't get the picture.
Please contact Crowd Supply for a return.

You can email me directly ldpinney@yahoo.com

azwirko, 2017-09-21:
Email sent with pix attached

azwirko, 2017-09-22:
Larry,
  Being an electrical engineer and having worked on SMT/SMD previously, I pulled out the hot air rework tool and reflowed the data pins on SATA5 connector, pushing them all flat.  I fired up the GnuBee with only one drive in that slot and syslog no longer indicates any errors.  I ran an "fdisk /dev/sda" and the SSD shows up as expected.  Looks like all is working fine now. Thanks for the pointers.

  I guess you'll have to pay a bit more attention to QA/QC on future boards from your assembly vendor.  Also I might note my board came wrapped in standard bubble wrap surrounded in foam peanuts.  I would have expected a ~$200 board to come in an anti-static bag.  I've gotten cheaper boards out of China wrapped and delivered with better protection than the GnuBee.  I'm just glad this failure wasn't related to losing a active component on the PCB due to some static discharge during transit!

Regards

andyz

Hi Andy :

Doesn't look good ... but it does look like an easy fix for someone with soldering skills such as a mobile phone repairman.

I don't know much about the return policy @ Crowd Supply ... yet.

Sorry about that.

Larry

pinney, 2017-09-23:
Hi AndyZ : 

It's very nice you could fix the problem...I'm sure most people couldn't do that.
It's our first project...we've learned many new things.
We shall improve on many things including QA/QC and packing.

Thanks!

closed 2017-09-23 by pinney