gnubee-git/GnuBee_Docs #16: Weird network settings

neheb, 2017-09-19:
No idea where to put this.

I noticed that by default, a bridge is set up that bridges the ethernet interface and...nothing else. Eliminating the bridge gives a small performance boost (~20mbps in iperf). Normally the bridge is used to bridge two wireless(2.4ghz and 5ghz) and the ethernet together but since there is no wireless this is just pointless overhead.

Speaking of overhead, the switch driver in use(mt7530) has the capability to do switching in hardware as opposed to software. This basically exposes the two ethernet ports as two interfaces which can then be bridged or whatever. The difference is in speed (needed here). DSA is faster than swconfig. I have not been able to set it up properly (only for a Netgear R7800) so I can't give hard numbers.

Speaking of numbers, I recently posted about a huge performance problem with the ethernet driver used in the GnuBee that reduces iperf speeds by overr 100mbps. Running "ethtool -K eth0 rx off" fixes it.  No idea what impact this has on anything else.

pinney, 2017-09-20:
@neheb 
There is certainly room for improvement :)
If you are using LEDE please push your code improvements :)
It seems there have been some recent commits related to DSA.
But I really haven't had time to look into that.

nariox, 2017-10-25:
@neheb would you mind explaining how to remove the bridge?

neheb, 2017-10-25:
Go to interfaces and edit the LAN to be just eth0.1. I tried removing it by
default but it doesn't seem to work.

neilbrown, 2018-05-13:
Is this still an issue?  If it is and you can tell me what firmware you are using, I can have a look and see if I can help.

neheb, 2018-05-13:
This is OpenWrt specific.

Mainline uses DSA instead of swconfig. Not sure if the staging drivers have been ported. Would be nice if John shares his code (he has DSA working in one of his trees).

The weird turning rx off thing is probably some CPU scheduling bug with the OpenWrt driver. iperf results are very inconsistent.

neheb, 2020-04-25:
OpenWrt master now uses DSA. This is irrelevant.

closed 2020-04-25 by neheb