gnubee-git/GnuBee_Docs #17: Turn off status leds

sorvis, 2017-09-20:
Is it possible to turn off the status leds on the front? I tried playing with the led section of the preinstalled firmware, but that did not seem to impact them any.

pinney, 2017-09-20:
@sorvis 
Which LEDs the two by the power switch or the LEDs on the Ethernet Jacks?

sorvis, 2017-09-20:
Thank you for getting back to me so fast. The two by the power switch. That's the configuration screen I tried playing with. Should leaving the default state checkbox unselected do the trick?

pinney, 2017-09-20:
I think option 'Trigger' 'none' should make those remain off.

sorvis, 2017-09-21:
Hmm, I think this should be it, but the LEDs still seem to be on after saving and even rebooting. Am I missing something?

[image]

pinney, 2017-09-21:
@sorvis 
What firmware is running on the device?
If you haven't updated the firmware please use one from here :
https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/LEDE/GnuBee
GnuBee-lede-17.01-full-0915.bin <--- a 'full' build with a large number of packages
GnuBee-lede-17.01-lite-0915.bin <--- a smaller build to allow one to install their own package set.

sorvis, 2017-09-21:
@pinney that did the trick to update the firmware. It now shows all four devices like your screen shot did and the light turned right off. Thank you

closed 2017-09-21 by sorvis