gnubee-git/GnuBee_Docs #18: OMV install

menumsg, 2017-09-21:
I'm trying to install the omv-jessie-install.  Debian Jessie and Debian modules installed fine.  I run omv-jessie-install and it gets to a point after generating locales and then the kernel panics.  I've attached a copy of the output.  The GnuBee will not boot after this and I have to start with installing Debian again.  I've tried three times with the same results.  Any idea what is happening?
[openmediavault.txt](https://github.com/gnubee-git/GnuBee_Docs/files/1322491/openmediavault.txt)

dgazineu, 2017-09-21:
Which of the available firmware/kernel versions are you with Debian using
before the OMV install?

menumsg, 2017-09-21:
I used the librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-07-22.bin ingnubee-git.github.io/debian/.

dgazineu, 2017-09-21:
Ok, as the kernel panic is not giving us much details, I will repeat an
installation from scratch with the same firmware to see if anything in
Debian repositories/packages have changed that could lead to this situation.

FHulot, 2017-09-23:
The webgui is actually working after installation but before omvinitsystem. Maybe you can see something there?

Honos2014, 2017-09-25:
I've got the same problem after installing OMV on Debian using those instructions:
https://github.com/gnubee-git/gnubee-git.github.io/blob/master/omv/README.DEBIAN.txt

ohrensessel, 2017-09-27:
Same with me. I used the omv apt-repository from https://github.com/gnubee-git/gnubee-git.github.io and after installing:

```
root@gnubee-n1:/etc/apt# omv-initsystem

Message from syslogd@gnubee-n1 at Sep 27 07:48:28 ...
 kernel:[ 1857.790000] Fatal exception: panic in 5 seconds

Message from syslogd@gnubee-n1 at Sep 27 07:48:28 ...
 kernel:[ 1857.980000] Fatal exception: panic in 5 seconds
```

Now, GnuBee is not reachable any more. As I'm remotely connected I now have to wait till the evening when I can power cycle my GnuBee again.

dgazineu, 2017-09-27:
I have been able to reproduce the failure, and obtained the stack failure, on the I/O scheduler of kernel 4.4.52-gnu.

The alternative to run OMV now is to use the firmware with "official" kernel from here : [http://gnubee-git.github.io/debian/openwrt-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-03-01.bin](http://gnubee-git.github.io/debian/openwrt-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-03-01.bin)

After flashing the firmware above, be sure to install modules with the following two commands and restart your GB-PC1, and OMV will work right away with your already done installation (if nothing got corrupted on previous hangs):
```
  wget https://raw.githubusercontent.com/gnubee-git/gnubee-git.github.io/master/debian/debian-modules-install
  sh debian-modules-install
```

ohrensessel, 2017-09-29:
thanks, that helped. had to flash the firmware via fail safe (usb).
will there be an updated firmware with 4.4.52 fixing the bug at some time?

dgazineu, 2017-09-29:
Glad it helped.

I think so. As both kernel and OMV have updates since this version, I will
work on a newer firmware to fix this bug with OMV.

I will keep this issue open to inform about progress.

menumsg, 2017-10-04:
I tried this method.  When I run omv-initsystem I still get a kernel panic and it won't boot anymore.  I tried not running omv-initsystem and it worked until I reboot the system.  During the boot, it had a kernel panic and wouldn't boot.  I tried running it on a USB drive, SD card, and a hard drive and all had the same problem.

dgazineu, 2017-10-04:
@menumsg Are you running firmware version "openwrt-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-03-01.bin" this time?

menumsg, 2017-10-04:
I thought I flashed that version but it didn't work so I've been using the one that doesn't work still.  I saw that it flashed this time so I will try it again.

dgazineu, 2017-10-04:
@menumsg Ok, please remember to redo the module installation for the kernel version on this firmware, after flashing and booting Debian:
```
wget https://raw.githubusercontent.com/gnubee-git/gnubee-git.github.io/master/debian/debian-modules-install
sh debian-modules-install
```

daniellandau, 2017-11-16:
The kernel recommended here doesn't have the tun module (for openvpn) in the provided modules pack. Any progress on fixing the newer kernels, or of getting more modules in the older kernel?

dgazineu, 2017-11-16:
I am testing two different versions this week.
If all goes well, next week should be updated firmwares with all reported
needs (as this one)  inside.

dgazineu, 2017-11-20:
New firmware available @
https://github.com/gnubee-git/gnubee-git.github.io/raw/master/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-18.bin,
with Debian/OMV support.
    Install instructions are the same as before, updated on
http://gnubee.org/install_debian.html as well.

    The updated sources are @
https://gogs.librecmc.org/dgazineu/GnuBee-libreCMC

Please give a try and report back.

daniellandau, 2017-11-21:
Works for me, thanks!

dgazineu, 2017-11-29:
@menumsg Have you been able to install OMV, or are you still experiencing issues?

menumsg, 2017-11-29:
I was able to install it today and it worked.

dgazineu, 2017-11-30:
Glad to hear.

closed 2017-11-30 by dgazineu