gnubee-git/GnuBee_Docs #19: Howto configure GnuBee-lede?

ngiger, 2017-09-21:
As a newbie to LEDE, I clone git@github.com:gnubee-git/GnuBee-lede.git, installed the needed packages (on Debian) and starting working using 

    sudo apt-get install gcc binutils bzip2 flex python perl make findutils grep diffutils unzip gawk util-linux subversion libz-dev
    ./scripts/feeds update -a
    make menuconfig

Now my question is, how can I import/use default configuration used to build the LEDE firmware?

Thanks in advance

dgazineu, 2017-09-21:
The default ".config" for our LEDE build is saved as "GB-PC1_defconfig" in
the build root, after you had cloned the repository.
You can copy GB-PC1_defconfig to .config after cloning it.
When you do enter "make menuconfig", the configuration will be the same we
have used to make the firmware.
Upon compiling, further changes may be needed to it, due to upstream LEDE
evolution (packages defaults may override tue default chosen by us), and
need some adjustments.

ngiger, 2017-09-21:
Thanks for your quick response. After search around I found out that the GB-PC1_defconfig only exists in the origin/lede-17.01 branch.

Checking out this branch and make menuconfig worked with your tipp. Now waiting for make to finish, which will probably take some time.

ngiger, 2017-09-22:
Documented building LEDE under https://github.com/gnubee-git/GnuBee_Docs/wiki/build-LEDE

closed 2017-09-22 by ngiger