gnubee-git/GnuBee_Docs #20: UART voltage

BzukTuk, 2017-09-23:
Hi GnuBee team,
maybe a silly question, but I did not find any information regarding J1 UART voltage. 
(did look on schematics, and MT7621 datasheets, but with no definitive answer - I'm not HW/embeded guy and everywhere I searched there is either 1.8V, 3.3V and 5V logic mentioned, so I'm not sure, and don't want brick my device first day :-) ) 
I did not buy your USB->UART adapter, because it would be above our customs price threshold, and I'd pay much more fee/tax (sorry). 

I used Raspberry Pi for UART console before, and it uses 3.3V. Is it safe to use RPI for GnuBee`s UART console ?

Thank you very much

(consider including this information into USB_to_UART\README.md)

pinney, 2017-09-23:
There is no voltage pin @ the J1 header.
Only TX/RX and ground... DO NOT connect to any voltage.
I've tested the USB to UART adapters using the PL2303,CP210x, and HL-340.
They all work OK.
Please see : 
https://github.com/gnubee-git/GnuBee_Docs/tree/master/USB_to_UART

BzukTuk, 2017-09-23:
Hi pinney,
thank you for quick answer. By voltage I meant 0 and 3.3 V logic levels for TX/RX. I'll look what logic levels is used in PL2303, CP210x, and HL-340 chips.

Thanks

pinney, 2017-09-23:
OK just checked those too...
Without being connected.
The CP2102 and PL2303 are ~ 3v with HL-340 is ~ 5v
When plugged in the first two come up a bit...and the last one comes down a bit.

closed 2017-09-23 by pinney

BzukTuk, 2017-09-23:
Ok, I was not able to connect to serial console using RPI2. Will buy CP2102 next week... RPIs UART is still working, so everything should be fine... I hope. (maybe it was just issue with RPI unable to handle baudrate correctly)
Thanks

BzukTuk, 2017-09-25:
USB->UART CP2102 does not work also :( It behaves exactly same way as raspberry pi's serial console.

When serial console is attached before I power gnubee on, it does not start. Two green leds are very dim. (very little green light)

When I attach serial console cable on already running gnubee, green leds are full green for like 3 seconds, and then leds dim and system is unresponsive until I unplug serial console and do PowerOff/PowerOn.

Maybe something is shorted (with capacitor in play, so green lights stay full green for few seconds on running system?).

What can I try? Everything else is working on my gnubee. All SATA slots, all USBs, uSD card...

Thanks

pinney, 2017-09-25:
@BzukTuk 
Can you send some photos of your setup?
Such as the USB to UART adapter and your wire connections to the device.

BzukTuk, 2017-09-26:
@pinney 
Sure, [here it is](https://imgur.com/a/pIFSz). If you can`t access imgur, pls post me a image hosting service you can access.

Strange - when I was trying to capture picture of under-powered LED, something happened and I can connect UART without ssh session freeze and LED is not dimming any more... However, "screen /dev/ttyUSB0 57600" still shows nothing during reboot. Maybe something finally broke?

Between GND and RX there is 3.36V, and between GND and TX 3.07V.

UART USB dongle and wires I`m using are still working with Raspberry Pi.

Thank you

pinney, 2017-09-26:
@BzukTuk 
The wiring in the picture looks correct...
I duplicated your setup here and it works...
My CP2102 USB to UART looks a bit different than yours.
However they should work the same :) 
Have you tried swapping the RX/TX wires?
Also please have a close look at R1781 and R1782 between the mount hole and the 3.5mm jack.

BzukTuk, 2017-09-26:
@pinney 

I grabbed my CP2102 from local arduino hobby shop for like 4 USD. Probably some local production or different vendor/assembler. There is CP2102 on the main chip and it works :-)

No, I did not switch RX/TX. (with RPI, I always triple-check wiring before powering both sides of UART. I read somwhere that Rpi is pretty sensitive to this). Do you think it is safe to try connecting TX to TX, and RX to RX? I will try tomorrow (maybe only 3.5mm jack is wired according schema, and RX TX are switched on GnuBee pins)

Both resistors look fine, I`ll try to make a macro picture tomorrow, but don`t think there will be anything to see. How many Ohm are they? ( I can see just 0R ±1％ 1/16W SMD0402  in *BOM.ods)

I removed aluminium plates, and I get LED dimed again. I'm starting to think that this has something to do with screw-hole for L shaped part of "case".

Thank you, I`ll let you know how it goes tomorow

BzukTuk, 2017-09-26:
@pinney 
Hmmm... could not sleep... Let's try something risky... like switching RX to RX and TX to TX...

IT WORKS! :-) I'm in U-boot cmdline now... hoooray

So, the label on GnuBee PCB has logic: ''put RX into this pin, TX into this pin, and GND into this pin''

Thank you for encouraging me into trying this :-) Have a nice day, I'm going to sleep

BzukTuk, 2017-09-26:
I assembled gnubee again, and again, it started to dim the leds after UART connection. Two problems hidden as one. Removed Aluminium plate on UART pin side, and it is working again.

I will put some insulation between PCB and screws... Tomorrow...

pinney, 2017-09-26:
@BzukTuk 

USB to UART ... 
I think that needs to documented ... 'somewhere'

LEDs dim on UART connection seems strange ... 
The mount holes are not connected to anything. (such as ground)
Please post your findings

Thanks!

BzukTuk, 2017-09-30:
@pinney 
I found out how to connect to serial console, without LED dimm... I need to plug UART pins into already running GnuBee. When serial console is working, and I power off and power on gnubee, it will not boot again, leds will stay dimmed and device is without any activity. (just once I managed to press power button fast enough, to actually restart device clean with uart connected).

Strange is that for device to properly boot up, unplugging the USB-UART adapter from my computer is enough.

When restarting the device via default firmware 'reboot' command, I can see booting from the start, enter U-Boots command line etc...

I also found out that this issue is not related to aluminium plates and/or screws. It behaves same even without its 'case'. With raspberry pi, I always had serial console cable connected before powering on both devices (and RX TX switched all the time, because I followed label on PCB :-) ).

I'm open to suggestions or experiments (non destructive :-) ), if any, but I'm pretty satisfied with GnuBee's functionality.

Thanks