gnubee-git/GnuBee_Docs #21: "invalid ELF header" when trying to start Debian Stretch

ngiger, 2017-09-24:
I installed Debian stretch by replacing jessie->stretch in debian-jessie-install

I completed successfully, but booting failed with

    [ 16.940000] Partition GNUBEE-ROOT found. Starting...
    chroot: error while loading shared libraries: /lib/mipsel-linux-gnu/libc.so.6: invalid ELF header
    [ 21.080000] Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00
    [ 21.080000]
    [ 21.080000] Rebooting in 1 seconds..

Any ideas whether this fails? Is there a way to upgrade to stretch?

dgazineu, 2017-09-24:
Hi, I haven't worked on/tried this upgrade yet.
Which firmware version are you using?

ngiger, 2017-09-24:
**librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-07-22.bin** 
with sha25sum cac672da5dd3c1f055721f08909af5139416f387d4e9603021dc34de10ede5a8
reportsLinux version 4.4.52-gnu (dgazineu@olimpo) (gcc version 5.4.0 (LEDE GCC 5.4.0 r20-646beea) ) #0 SMP Wed Jul 5 16:50:03 2017

This image boots the jessie without any problem.

Changing to **openwrt-ramips-mt7621-gb-n1-squashfs-sysupgrade.bin** with sha256sum 877a830ed829d83b012d0d688968e50040e0da277353cd44e5f2aefc9e7f3fb3 
which reports
[    0.000000] Linux version 3.10.14 (dgazineu@olimpo.gazineu) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 unknown) ) #2 SMP Fri Jun 24 19:32:11 BRT 2016

Fixes the problem. I should have thought at this possibilty, too.

Adirelle, 2017-10-09:
I installed jessie then dist-upgraded to stretch. It boots and a lot of (but not all) software seems to work.

dgazineu, 2017-11-29:
Direct Debian stretch install is not supported (as LibreCMC/Openwrt itself still show problems), hence only Debian Jessie is provided (at this time)

closed 2017-11-29 by dgazineu