gnubee-git/GnuBee_Docs #22: LUKS encrypted drives with cryptsetup

adborden, 2017-09-24:
With the factory LibreCMC firmware, I'm getting an error trying to setup LUKS with cryptsetup.
```
device-mapper: table: 254:0: crypt: Error allocating crypto tfm
```

Which is usually caused by missing crypto modules in the kernel. By default, LUKS uses aes-xts-plain64, and I can confirm `cat /proc/crypto | grep xts` yields no results. Looking at the available modules in `opkg`, I'm not sure if there is one I can install that would provide this. I'm also wondering if there is a cipher spec that you would recommend for the GnuBee instead?

adborden, 2017-09-24:
I upgraded the firmware and now there's an xts package, but I'm having trouble installing it for some reason:
```
root@libreCMC:~# opkg install kmod-crypto-xts
Installing kmod-crypto-xts (4.4.52-1) to root...
Downloading https://librecmc.org/librecmc/downloads/snapshots/v1.4-core/ramips/mt7621/packages/kmod-crypto-xts_4.4.52-1_mipsel_24kc.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-crypto-xts:
 *      kernel (= 4.4.52-1-5628dfe5e3bdec27b47408e47a6b79b0) *  kernel (= 4.4.52-1-5628dfe5e3bdec27b47408e47a6b79b0) *
 * opkg_install_cmd: Cannot install package kmod-crypto-xts.
```

Those kernel versions look like they match O.o

adborden, 2017-09-25:
ah, I guess the installed kernel is not that version:
```
root@libreCMC:~# opkg list-installed | grep kernel
kernel - 4.4.52-1-a1d847d3569ef5d2e2051376d1a04b42
```

pinney, 2017-09-25:
Try opkg --force-depends install kmod-crypto-xts

neilbrown, 2018-05-13:
I think this has been answered, so I'll close the issue.  If you still have problems, please open another issue.

closed 2018-05-13 by neilbrown