gnubee-git/GnuBee_Docs #23: Cannot create RAID with LVM2 on factory LibreCMC

adborden, 2017-09-24:
Trying to create a raid1 in the factory LibreCMC but running into this error.
```
root@libreCMC:~# lvconvert --type raid1 -m1 data0/data
  raid1: Required device-mapper target(s) not detected in your kernel.
```

Not sure if there is a kernel module I can install that would provide this. I'll try using mdadm for now.

pinney, 2017-09-24:
@adborden 
Please upgrade your firmware the latest libreCMC can be found here : 
https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/libreCMC

You may want to use the "lite" version should have enough space to tailor your own package set.

With the current libreCMC firmware you will need to set the Distribution feeds : 

src/gz frivolous_fred_core https://librecmc.org/librecmc/downloads/snapshots/v1.4-core/ramips/mt7621/packages

Just add -core to v1.4 as shown above.

Thanks!

adborden, 2017-09-24:
Thanks! I upgraded the firmware to the lite version, `opkg update` works, and it looks like the system is up to date. I'm still getting an error when creating an LVM2 RAID though.

```
root@libreCMC:~# lvcreate --type raid1 -m1 -n data data0
  raid1: Required device-mapper target(s) not detected in your kernel.
  Run `lvcreate --help' for more information.
```

I've got the device-mapper package installed (`kmod-dm`), but I think dm raid targets are not included. I don't see any other device-mapper packages.

neheb, 2017-11-18:
@adbor try kmod-md-raid1 and kmod-md-raid0

dgazineu, 2017-11-29:
@adborden Have you been able to solve the problem?

adbor, 2017-11-29:
@adbor - is my nickname, but I have nothing common with this project. Sorry.

adborden, 2017-12-31:
I tried updating to the latest firmware to try this out, but opkg cant pull the latest packages. I get the same error described in https://github.com/gnubee-git/GnuBee_Docs/issues/9

neilbrown, 2018-05-13:
Where did you get with this?  Did you particularly need librecmc firmware, or will any kernel that supports Debian do?  If the later, please see https://groups.google.com/forum/#!topic/gnubee/VRNhAw1csnw

adborden, 2018-05-28:
I'm not attached to librecmc, I just want to use something "supported" by the gnubee team. I tried your prebuilt binary, but the vlan interfaces aren't configured correctly and I can't install anything.