gnubee-git/GnuBee_Docs #24: Jack connector pin out ?

Adirelle, 2017-09-28:
I could not found the serial jack connector pin out on the schematics of the GB1 nor the USB-to-UART. I am wondering if it is compatible with the FTDI TTL-232R-3V3-AJ, see the pinout from the datasheet here : http://i.imgur.com/4FY9xHq.png

Could you confirm it uses the same pin out ? or, else, which pin out is it ?

pinney, 2017-09-29:
That looks OK...
Please see https://github.com/gnubee-git/GnuBee_Docs/issues/20
for more info...

Adirelle, 2017-09-29:
Thanks for the quick reply. I have finally found a USB->UART CP2102 bundle (USB connector with the CP2102 on PCB and connection wires). I will keep you informed.

However, this audio pinout should me documented somewhere (maybe on the wiki ?).

closed 2017-09-29 by Adirelle