gnubee-git/GnuBee_Docs #25: GPIO usage with Linux

Adirelle, 2017-09-28:
I may have overlooked something. But are the GPIO on the J4 header already used for something ?

The same question goes for GPIO18 (push button): is it already used somehow (using the linux device tree maybe) ?

pinney, 2017-09-29:
GPIO @ J4 
These GPIOs are available for the user.
GPIO 26, 28, 29 and 30
Shown on page 13 
https://github.com/gnubee-git/GnuBee_Docs/blob/master/GB-PC1/Documents/GB-PC1_V1.0_Schematic.pdf

Reset Key from GB-PC1.dts : 
	gpio-keys-polled {
		compatible = "gpio-keys-polled";
		#address-cells = <1>;
		#size-cells = <0>;
		poll-interval = <20>;

		reset {
			label = "reset";
			gpios = <&gpio0 18 GPIO_ACTIVE_HIGH>;
			linux,code = <KEY_RESTART>;
		};
	};

Adirelle, 2017-09-29:
Thanks for the quick reply.

closed 2017-09-29 by Adirelle