gnubee-git/GnuBee_Docs #27: GNUBee-PC1No network connection after installing Debian

indiebrain, 2017-09-29:
Ohai y'all. I ran into a networking snag while trying to get Debian and Open-Media-Vault setup on my GNUBee-PC1. I'm all out of ideas on what to look for and was hoping someone might have ideas of things to try.

My endgame is to have Debian running open-media-vault on this GNUBee-PC1. I've been following the instructions at http://gnubee.org/install_debian.html.

At this point I'm connecting to the GNUBee via USB-to-UART with [this cable](https://www.amazon.com/gp/product/B00XSPECIA/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1) as described [here](http://gnubee.org/USB_to_UART/README.html).
> Larry uses GNU Screen AND tftpd-hpa and starts it calling
> screen /dev/ttyUSB0 57600

So far I've:

1. Flashed the bios with [librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-07-22.bin](https://github.com/gnubee-git/gnubee-git.github.io/blob/a1dfd44/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-07-22.bin)
2. Partitioned the drive `/dev/sda` I installed into slot 1 as  the [article](http://gnubee.org/install_debian.html) suggest:
> /dev/sda1 1 GB ext4 named boot
> /dev/sda2 10 GB ext4 named data
> /dev/sda3 rest ext4 named data
3. Installed Debain as suggested in the README.md of [https://github.com/gnubee-git/gnubee-git.github.io/debian/](https://github.com/gnubee-git/gnubee-git.github.io/blob/938e493/debian/README.DEBIAN.txt)
```
wget --no-check-certificate gnubee-git.github.io/debian/debian-jessie-install
chmod +x  debian-jessie-install
./debian-jessie-install
```
4. Connected the GNUBee to my router on the Blue RJ-45 ethernet port
5. Reboot

After the reboot I tried to install the modules It appears that the GNUBee leased an IP address via DHCP from the router, but the GNUBee isn't able to get to the internet to download the modules script

```
root@gnubee-pc1:~# ifconfig eth0.2
eth0.2    Link encap:Ethernet  HWaddr 78:a3:51:ab:b7:d0  
          inet addr:192.168.1.198  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::7aa3:51ff:feab:b7d0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:800 errors:0 dropped:0 overruns:0 frame:0
          TX packets:20 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:203511 (198.7 KiB)  TX bytes:1724 (1.6 KiB)
```

```
root@gnubee-pc1:~# wget gnubee-git.github.io/debian/debian-modules-install
converted 'http://gnubee-git.github.io/debian/debian-modules-install' (ANSI_X3.4-1968) -> 'http://gnubee-git.github.io/debian/debian-modules-install' (UTF-8)
--1970-01-01 00:09:11--  http://gnubee-git.github.io/debian/debian-modules-install
Resolving gnubee-git.github.io (gnubee-git.github.io)... failed: Name or service not known.
wget: unable to resolve host address 'gnubee-git.github.io'
```

`traceroute` seems to be giving me some interesting information, but I'm not quite sure what it's telling me:

```
root@gnubee-pc1:~# traceroute gnubee-git.github.io 
gnubee-git.github.io: Name or service not known
Cannot handle "host" cmdline arg `gnubee-git.github.io' on position 1 (argc 1)

# gnubee-git.github.io resolves to 151.101.201.147 on the other hosts on my network
root@gnubee-pc1:~# traceroute 151.101.201.147
traceroute to 151.101.201.147 (151.101.201.147), 30 hops max, 60 byte packets
 1  * * *
 2  * * *
 3  * * *
 4  * * *
 5  * * *
 6  * * *
 7  * * *
 8  * * *
 9  * * *
10  * * *
11  * * *
12  * * *
13  * * *
14  * * *
15  * * *
16  * * *
17  * * *
18  * * *
19  * * *
20  * * *
21  * * *
22  * * *
23  * * *
24  * * *
25  * * *
26  * * *
27  * * *
28  * * *
29  * * *
30  * * *
```

I feels to me like a DNS issue as I am able to successfully ping my router and various IP addresses around the internet, but not their names. The GNUBee DNS setup doesn't look incorrect to me compared to the other Linux hosts on my network:

```
root@gnubee-pc1:~# cat /etc/resolv.conf 
domain fios-router.home
search fios-router.home.
nameserver 192.168.1.1
```

All other hosts on my local network are able to communicate with the internet just fine so I'm scratching my head on this one. Any ideas / suggestions for diagnosis or treatment would be greatly appreciated - you have my greatest thanks.

dgazineu, 2017-09-29:
Hi,
   The recommended firmware for Debian/OMV at this point is the one with "official" kernel from here : [http://gnubee-git.github.io/debian/openwrt-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-03-01.bin](http://gnubee-git.github.io/debian/openwrt-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-03-01.bin)

Please refer to #18 for more information on other issues that he had with firmware librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-07-22.bin, which will have to be worked on.

You can replace firmware and resume the install process from your step 3.

indiebrain, 2017-09-29:
@dgazineu, thanks for the suggestion! I appreciate your reply. 

After reading through #18 I've flashed http://gnubee-git.github.io/debian/openwrt-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-03-01.bin to the GNUBee, but I'm seeing no change in behavior - debian boots, but I'm still not able to get out to the modules script with wget, or any other internet resource for that matter:

```
root@gnubee-pc1:~# wget gnubee-git.github.io/debian/debian-modules-install
converted 'http://gnubee-git.github.io/debian/debian-modules-install' (ANSI_X3.4-1968) -> 'http://gnubee-git.github.io/debian/debian-modules-install' (UTF-8)
--1970-01-01 00:04:29--  http://gnubee-git.github.io/debian/debian-modules-install
Resolving gnubee-git.github.io (gnubee-git.github.io)... failed: Name or service not known.
wget: unable to resolve host address 'gnubee-git.github.io'
```
I'm totally open to any other ideas.

bcupps, 2017-09-30:
@indiebrain check that your gateway is listed. I set a static IP in /etc/network/interfaces and added the gateway and it's resolving names now.

However, I have a new issue when trying to install OMV. apt-get is looking for mipsel in the OMV packages and that doesn't exist. Error is below. I've browsed all of the OMV distributions and none of them have a binary-mipsel version that I can find.

`W: Failed to fetch http://packages.openmediavault.org/public/dists/erasmus/InRelease  Unable to find expected entry 'main/binary-mipsel/Packages' in Release file (Wrong sources.list entry or malformed file)
`

Thoughts?
Thanks!

dgazineu, 2017-09-30:
Hi,

OMV for mipsel is not on their repositories, we have one compiled for GnuBee:
You will have to remove the openmediavault.org repository and add the following one:

   `deb https://gnubee-git.github.io/omv/ erasmus main`

If you prefer to use a script, these two commands will guarantee module installation and also install OMV:
```
  wget https://gnubee-git.github.io/debian/omv-jessie-install
  chmod +x omv-jessie-install
  ./omv-jessie-install
```

bcupps, 2017-09-30:
@dgazineu it looks like that worked! I will be going through the configuration tomorrow. Thanks!!

indiebrain, 2017-09-30:
@bcupps, thanks for the suggestion of using a static IP, I can now access network resources from the GNUBee! 

For reference, `/etc/network/interfaces` now looks something like this:

```
root@gnubee-pc1:~# cat /etc/network/interfaces
auto lo
iface lo inet loopback

auto eth0.2
iface eth0.2 inet static
    gateway 192.168.1.1
    netmask 255.255.255.0
    address 192.168.1.153
    broadcast 192.168.1.255
```

Also, FWIW before I statically assigned the network info to eth0.2, I attempted to verify that Debian was receiving a valid DHCP response from my router by looking at `/var/lib/dhcp/dhclient.eth0.2.leases` and everything looked OK to me. I'm not really sure why DHCP on this interface isn't working. I _think_ I'll stick with the static configuration for now, try to get the NAS functionally setup, then loop back to see if I can get any more info on this DHCP issue. 

Thank you, all!

closed 2017-09-30 by indiebrain

indiebrain, 2017-10-05:
To follow-up, I've managed to configure the GNUBee to use DHCP by removing the static `eth0.1` configuration. `/etc/network/interfaces` now looks like:

```
root@gnubee-pc1:~# cat /etc/network/interfaces
auto lo
iface lo inet loopback

auto eth0.2
allow-hotplug eth0.2
iface eth0.2 inet dhcp
```

With this configuration, I'm able to connect to network resources just fine, but it seems that the speed of the connection is terrible - file transfers shares on the NAS are capping out at about 5Mb/s - AFAIK, gigabit ethernet should be doing an order of magnitude or two better than this. I suspect that the GNUBee and my router aren't negotiating the correct connection speed/duplex settings and the [connection is falling back to something slower than optimal](https://wiki.debian.org/NetworkConfiguration#Setting_the_speed_and_duplex), but I'm still compiling information about this - more to come eventually.