gnubee-git/GnuBee_Docs #28: Host gnubee.org source on github

vincent-legoll, 2017-10-01:
So that we can contribute fixes easily, and tell us about it on this page: 
http://gnubee.org/contributing.html

pinney, 2017-10-07:
Hi Vincent : 

The pages are quite new and I think Niklaus is still on holiday.

However ... 
You can contribute in several ways including Pull Requests, patches, or by just opening an issue.

Adirelle, 2017-10-08:
I supports this. I finally  ave installed Debian Stretch on my GnuBee and have things to tell, those that were missing from the docs.

twotwenty, 2017-10-12:
I am interested in knowing about your debian stretch installation

Adirelle, 2017-10-12:
I firstly installed the firmware from there : https://github.com/gnubee-git/gnubee-git.github.io/blob/master/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-07-22.bin

Adirelle, 2017-10-12:
Well, I already installed a complete ArchLinux in a VM and have a prior experience with embedded systems, so I am used to install Linux from scratch using the command line.

I firstly installed the firmware from there : https://github.com/gnubee-git/gnubee-git.github.io/blob/master/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-07-22.bin The network configuration doesn't work but I am using a USB-to-UART connector so this is not an issue. I strongly recommend using the serial console; it is too easy to accidentally shut down the network or the SSH daemon and to get locked out.

I installed two disks in my GnuBee. On each disk I created two partitions using gdisk, one of 5GB, and a second one with the rest of the disk. I used mdadm to create a mirrored array using the two 5GB partitions. Note: as said in mdadm doc, be careful to create a disk with metadata 0.9, the kernel is unable to auto-assemble arrays with metadata 1.0+, thus it will now be able to boot from them. I created an ext4 partition labelled GNUBEE-ROOT on this mirrored volume.

I looked at the scripts at https://github.com/gnubee-git/gnubee-git.github.io/tree/master/debian and tried to debootstrap a debian stretch, but it failed for some reason about the repository (it seems the stretch repo only has compressed Release files and debootstrap is unable to use them with the LibreCMC firmware, but I am not sure), so I felt back to install debian jessie, and upgraded it to jessie afterwards.

About the Debian configuration, I wanted to use systemd-networkd and systemd-timesyncd, because they are installed anyway but I could not find how to tell systemd-networkd to configure the eth0.x interfaces (since the GnuBee provide a network switch chip, which is a bit different to configure than a "simple" network adapter), so I switched back to using the ifupdown package (e.g. /etc/network/interfaces configuration) and chronyd.

For the SSH server, dropbear is very lightweight. I prefer it when I don't need the fancy features of openssh-server. It is less memory- and cpu- intensive and it works with the sftp-server from the openssh suite. (BTW, @pinney, does the CPU have any built-in crypto functions, like AES ?)

Once the system was booting

So far most things are working but :
* acl, since the kernel-side support is disabled.
* ufw, which misses a nf_* module.
* minidlnad, because some illegal instruction error in one of the dependency. I filed a bug report to Debian about this.

I am considering rebuilding a kernel+initramfs myself using a more recent kernel but I am not sure about the required settings and drivers and the way to install them on the flash memory. (And cross-compiling the kernel is quite tedious)

twotwenty, 2017-10-22:
When I deboot strapped from gnubee I cannot remember what it was but I was able to debootstrap using my computer with a switch for alternative architecture. https://wiki.debian.org/Debootstrap

I want to know if I can boot from a USB or another drive and forget about using the microsd.

suppose I'll try when I get time. we should be able to pull the make menu config from some gnubee kernel already working.

dgazineu, 2017-10-23:
Hi,
   You can place the root filesystem for Debian in any ext2/3/4 partition
named GNUBEE-ROOT, in a microSD, USB drive or on a HDD/SDD, if you want.

vincent-legoll, 2017-10-23:
Do you mean: "with a filesystem label GNUBEE-ROOT", like done with:
`mkfs.ext4 -L GNUBEE-ROOT /dev/...`

dgazineu, 2017-10-23:
Yes Vincent, that's what I meant to.

dgazineu, 2017-11-29:
@vincent-legoll Thanks for your help and tips, if you feel right to do so, sources to contribute are @ https://gogs.librecmc.org/dgazineu/GnuBee-libreCMC
There is also a group to discuss and contribute @ https://groups.google.com/forum/#!forum/gnubee

Rgds,

closed 2017-11-29 by dgazineu