gnubee-git/GnuBee_Docs #29: Voltage for UART adapter?

nariox, 2017-10-01:
Hi, the CH340G supports both 5V and 3.3V. Which one should I use to connect to the GB-PC1?

dgazineu, 2017-10-01:
The interface is 3.3v, but you only have to connect TX, RX and GND.

nariox, 2017-10-01:
Thanks, I assumed so, but always good to be sure. Would anyone mind if I added this to the wiki?

closed 2017-10-01 by nariox

BzukTuk, 2017-10-01:
@nariox 
Please read issue #20. In my case, I had to connect gnubees PIN labeled RX to RX on my CP2102 and TX labeled pin to TX on CP2102. (GND to GND). Could you confirm this? Please let me know, if you have same issues as I did.
Thanks

nariox, 2017-10-17:
Hi @BzukTuk, I had figured out what was wrong my system without the UART, so it took me a while to actually test it out. For me it boots fine, but I did have to connect TX to TX and RX to RX.