gnubee-git/GnuBee_Docs #30: GnuBee USB-to-UART Adapter Not Working

leocrawford, 2017-10-04:
The GnuBee USB-to-UART Adapter I received gets hots, but doesn't create a USB connection on the PC. It also stops the device booting if plugged in. On taking it apart I found the case to be loose and only three connections made (white, green, black) whilst the other (orange) appears to be broken. I assume it's faulty, but perhaps it's meant to be like that?

pinney, 2017-10-04:
Hi Leo : 
The USB-to-UART should not get hot.
Does it get hot when plugged into a PC without being connected to the GnuBee?

Only three wires required ... so the wiring seems to be OK.

leocrawford, 2017-10-05:
Not managed to get it hot again whether plugged in or not. The bee reliably won't boot when plugged in to UART (even when that isn't plugged into PC)

When plugged in, and power cycle green light doesn't even flicker. Multiple tries all fail. When unpugged  green light comes staright on and boots cleanly.

pinney, 2017-10-06:
Let's try a simple test : 
Short Tip and Ring together with a metal object such a bare wire.
Then type something into your terminal emulator to test the USB to UART device.
[image]

leocrawford, 2017-10-08:
Hi,

I'm not getting anything. On first boot I saw some dmesg activity, now
nothing. I assume it failed when it got hot. Tried it in two laptops,
multiple USB ports and nothing.

As an alternative I do have a RS232 to USB adapter (TTL UART PL2303). Could
I cut the cable you provided and connect to that? Which colours would I
need to connect?

pinney, 2017-10-09:
I've tested a PL2303 USB to UART and those work fine. 
You should be able to use the cable ... please open the backshell and check it for shorts.
You only need three wires TX/RX and Ground.
The colors are mark in the photo above.
You can also swap TX/RX if it doesn't work ... in the past I have not noted any damage from reversed TX/RX lines ... other than you get a blank screen.

leocrawford, 2017-10-12:
Got that working thanks. There was no short on the 3.5mm connector, so I suspect whatever fault there was was on the USB connector itself.

closed 2017-10-12 by leocrawford