gnubee-git/GnuBee_Docs #31: OMV Update

menumsg, 2017-10-06:
OMV version that is installed is 3.0.65.  Is there a way to update to OMV 3.0.88 or at least 3.0.67?  I need version 3.0.67 or later for the URBACKUP plugin to install.

dgazineu, 2017-10-06:
Yes, and I am already tasked with OMV version upgrade on our repository.

ThomasKaiser, 2017-10-11:
You can update to latest OMV 3.x version and if you're somewhat adventurous even to OMV 4 Arrakis already (tested this a few times the last weeks doing a simple `omv-release-upgrade` but this will also upgrade Debian from Wheezy to Stretch so you should do a backup first)

Also I would strongly recommend to improve the OMV installation routine and most probably also cpufreq scaling settings (though never tested on MIPS so far).

ThomasKaiser, 2017-10-11:
Oh, please forget what I've written above. The moment I wrote MIPS it came to my attention that you have most probably prepare an own `mipsel` repo with stuff :(

ohrensessel, 2017-11-03:
Are there any news regarding the OMV update?

amatter, 2017-11-27:
I am as well interested in an OMV/debian update. 
So far my gnubee OMV is running with the latest firmware librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-18.bin.
Thank you

ohrensessel, 2017-12-30:
Hi,

I compiled the current OMV version and some additional packages, similar to what can be found in the [gnubee-git.github.io omv](https://github.com/gnubee-git/gnubee-git.github.io/tree/master/omv) repository.  Please note, that I take no responsibility for problems resulting from the use of these repositories. Most of the stuff is untested, as I only use a subset of the packages myself. For now, packages are unsigned.

`deb http://apt.ohrensessel.net/omv erasmus main
deb http://apt.ohrensessel.net/omv-extras erasmus main
deb http://apt.ohrensessel.net/gnubee-extras erasmus main`

As my repositories do not contain all packages found in the gnubee repository (e.g. proftpd-mod-vroot) mentioned above, I recommend using them additionally.

Let me know if this helps you and if you need any additional packages.

amatter, 2017-12-31:
hello,

I added to above 3 lines to /etc/apt/sources.list.d/omv.list : 
```
 root@gnubee-n1:/etc/apt/sources.list.d# less omv.list
 deb https://gnubee-git.github.io/omv/ erasmus main
 # added according to: ttps://github.com/gnubee-git/GnuBee_Docs/issues/31
 deb http://apt.ohrensessel.net/omv erasmus main
 deb http://apt.ohrensessel.net/omv-extras erasmus main
 deb http://apt.ohrensessel.net/gnubee-extras erasmus main
```
after running 
apt-get update
apt-get upgrade 

I get : 
root@gnubee-n1:/etc/apt/sources.list.d# apt-get upgrade
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Paketaktualisierung (Upgrade) wird berechnet... Fertig
Die folgenden Pakete sind zurückgehalten worden:
  openmediavault
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 1 nicht aktualisiert.

I guess I could force it ... but this could cause an eventually unstable system?

Actualy I like the gnubee but I'd really like to keep my "gnubee OMV System" on an actual base. 

Thanks to ohrensessel for his effort!

ohrensessel, 2018-01-02:
Hi @amatter,

you have to run `apt-get dist-upgrade`. See the [man page of apt-get](https://linux.die.net/man/8/apt-get) for an explanation. Seems the newer openmediavault version brings updated dependencies (some added, some removed) which makes this step necessary.

amatter, 2018-01-02:
Good evening @ohrensessel 
Thanks for your explanantion. I did `apt-get dist-upgrade` => the system seems to run without problems.

regards Adrian

noblepepper, 2018-01-03:
Seems to work here too, fixed a couple of glitches I had with 3.0.65!

Thanks!

rogermartinor, 2018-04-02:
Hello - Thanks go to orhensessel for providing an updated version of OpenMediaVault for GnuBee!
I'm trying to install the SyncThing package on OMV, __openmediavault-syncthing__.  
Dependencies for the current version are listed at https://bintray.com/openmediavault-plugin-developers/erasmus-sync/openmediavault-syncthing/view.

OMV  |  Syncthing  |  SyncThing Date  |  Indexed Architecture  |  Explanation
------  |  ---------  |  ----------------  |  ---------------------  |  ------------
3.0.65  |  none  |  n/a  |  MT7621A  |  GnuBee release              
3.0.40  |  3.8  |  2016-09-28  |  armhf, i386, amd64  |  Quite obsolete
3.0.67  |  3.10  |  2017-05-25  |  arm64, armhf, i386, amd64  |  Obsolete
3.0.67  |  3.11  | current version  |  arm64, armhf, i386, amd64  |  Current versoin
3.0.96  |  none  | n/a  |  MIPS 1004 Kc V2.15  |  orhensessel build

My assumption was that __openmediavault-omvextrasorg__ plugin could install other plugins. It appears that all other plugins also need to be compiled into the build. Is this correct? Is this due to the MIPS dependency?

How big an effort was it to build OMV from source? I'm not quite sure where to start; I consider myself to be a noob. How do you determine which GnuBee source to merge with OpenMediaVault source? Are there sections where you have to hand-code GnuBee fixes in new OpenMediaVault sources? Do you create a difference list? I see references in the error message that I got when I tried to install the package.

http://apt.ohrensessel.net erasmus Release.gpg Hit http://httpredir.debian.org jessie/main Translation-en Hit http://apt.ohrensessel.net erasmus Release Hit http://apt.ohrensessel.net erasmus Release Hit http://apt.ohrensessel.net erasmus Release Ign http://apt.ohrensessel.net erasmus/main mipsel Packages/DiffIndex Ign http://apt.ohrensessel.net erasmus/main mipsel Packages/DiffIndex Ign http://apt.ohrensessel.net erasmus/main mipsel Packages/DiffIndex Get:14 http://apt.ohrensessel.net erasmus/main mipsel Packages [5751 B] Ign http://apt.ohrensessel.net erasmus/main Translation-en Get:15 http://apt.ohrensessel.net erasmus/main mipsel Packages [5950 B] Ign http://apt.ohrensessel.net erasmus/main Translation-en Get:16 http://apt.ohrensessel.net erasmus/main mipsel Packages [844 B] Ign http://apt.ohrensessel.net erasmus/main Translation-en Fetched 27.9 kB in 34s (819 B/s) W: Failed to fetch https://dl.bintray.com/openmediavault-plugin-developers/erasmus/dists/jessie/Release Unable to find expected entry 'main/binary-mipsel/Packages' in Release file (Wrong sources.list entry or malformed file) E: Some index files failed to download. They have been ignored, or old ones used instead.

Any help or insights would be appreciated.
Thanks, Roger

riverside6, 2018-11-25:
Hello @dgazineuWill there be a current OMV 4 release for the GnuBee PCx?

abesnier, 2019-09-09:
Sorry to revive an old post, but I just reinstalled everything on my PC1, so here's my feedback.
OMV4 can be built easily. [The wiki ](https://github.com/gnubee-git/GnuBee_Docs/wiki/Stretch-OMV)gives the proper directions to do so.
It works fine up to the latest commit on branch 4.x (commit 42f2bf03ee3dc391e6f15e048e3e977e0f9ae595 as of today). It requires to make a package for [WSDD ](https://github.com/openmediavault/wsdd)too, and the extras plugins (such as [MySQL ](https://github.com/OpenMediaVault-Plugin-Developers/openmediavault-mysql)or [Nginx](https://github.com/OpenMediaVault-Plugin-Developers/openmediavault-nginx), the ones I use...) need to be built separately. But if you can manage to build the main OMV, plugins are a breeze !

abesnier, 2019-09-10:
I've copied the deb packages here, if someone is interested:
https://github.com/abesnier/gnubee-omv4-packages/