gnubee-git/GnuBee_Docs #32: Is ACL enabled in the librecmc firmware ?

Adirelle, 2017-10-09:
I have installed my GnuBee using the LibreCMC firmware from 2017-07-22 (i.e. librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-07-22.bin).

I want to enable the acl but it does not seem to work: when using the "acl" option at (re)mount, it is not listed when checking the mount with "mount" and trying to use setfacl causes "operation not supported" errors.

I think these options were not enabled in the kernel configuration (see http://www.projectenvision.com/blog/4/Enable-Support-for-ACL-in-Debian-Ubuntu) but I could not find the kernel config, neither as /proc/config nor in the build tools (but maybe I overlooked them).

Are these options enabled in this firmware ?

By the way, what is the firmware build procedure ? (to produce a *.bin file)

pinney, 2017-10-09:
ACL is not enabled by default in the firmware.

To do so one only needs to run 'make kernel_menuconfig' and enable ACL

To build firmware basically : 

git clone 'target repository' for libreCMC 
You can use : 
https://gogs.librecmc.org/ldpinney/GnuBee-libreCMC

Then : 
'make menuconfig'
and 
'make'

for more details please see : 
https://wiki.openwrt.org/doc/howto/build

Adirelle, 2017-10-09:
Could you please provide the .config file of your build root after a successful build ? This is where the menuconfig stores it configuration (and what is usually made available through /proc/config). Starting from this file would be easier than trying to guess all parameters...

pinney, 2017-10-09:
https://gogs.librecmc.org/ldpinney/GnuBee-libreCMC/src/master/GB-PC1_defconfig

[image]

Adirelle, 2017-10-11:
I have compiled it but I didn't have time to test it yet.

However, your screenshot shows how to compile the acl user-space tools, which are already included in Debian distributions (which I use).

What is missing is the kernel-side support of these features for different filesystems. You might want to include it in the base firmware, as users might want to use them. We are speaking of a NAS, after all.

dgazineu, 2017-10-11:
You are right, the kernel support for acl will be included for the next
base firmware iterations (both older and newer kernels).

Adirelle, 2017-10-11:
By the way, nf_conntrack_netbios_ns is also missing, which prevents to use ufw. I learnt it the hard way at  activation: it could not add any ALLOW rule after denying all, including SSH...

Adirelle, 2017-10-13:
I came across this while looking at the compilation : https://wiki.openwrt.org/doc/howto/building_kernels_for_debian_binaries . Some errors (like #33) might be linked with the default openwrt options for the kernel. I will try to compile the kernel with FPU emulation enabled.

pinney, 2017-10-14:
New GB-PC1_defconfig: add support for filesystem ACL and Ath9k Wireless
https://gogs.librecmc.org/ldpinney/GnuBee-libreCMC/commit/8eb5bdb31c74345e53becf9003a00cdec465a5d1

Adirelle, 2017-10-15:
I have finally tested the firmware I compiled. It seems to work but does not search nor pivot to GNUBEE-ROOT partition. Is this customization included in https://gogs.librecmc.org/ldpinney/GnuBee-libreCMC ?

dgazineu, 2017-11-29:
Updated LibreCMC firmware @ http://gnubee.org/install_debian.html (https://github.com/gnubee-git/gnubee-git.github.io/raw/master/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-28.bin)
Sources updated @ https://gogs.librecmc.org/dgazineu/GnuBee-libreCMC

closed 2017-11-29 by dgazineu