gnubee-git/GnuBee_Docs #33: Illegal instruction when starting minidlna

Adirelle, 2017-10-09:
I installed Debian Jessie and upgraded to Stretch aftewards. Everything seems to work fine, but minidlna.

When I start it, it crashes with an "Illegal instruction" error. I am not sure if it is related to the firmware (librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-07-22.bin), some incompatible option in the stretch configuration for mipsel of another thing.

I'm am not fond of reinstalling the whole system with jessie without be sure it will fix the issue. Could someone test minidlna from jessie with that firmware and tell me if it also crashes ?

Adirelle, 2017-10-10:
Ok, so it seems the culprit is actually libopenmpt. This is what I obtain when I try to recompile it:

```(gdb) run
Starting program: /home/adirelle/build/libopenmpt-0.2.7386~beta20.3/libopenmpttest

Program received signal SIGILL, Illegal instruction.
OpenMPT::izero (y=<optimized out>) at soundlib/Tables.cpp:852
852                     d = d + 2; ds = ds * (y*y)/(d*d);
```

I suspect that changing something in the compilation option could fix it but I am not competent enough to know what.

closed 2017-10-10 by Adirelle

Adirelle, 2017-10-13:
For the record, this is probably related to this:  https://wiki.openwrt.org/doc/howto/building_kernels_for_debian_binaries#floating_point_unit_mips

dgazineu, 2017-10-16:
> For the record, this is probably related to this:
> https://wiki.openwrt.org/doc/howto/building_kernels_for_
> debian_binaries#floating_point_unit_mips

Thanks for pointing out your experience, those options are already
turned on, or else Debian Jessie did not even boot.
   The installation of Stretch was unstable in some trials, hence the
supplied scripts are still for Jessie, while we work out the issues.
   As more people, like you, experiment and share its findings, it will be
possible to evolve even more in this platform.

Rgds,

Adirelle, 2017-10-16:
Hello @dgazineu,

And yes, I experienced the same issue using a kernel I recompiled myself with these options enabled, so this is not related to the FPU emulator (unless it is buggy).

How can I reproduce the build you used for https://github.com/gnubee-git/gnubee-git.github.io/blob/master/debian/openwrt-ramips-mt7621-gb-n1-squashfs-sysupgrade.bin ? Larry pointed me to this repository: https://gogs.librecmc.org/ldpinney/GnuBee-libreCMC, but it seems that it does not include the scripts to find and pivot to the GNUBEE-ROOT partition.

Several versions of the firmware are available (labelled "libreCMC", "Debian", "LEDE") but none of them are clearly linked to their complete sources and/or compilation instructions. So I cannot reproduce them identically. It kinda feels like "closed" source software, e.g. : report bug then wait until the editor fixes them and provides a new binary. I do not mind reporting what I find but I also would like to test things myself. You already have done a lot of work on the kernel and the firmware and without the sources and build instructions, it is like I have to restart from scratch (or at least from the "stock" openwrt builds). Moreover, as it is not clear how each of the firmware versions were compiled and which features were enabled or not, it is hard to tell which one we should pick and what we should expect to test. Note this is mostly documentation issues, but ones that cannot be fixed by third persons.

dgazineu, 2017-10-16:
Right now, the two provided firmware with Debian support are on:

 - https://github.com/gnubee-git/gnubee-openwrt (branch master, with official kernel from Soc vendor, a little old, but enough for the device as a NAS, more stable at the moment) 
and
 - https://github.com/gnubee-git/GnuBee-lede/tree/lede-17.01 (branch lede-17-01, the kernel is updated, but still a work in progress, some kirks appearing and being worked out)

On the root folder of both there is a file, (GB-Deb.config / GB-PC1_defconfig) to be copied to .config, that have the main config ( as seen on "make menuconfig" ). Kernel configuration ( done by "make kernel_menuconfig" ) sets non-default configuration needed for Debian and the choosen modules to be built-in (something that would be loaded after boot in the pure firmware, but is needed to boot, for example, or specific kernel features), and is already saved to "target/linux/ramips/mt7621/config-x.yy" respective files.

We are encouraging the use of a group on [https://groups.google.com/group/gnubee](https://groups.google.com/group/gnubee) to discuss ideas and hence improve the documentation we have.

Rgds,
For instructions on building, please refer to h[ttps://wiki.openwrt.org/doc/howto/build](ttps://wiki.openwrt.org/doc/howto/build)

cmm, 2018-02-13:
in case anyone stumbles on this: one work-around I accidentally found for minidlna is to build the libopenmpt0 package from sources with optimization off (I just wanted to debug it, per https://wiki.debian.org/HowToGetABacktrace, but the SIGILL didn't reproduce without optimization).

minidlna still crashes when started the first time (a SIGSEGV somewhere outside libopenmpt, so probably unrelated to the problem here), but after restarting seems to run just fine.

cmm, 2018-02-13:
Digging some more:

The offending instruction is MTHC1, which is a MIPS32R2 instruction.  

Now, one would be tempted to suspect the fact that Debian Stretch is compiled for mips32r2 (Jessie and earlier were mips32) -- I know I was! -- but compiling the offending source with explicit -march=mips32 yields the innocent-looking MUL.D instead, which is _also_ causing SIGILL.  So, I dunno.

(Does MT7621A implement MIPS32R2, anyway?  I wasn't able to determine by googling.  If it doesn't, I guess we are royally screwed...)

Adirelle, 2018-02-13:
Isn't there a way to disable hardware FPU at compilation to use (slow, )software emulation from libraries ?

cmm, 2018-02-14:
There is -msoft-float and it would actually be slightly faster than the in-kernel FP emulation, but gcc in Debian doesn't have it configured (because they argue, quite rightly, that since there is a working FP emulator in the kernel, it's simpler to just pretend every MIPS CPU Debian runs on has an FPU and be done with it).  And it would be another work-around anyway, I'd rather someone smart would figure out the actual problem :/