gnubee-git/GnuBee_Docs #34: Power-on issue

Libbum, 2017-10-11:
Received my kit today and it doesn't seem to be powering up at all. Using a 12 V, 3A adaptor (center-positive); but I see no status lights coming on when I press the power button.

Where's the best place on the board to troubleshoot with a multimeter? I can confirm there's 12V on-board at the power sockets feet, but can't easy follow the power distribution from there.

pinney, 2017-10-11:
@Libbum 
The schematic is here : 
https://github.com/gnubee-git/GnuBee_Docs/blob/master/GB-PC1/Documents/GB-PC1_V1.0_Schematic.pdf

You may want to check the switch.

Are any drives installed? or any other things such as a USB drive or UART cable?

Libbum, 2017-10-12:
@pinney 
No drives, usb or uart in yet - haven't got that far. 
Switch is fine from now checking power across multiple components. All the following capacitors have their expected, associated voltages:

C4250: 12V
C0176: 5V
C1078: 3.3V
C1713: 1.0V
C1709: 1.14V

So power distribution up to that point seems good. Not sure what to look at next though...

pinney, 2017-10-12:
@Libbum 

I'm not sure what happened I powered on every board then changed the bootloader and firmware.

I think this board need exchanged under warranty.

Libbum, 2017-10-13:
@pinney 
Ok, thanks for the info. I'll get in contact with Crowd Supply. 
(Will close this issue for now too).

closed 2017-10-13 by Libbum