gnubee-git/GnuBee_Docs #35: howto use microsd

twotwenty, 2017-10-12:
can I put an image on the microsd?

leocrawford, 2017-10-12:
Good quetsion; the order form read "Run your GnuBee Personal Cloud 1 directly from the microSD card. " yet my only uboot options are

    Please choose the operation: 
    1: Load system code to SDRAM via TFTP.
    2: Load system code then write to Flash via TFTP.
    3: Boot system code via Flash (default).
    4: Enter boot command line interface.
    5: Load system code then write to Flash via USB Storage.
    6: Load system code then write to Flash via Httpd.
    9: Load U-Boot code then write to Flash via TFTP.      

Selecting 5 gives:

    No USB Storage found. Upgrade FW failed!

Booting via flash mounts the debian MicroSD card so I know it's working

Adirelle, 2017-10-12:
The firmware looks for a partition labelled "GNUBEE-ROOT" on any available device. Create a proper rootfs with that label on the microSD card and you should be good to go.

pinney, 2017-10-13:
To run Debian directly from the microSD card you will need to use this firmware 
https://github.com/gnubee-git/gnubee-openwrt/blob/master/binaries/openwrt-ramips-mt7621-gb-pc1-squashfs-sysupgrade.bin

This one uses an older kernel.
The later kernels support the microSD card but can not boot from it yet.

twotwenty, 2017-10-13:
can have the /boot from the internal memory and / from micro sd?

twotwenty, 2017-10-13:
can a newer kernel boot from the internal flash and / be from the microSD?

pinney, 2017-10-13:
@twotwenty 
The problem with using the newer kernel is the mmc module is only available after 'init' phase.
Which is needed to use a SD card for the root filesystem.

my two cents : 
One shouldn't use a SD card for the OS.
A search for 'raspberry pi corrupt sd card' should show ample reason.

I think a better solution might be using a pair of USB Drives with linux mdraid

twotwenty, 2017-10-14:
"To run Debian directly from the microSD card you will need to use this firmware"
can I run debian from a usb drive and have a newer kernel?

it is true microSD is plagued by poor quality but you can also purchase one that is reliable, power issues can cause corruption in non journal file systems ie fat  ... it is a large subject of how corruption is caused.

twotwenty, 2017-10-14:
can I use the newer kernel with debian on a USB?

pinney, 2017-10-15:
Yes ... you should be able to use the script found here : 
https://github.com/gnubee-git/gnubee-git.github.io/blob/master/debian/README.DEBIAN.txt

To install Debian to a USB Drive or SATA Drive.

Adonai, 2017-10-15:
@pinney 

> The problem with using the newer kernel is the mmc module is only available after 'init' phase.
Which is needed in the 'init' to use a SD card for the root filesystem.

Isn't this an initramfs' job to initialize all devices? Can't figure out what exactly changed between the kernel versions...

twotwenty, 2017-10-15:
I second some other threads"Thank you for bringing us freedom!"

 FSF or GNU.org let's you keep the name gnubee maybe we could chip in and donate them one loaded anyway cause they are great.

Adonai, 2017-10-15:
Just installed Debian on 32GB Class 10 sdcard with kernel 3.10, works like a charm.
Thanks @pinney ! Let's work together to bring 4.4.x kernels to it!

Adirelle, 2017-10-15:
@twotwenty I used Debian's "mipsel" architecture. There is something fishy with the FPU emulator (see #40) which can bite you when using some packages, like minidlna.

twotwenty, 2017-10-22:
I am satisfied with the answer so closing this ticket is ok.
Thanks @pinney

twotwenty, 2017-10-22:
To boot from MicroSD as currently requires an older firmware/kernel

refacto, 2017-11-27:
@pinney I'd like to install Debian on a MD RAID 1 on two USB sticks, but the script only looks for /dev/sd* and /dev/mmcblk0p devices. Is it okay to modify it so that it also accepts /dev/md* devices? I'll make a pull request for it to show it more precisely.

dgazineu, 2017-11-29:
As this issue turned into several ones, I am closing it for now. Please re-open separatelly any remainning issues, thanks.

closed 2017-11-29 by dgazineu