gnubee-git/GnuBee_Docs #37: No interface for USB-Wifi-adapter

DurtalDeepDown, 2017-10-12:
Dear GnuBees, 

I do not have permanent physical access to my router. So I planned to connect the GnuBee to my network via an USB-Wifi-adapter and bought this one https://www.thinkpenguin.com/gnu-linux/penguin-wireless-n-usb-adapter-gnu-linux-tpe-n150usb . I installed debian and the modules according to the manual. The USB-adapter is listed by lsusb: 
`Bus 001 Device 002: ID 0cf3:9271 Atheros Communications, Inc. AR9271 802.11n`
But whatever I tried, I couldn't manage to get up a wireless interface. Most notably I followed the instructions here: https://www.thinkpenguin.com/gnu-linux/penguin-wireless-n-usb-adapter-gnu-linux-tpe-n150usb-debian-firmware-install , tried but failed to build the firmware on my own according to https://wiki.debian.org/ath9k_htc/open_firmware, finally installed firmware-ath9k-htc. But nothing changed. ip link still gives only: 
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1
link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00 
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 1000
link/ether 78:a3:51:ab:b5:90 brd ff:ff:ff:ff:ff:ff 
5: eth0.1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
link/ether 78:a3:51:ab:b5:90 brd ff:ff:ff:ff:ff:ff 
6: eth0.2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
link/ether 78:a3:51:ab:b5:90 brd ff:ff:ff:ff:ff:ff 
```

I researched for two nights but didn't find anything useful and so I am stuck here. May that be because there are no ath9k.... modules available?

Adirelle, 2017-10-12:
This is probably the case.

DurtalDeepDown, 2017-10-13:
Hum, too bad. Is there a way to fix this? Can I just copy a ko-file to /lib/modules/`uname -r`? But where do I get it?
Sorry for these naive questions. I never fiddled around with the kernel.

pinney, 2017-10-14:
New GB-PC1_defconfig: add support for filesystem ACL and Ath9k Wireless
https://gogs.librecmc.org/ldpinney/GnuBee-libreCMC/commit/8eb5bdb31c74345e53becf9003a00cdec465a5d1

DurtalDeepDown, 2017-10-16:
Thank you for the update.
I tried a lot but didn‘t manage to achieve anything because I lack essential coding skills. I cloned the repository, renamed "GB-PC1_defconfig" to ".config" and run make. But this produces errors I do not understand:
```
Collected errors:
* satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-ath9k-htc:
 *      ath9k-htc-firmware *
 * opkg_install_cmd: Cannot install package kmod-ath9k-htc.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-carl9170:
 *      carl9170-firmware *
 * opkg_install_cmd: Cannot install package kmod-carl9170.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-ddns:
 *      ddns-scripts *
 * opkg_install_cmd: Cannot install package luci-app-ddns.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-hd-idle:
 *      hd-idle *
 * opkg_install_cmd: Cannot install package luci-app-hd-idle.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-minidlna:
 *      minidlna *
 * opkg_install_cmd: Cannot install package luci-app-minidlna.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-mjpg-streamer:
 *      mjpg-streamer *
 * opkg_install_cmd: Cannot install package luci-app-mjpg-streamer.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-p910nd:
 *      p910nd *
* opkg_install_cmd: Cannot install package luci-app-p910nd.
```

Tested on an uptodate arch- and trisquel-system. I guess I skipped an important step but couldn't find help anywhere. Can anyone give me a hint how to fix this? 
Furthermore: Will the updated firmware lead to a modified debian-install? It would be nice to run libreCMC via WLAN. But I hope to run the GnuBee with a WLAN-connected debian some day.

pinney, 2017-10-16:
@DurtalDeepDown 

Before you run 'make' 
run 'make menuconfig'

If you email me I'll send you my current configs

pinney, 2017-10-16:
######## REF ########

https://wiki.openwrt.org/doc/howto/buildroot.exigence

https://wiki.openwrt.org/doc/howto/build

######## GIT #########

git pull && ./scripts/feeds update -a && ./scripts/feeds install -a

######## BUILD ########

make menuconfig

make -j1 V=s download

ionice -c 3 nice -n 19 make -j1 V=s | tee build.log

######## CLEAN ########

make dirclean

make target/linux/clean

DurtalDeepDown, 2017-10-26:
Sorry for leaving this issue open but I spent a terrible frustrating week trying to build the firmware. All attempts (with different config-files on various systems) failed even on a brand new debian stretch. 
Strangely enough, after running `make menuconfig`, selecting a config-file, and exiting, I am not able to run `make`. This will get me back to the menu. The only way to fix this is to save the configurations to .config. Don't think this to be crucial but it seems strange  to me. 
After renaming the file, `make -j1 V=s download ` exits seemingly ok. But a closer inspection gives these errors for which I couldn‘t find any help online. 
```
bash: mipsel-openwrt-linux-musl-gcc: command not found
bash: mipsel-openwrt-linux-musl-gcc: command not found
bash: /home/matthias/GnuBee-libreCMC/staging_dir/target-mipsel_24kc_musl-1.1.16/host/bin/gettext: No such file or directory
bash: /home/matthias/GnuBee-libreCMC/staging_dir/host/bin/sed: No such file or directory
can't open file '/home/matthias/GnuBee-libreCMC/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7621/linux-4.4.52/.config' at /home/matthias/GnuBee-libreCMC/scripts/kconfig.pl line 32.
grep: /home/matthias/GnuBee-libreCMC/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7621/linux-4.4.52/.config: No such file or directory 
```
Eventually `make -j1 V=s ` exits with 
```
Configuring kmod-ledtrig-netdev.
Collected errors:
satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-ath9k-htc: 
* ath9k-htc-firmware *
* opkg_install_cmd: Cannot install package kmod-ath9k-htc.
* satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-carl9170:
* carl9170-firmware *
* opkg_install_cmd: Cannot install package kmod-carl9170.
package/Makefile:60: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/matthias/Gnubee-libreCMC'
package/Makefile:102: recipe for target '/home/matthias/Gnubee-libreCMC/staging_dir/target-mipsel_24kc_musl-1.1.16/stamp/.package_install' failed
make[1]: *** [/home/matthias/Gnubee-libreCMC/staging_dir/target-mipsel_24kc_musl-1.1.16/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/matthias/Gnubee-libreCMC'
/home/matthias/Gnubee-libreCMC/include/toplevel.mk:197: recipe for target 'world' failed
make: *** [world] Error 2 
```
For reasons I do not understand `make -j1 V=s | tee build.log` omits some of these lines... 
I did a lot of research but couldn’t find anything. So I wonder if anybody succeeded in building the firmware from the repository https://gogs.librecmc.org/ldpinney/GnuBee-libreCMC.git Where is the blunder? What am I missing?

dgazineu, 2017-10-26:
Hi,
   The first error shows a failure on toolchain (cross compiler) files
generation/location:

bash: mipsel-openwrt-linux-musl-gcc: command not found

I'm in process of fixing package lists for
"https://github.com/gnubee-git/gnubee-openwrt" branch, and will take a
look in libreCMC code.

As openwrt/lede/librecmc attempt to build the cross-compiler from
external sources/packages (and also use external sources to include
packages on firmware), they are basically rolling-code generation,
something get broken on the build environment from time to time.

dgazineu, 2017-11-20:
New firmware available @
https://github.com/gnubee-git/gnubee-git.github.io/raw/master/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-18.bin,
with Debian/OMV support.
    Install instructions are the same as before, updated on
http://gnubee.org/install_debian.html as well.

    The updated sources are @
https://gogs.librecmc.org/dgazineu/GnuBee-libreCMC

Please give a try and report back.

DurtalDeepDown, 2017-11-22:
Nope. I saved GB-PC1_defconfig as .config. (The same holds for the unchanged GB-PC1-Debian_defconfig.) Building the firmware fails with:

```
make[5]: Entering directory '~/GnuBee-libreCMC/build_dir/target-mipsel_24kc_musl-1.1.16/linux-ramips_mt7621/linux-4.4.87'
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/bounds.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  CC [M]  drivers/mmc/host/mtk-mmc/sd.o
scripts/Makefile.build:258: recipe for target 'drivers/mmc/host/mtk-mmc/sd.o' failed
scripts/Makefile.build:403: recipe for target 'drivers/mmc/host/mtk-mmc' failed
scripts/Makefile.build:403: recipe for target 'drivers/mmc/host' failed
scripts/Makefile.build:403: recipe for target 'drivers/mmc' failed
Makefile:964: recipe for target 'drivers' failed
make[5]: Leaving directory
```

What does the Script GB-PC1_debian-modules.sh do?

DurtalDeepDown, 2017-11-22:
The updated firmware-bin also comes with a weird issue. No problems with logging in into libreCMC. But the debian-install-script gives: 
`W: Cannot check Release signature; keyring file not available /usr/share/keyrings/debian-archive-keyring.gpg `
Not sure whether it is because of that, but afterwards retrieving and validating runs in super slowmo. It now runs for more than 24h. I am afraid I have to abort it today. I have no clue what may have caused this but I can reproduce it. There are no connection problems. Did anyone else encounter this?

dgazineu, 2017-11-22:
The script "GB-PC1_debian-modules.sh" create a tarball with compiled modules from kernel, to be uncompressed @ /lib/modules after Debian installation (or firmware/kernel upgrade).

I'll check again into mtk-mmc module, as there may be a missing patch for it on the buildroot.

dgazineu, 2017-11-22:
The mmc patch for mt7620 was indeed missing. 
Just updated source @ https://gogs.librecmc.org/dgazineu/GnuBee-libreCMC 
Update from that repo, clean and prepare the kernel ( make target/linux/{clean,prepare} ) and try compiling again.

DurtalDeepDown, 2017-11-22:
Thanks for the update. That brought me further but now I receive the known errors from the old repository:

```
Configuring luci-apCollected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-ath9k-htc:
 * 	ath9k-htc-firmware * 
 * opkg_install_cmd: Cannot install package kmod-ath9k-htc.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-carl9170:
 * 	carl9170-firmware * 
 * opkg_install_cmd: Cannot install package kmod-carl9170.
p-commands.
```

dgazineu, 2017-11-22:
These specific drivers need blobs (firmwares) that LibreCMC doesn't have.

DurtalDeepDown, 2017-11-22:
Does that mean that I cannot use this wifi-dongle (or any other, since this one is 100% libre) on the GnuBee?

dgazineu, 2017-11-22:
There are other ath9k (without htc) in the kernel options. Don't know about
carl9170...
Try unselecting those *firmware options and see what you got working.

There is also the openwrt-based firmware for it that may support this
hardware.

neheb, 2017-11-22:
ath9k is pcie only. _htc is for a select few USB cards.

BTW LibreCMC seriously needs to include the firmware for the _htc cards. The firmware is completely open source. This sounds like an accident or oversight on LibreCMC's part.

pinney, 2017-11-22:
Please see :https://gogs.librecmc.org/libreCMC/libreCMC/pulls/20

DurtalDeepDown, 2017-11-23:
OK. I give up the libreCMC approach for now and turn to openWRT. Building from the repository https://github.com/gnubee-git/gnubee-openwrt This fails with:

```
In file included from ~/gnubee-openwrt/build_dir/toolchain-mipsel_1004kc_gcc-4.8-linaro_uClibc-0.9.33.2/gcc-linaro-4.8-2014.04/gcc/cp/except.c:1008:0:
cfns.gperf: In function 'const char* libc_name_p(const char*, unsigned int)':
cfns.gperf:101:1: error: 'const char* libc_name_p(const char*, unsigned int)' redeclared inline with 'gnu_inline' attribute
cfns.gperf:26:14: note: 'const char* libc_name_p(const char*, unsigned int)' previously declared here
cfns.gperf: At global scope:
cfns.gperf:26:14: warning: inline function 'const char* libc_name_p(const char*, unsigned int)' used but never defined
Makefile:1058: recipe for target 'cp/except.o' failed
make[6]: *** [cp/except.o] Error 1
```

dgazineu, 2017-11-23:
I got similar errors to build the toolchain for openwrt when tried to
compile everything from scratch on an Ubuntu 17.04 machine. Latest LibreCMC
went fine, but the older openwrt brought those errors.

Right now, my environment for both of them is based on Ubuntu 16.04.

Rgds,

DurtalDeepDown, 2017-11-24:
Whew hew, whew hew, whoo hoo! That finally did it!

I can confirm that libreCMC (without the wireless modules) builds fine on a uptodate debian stretch. Building the openWRT firmware (with those modules) requires a somewhat older system. Ubuntu16.04 does it. I also had success on a reasonably recent Trisquel system. Maybe you should mention this requirement in the README.

It also should be mentioned (because I didn’t read it anywhere) that you reach the openWRT system not on http://192.168.10.1 but on http://192.168.1.1 . The password is GnuBee.

Installing Debian went fine without the aforementioned "bug"(?).

After installing and loading the modules and the free firmware in Debian, I got a wlan interface. For reasons unknown, iwconfig shows that this interface has no wireless extensions. But following [these instructions](https://wiki.debian.org/WiFi/HowToUse#WPA-PSK_and_WPA2-PSK)   I was able to connect.

@pinney: I fully understand  [your scepticism](https://github.com/gnubee-git/GnuBee_Docs/issues/38) about running a NAS on WLAN. But in a shared living space not everything can be set up according to technical perfection. 

Thank you for your amazing effort and for your patience. Closing the issue, opening a beer.

closed 2017-11-24 by DurtalDeepDown