gnubee-git/GnuBee_Docs #38: Firmware questions

Adirelle, 2017-10-13:
The current firmware is good for bootstrapping but it still has a couple of issues with the provided kernel :
* There is a terrible clock skew, to the point that neither ntpd nor chrony are able to correct it ; which prevents any kind of serious backuping or synchronisation.
* Some modules are missing : acl support, nf_*, wireless drivers ...
* It is based on an old version which might have a buggy software FPU emulator (unconfirmed).
* The boot sequence is quite slow (low priority).

I am considering building a kernel and initramfs from the mainstream kernel. I do not care about the embedded distribution. Ideally, I would put the initramfs and kernel on my root partition and load them directly from U-boot. That would allow me to easily update the kernel, as well as fall back to the firmware if something goes wrong. But before I eventually start, I have a some questions about the current firmware:

* Concerning U-boot:
  * Would it be able to load a kernel and a initramfs from an fat/ext4 partition on a SATA disk ? i.e. have the necessary features been enabled at compile-time ?
  * Has it some persistent storage to save the configuration between reboot ? I think this is called "environment" in u-boot terminology.
* What are the kernel specificities ?
  * Has it be source-patched ? i.e. patches to the C sources. If yes, where are these patches ?
  * Is there specific configuration ? Which ? And where can they be found ?
  * What are the drivers required to handle all GnuBee-PC1 hardware "out-of-the-box" ?

Thanks.

Adirelle, 2017-10-14:
I have found some answers myself reading the openwrt source and documentation:
* Yes, the kernel has source patches.
* Yes, openwrt handle some specific configuration.
* The "illegal instruction" error may be related to the FPU emulator.

pinney, 2017-10-14:
Actually I don't plan to do anything with this 'laundry list'

Please close it and open an issue for each umm... issue?

Adirelle, 2017-10-14:
@pinney Sorry if it was unclear but I this is not a list of "action requests" but "information requests". I already tinkered embedded device (including configure u-boot, installing kernel and initramfs on flash/internal emmc) but I am not used to the OpenWRT distribution, and not to the way you taylored it for the GnuBee PC-1. So I asking questions to understand what is going on and how I could change it myself to fit my needs.

pinney, 2017-10-14:
@Adirelle 
Please don't take this the wrong way ... I would only like 'issues' to be more specific.
For example the ACL 'issue' is covered in another issue...and the Ath9k drivers as well.
Personally I see no good reason to run a NAS on a 150 USB wireless ...
Or use ACL on a personal cloud ... but hey it's a highly configurable device ... 

I encourage others to fork the PCx firmware and bootloader and make 'improvements'
Even to the point of offering the binaries to umm ... non coders ??
The bootloader does have a built-in delay to allow EZ USB upgrades.
I don't see this as a problem since this device should be powered on continuously in most cases.

I think currently only fat works in u-boot ... 

FPU emulator ... libreCMC / LEDE /OpenWrt doesn't enable this by default.
However it's required for Debian ...

pinney, 2017-10-14:
If one intends to play around with the bootloader or it's environment I would suggest making a backup copy.

I would put a USB Stick in the device ... mount it on /mnt and : 

'dd if=/dev/mtd0 of=/mnt/PC1.mtd0'
'dd if=/dev/mtd1 of=/mnt/PC1.mtd1'
'dd if=/dev/mtd2 of=/mnt/PC1.mtd2'

I normally do not save the other partitions as they are changed by firmware updates (easily replaced)

cat proc/mtd
dev:    size   erasesize  name
mtd0: 00030000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00010000 00010000 "factory"
mtd3: 01fb0000 00010000 "firmware"
mtd4: 0019dcc2 00010000 "kernel"
mtd5: 01e1233e 00010000 "rootfs"
mtd6: 01c80000 00010000 "rootfs_data"

pinney, 2017-10-14:
Clock skew is also covered in another 'issue'
https://github.com/gnubee-git/GnuBee_Docs/issues/39

Adirelle, 2017-10-14:
Thanks for information.

closed 2017-10-14 by Adirelle

Adirelle, 2017-10-18:
> Personally I see no good reason to [...] use ACL on a personal cloud ... 

Security by the principle of least privilege: restrict each service on the NAS to access only the files it is supposed to have access to. This is the very same reason you would like to install a firewall on such a device: reducing the risks of intrusion and containing the damages when it happens.