gnubee-git/GnuBee_Docs #39: Date way in the future after boot

bhkolts, 2017-10-13:
Hello,
After boot date returns a date in September 2018. This really messes up systemd timers as none of them will trigger until September 2018.
I used this firmware  https://github.com/gnubee-git/gnubee-git.github.io/raw/master/debian/openwrt-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-03-01.bin and installed Debian with the Jessie install script. I upgraded to Stretch. All seems well except the time skew, which will correct itself after 2 to 3 minutes.
More info:
I have installed smartmontools and set it to email me the health of each of the six hard drives, so 6 emails.. The time stamps of those emails is interesting. Three of them are dated  November 3, 2016 and three are dated September, 21, 2018. So it appears the date-time go through changes during boot. I also installed ntp so the correct date and time should show up as soon as the ntp daemon starts, which shouldn't take 2 to 3 minutes, I believe. 
Should I have flashed a different firmware?
I'm happy to provide any debugging information you need, just tell me what and how to get it.

Thanks in advance for your help,
Barry

pinney, 2017-10-14:
Hi Barry 

This device does not have a RTC as such it depends on a network connection for time/date.

Time skew ... Please try latest libreCMC firmware.
https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/libreCMC.

Using the shipped bootloader the device tree source needs this to set the clock : 

&sysclock {
                      compatible = "fixed-clock";
                      clock-frequency = <90000000>;
};

An alternative to this is to change the bootloader and set the clock @ 880
https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/uBoot

bhkolts, 2017-10-14:
Hi Larry,
Thanks for your response. So I used the "full" version of https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/libreCMC. It dropped into BusyBox and did not load Debian. However issuing the "date" command in BusyBox returned the correct date and time. The only error I saw as it was booting was "Failed to connect to ubus". So we can try to debug this version or I can try the "lite" version of the firmware. I will need your help  debugging since I'm breaking new ground here.

Barry

bhkolts, 2017-10-14:
Sorry about that hit wrong button

pinney, 2017-10-14:
Hi Barry 

Yes ... that firmware will only run libreCMC
Douglas is the 'go-to' man for  Debian / OMV .. I handle libreCMC/LEDE
So new firmware for loading Debian needs built using the new config/dts.

If you would ... please comfrm this fixes the clock skew.
Also please report any other findings :)

bhkolts, 2017-10-14:
Hi Larry,

So I left my GB-PC1 off over night, so we are starting with a cold boot this morning. Sill using the "full" version it booted into BusyBox and I immediately ran the "date" command. It returned the correct date and time for UTC. So I would assume that if it had loaded Debian the time would be correct. So yes this fixes the time skew.

Would you like me to test the "lite" version?

Barry

bhkolts, 2017-10-14:
Larry,

Just tried the "lite" version. Issuing "date" as soon as it drops into BusyBox gives the wrong data  Sat Sep 30 02:52:18 UTC 2017, but after a minute it comes up with the correct date and time.

Barry

bhkolts, 2017-10-17:
Tried to build firmware on my own, but I'm to new at building firmware so failed.
Re-flashed with openwrt-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-03-01.bin and will live with the time skew for now. I'm still happy to help out with what I can.

Closing this for now.
Barry

closed 2017-10-17 by bhkolts

pinney, 2017-10-17:
Please see : 
https://github.com/gnubee-git/GnuBee_Docs/issues/37