gnubee-git/GnuBee_Docs #44: update images, due to critical remote vulnerabilities (dnsmasq, KRACK, openvpn).

rdslw, 2017-10-31:
Could you please update firmwares available at https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/ ? Especially LEDE and LibreCMC. In the last 4 weeks, critical vulnerabilities were revelead and both LEDE and libreCMC released patched versions.

rdslw, 2017-11-06:
Thank you @pinney 
Do I assume correctly that https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/GnuBee-LEDE-17.01 is the firmware built with this patch?

also what about "LEDE-17.01" and "LEDE-trunk" directories containig one month old files? If not needed, maybe it would be good to remove them, as they might be confusing for new users.

pinney, 2017-11-06:
@rdslw 
Yes https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/GnuBee-LEDE-17.01 has the patches installed.
As does https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC1/firmware/GnuBee-libreCMC
##
Remove "LEDE-trunk" ... on the 'to-do list' as it has been merged upstream.
However "LEDE-17.01" still needs to be merged upstream.

rdslw, 2017-11-06:
Got it. THey contained built firmware blobs, so I wrongly assumed they were mostly for firmware availability.

Closing this issue now, as firmwares are available.  Thank you once more.

closed 2017-11-06 by rdslw