gnubee-git/GnuBee_Docs #46: Updated Debian images?

nariox, 2017-11-10:
Hello,
I've received an e-mail about the new firmware images, but they seem to include only LibreCMC and Lede. Any plans to update the "Debian" image (I'm aware that the debian image is simply a LibreCMC image that mounts the debian partition overlayed on root)?
Thanks

dgazineu, 2017-11-11:
Updates for OpenWrt and LibreCMC-based firmwares with Debian init scripts
are in the works. We use the same chroot/pivot_root that "stage 1" boot
from ramdisk to the root partition as desktop distribution use to employ,
only that "stage 1" is the firmware itself, doubling usefulness as recovery
environment or plain software base ( as OpenWrt/LibreCMC can be used on
their own).
As we proceed to (updating) them, we will test with the Stretch version of
Debian together, and update the installation procedures.
The resulting image can be put to download, as with Jessie earlier, if we
find it stable enough.
Some have reported successful upgrades already, but as ready-made images
go, we need to test first, as those who will use them will vary on
knowledge to solve problems by themselves later. It's not a total
guarantee, but it's the process we have to go through.

Also are being considered the feedback already received from everyone and
the upcoming GB-PC2 to take advantage of this work.

Rgds,

Adirelle, 2017-11-12:
Is there a git repository for building these "debian" images somewhere ?

dgazineu, 2017-11-12:
They're the same, and will be updated when ready.

dgazineu, 2017-11-20:
New firmware available @
https://github.com/gnubee-git/gnubee-git.github.io/raw/master/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-18.bin,
with Debian/OMV support.
    Install instructions are the same as before, updated on
http://gnubee.org/install_debian.html as well.

    The updated sources are @
https://gogs.librecmc.org/dgazineu/GnuBee-libreCMC

Please give a try and report back.

bhkolts, 2017-11-21:
Douglas,
Just tried the updated firmware. Works fine with the Debian Stretch I had installed on a ssd.

Thanks for your effort.

Barry

nariox, 2017-11-22:
So far so good here as well.
I had already upgraded to stretch as well. Hopefully no more panics (:

Pedro Nariyoshi

DurtalDeepDown, 2017-11-22:
@bhkolts @nariox 
Do you use an existing Debian or did you install a new one from the updated firmware? As I reported #37, installing Debian from the new firmware takes ages - at least for me.

dgazineu, 2017-11-22:
I do use existing Debian, but a debootstrap for jessie (as done with
provided script), takes about an hour here.

Also, please note that the process for debootstrap into stretch fails
(altough an upgrade can be done later). Not sure it debootstrap package
need any changes.

DurtalDeepDown, 2017-11-22:
Well, I also use the provided script. If you succeed in installing Jessie from the updated libreCMC I really have to wonder what I am doing wrong. I updated the firmware and immediately downloaded and executed the script. I do not see any opportunity to screw things up. 
After my first attempt I tried to update debootstrap but it was already uptodate. May the slowness be due to the missing keyring? [Here ](https://github.com/dnschneid/crouton/issues/2988)they say, this is not important, but it seems to be the only difference to the installation process done from the old libreCMC which worked without problems.

dgazineu, 2017-11-22:
I had no problems at all, also got the keyring complain here.

 - What steps did take the most time?
 - Can you open a ssh session to see if we can get a "top" output to narrow
down the slowness, while installing?

Rgds,

DurtalDeepDown, 2017-11-22:
Debootstrap is still Retrieving and Validating, right now at libncurses5. There were no others steps yet.
top returns:
```
Mem: 114296K used, 397720K free, 792K shrd, 7100K buff, 83620K cached
CPU:  0.3% usr  0.2% sys  0.0% nic 99.4% idle  0.0% io  0.0% irq  0.0% sirq
Load average: 0.00 0.00 0.00 3/96 6414
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
 6390  6363 root     R     1212  0.2   3  0.0 top
 6356   749 root     S     1128  0.2   1  0.0 /usr/sbin/dropbear -F -P /var/run
 5899  8720 root     S     3460  0.6   3  0.0 wget -q -O /tmp/newroot//var/cach
  911     1 root     S     2472  0.4   1  0.0 /usr/sbin/nmbd -F
  910     1 root     S     2388  0.4   3  0.0 /usr/sbin/smbd -F
  686     1 root     S     1704  0.3   2  0.0 /sbin/netifd
    1     0 root     S     1540  0.3   2  0.0 /sbin/procd
  867     1 root     S     1532  0.3   1  0.0 /usr/sbin/uhttpd -f -h /www -r li
  627     1 root     S     1520  0.3   2  0.0 /sbin/rpcd
  704     1 root     S     1420  0.2   3  0.0 /usr/sbin/odhcpd
 8720  8718 root     S     1360  0.2   2  0.0 {debootstrap} /bin/sh /usr/sbin/d
 8718  7068 root     S     1360  0.2   3  0.0 {debootstrap} /bin/sh /usr/sbin/d
 7068  6959 root     S     1352  0.2   3  0.0 {debootstrap} /bin/sh /usr/sbin/d
  618     1 root     S     1228  0.2   1  0.0 /sbin/logd -S 64
 6959  6932 root     S     1204  0.2   2  0.0 sh debian-jessie-install
  950     1 root     S <   1200  0.2   3  0.0 /usr/sbin/ntpd -n -N -S /usr/sbin
 6932  6930 root     S     1200  0.2   3  0.0 -ash
  826   686 root     S     1196  0.2   3  0.0 udhcpc -p /var/run/udhcpc-eth0.2.
 6363  6356 root     S     1196  0.2   1  0.0 -ash
  532     1 root     S     1188  0.2   3  0.0 /sbin/ubusd
```

dgazineu, 2017-11-22:
As usage seems low on the device, we can check the connection debootstrap
makes.
The process selects a mirror from "http://httpredir.debian.org/debian"
(Defined @ debian-jessie-install), redirected here to a CDN, that may be
near you or not (http://cdn-fastly.deb.debian.org/debian/).
You can also choose a mirror, listed here:
https://www.debian.org/mirror/list , to try out another source .

DurtalDeepDown, 2017-11-22:
No, it has nothing to do with the mirror. I guess it is a problem of outdated packages. I encountered the same strange behavior of debootstrap on an old Trisquel-system. Here, a systemupdate fixed it.

bhkolts, 2017-11-22:
@DurtalDeepDown ,
I had installed Debian with the original firmware per instructions and then updated to Stretch. I don't remember installing Debian taking an unusually long time.

Barry

nariox, 2017-11-22:
Same thing here, I've installed Jessie and then upgraded to Stretch (change
the repos to point to Stretch, it's been a while since I've last used
Debian, but I think this was one way to do it). I wanted to use php7, so I
needed Stretch.

Boy, I wish the netatalk version Debian had was somewhat recent and not the
4 year old stable version...

Pedro Nariyoshi

DurtalDeepDown, 2017-11-22:
Just for curiosity: How does the updated firmware influence the already installed debian-system? Isn’t everything done from the HDD/SSD once debian is installed to it?
Not that I have accomplished this yet...

dgazineu, 2017-11-22:
The new firmware is a new kernel version. You will need to get the
additional modules copied/installed (that will get on its own folder @
/lib/modules in the Debian root) and run "depmod -a"  after the copy. For
the versions I am testing, those are being uploaded already, and the
procedure described in http://gnubee.org/install_debian.html takes care of
this.

If you don't erase your modules folder from the other version, the kernel
downgrade is granted to work too.

DurtalDeepDown, 2017-11-24:
Since I solved my problem with the openWRT firmware (#37) you may close this issue for now. Sorry for hijacking.

nariox, 2017-11-25:
The clock skew problem seems to be gone. Yay! No kernel panics so far.

I do have one concern, I have the impression hdparm used to report higher
speed numbers on my drives. I have an SSD and an HDD, but now both report
around 75MB/sec, whereas the SSD should be much higher (the HDD should be
about 50% higher if I recall correctly). Should I report this as a new
issue?

Pedro Nariyoshi

Adirelle, 2017-11-25:
I have just tried the new firmware with Debian stretch and it is better than the previous one. 

Thanks for the added modules, ufw and acl are now working. 

I still have a noticeable clock skew but ntpd seems able to cope with it. 

minidlnad is still dying with illegal instruction, though I was not expecting this to be fixed. The debian maintainer could not reproduce the issue -- he supposes an hardware bug -- and I am stuck with this now. I will probably use NFS to expose the files from another server.

dgazineu, 2017-11-29:
Updated firmware:
https://github.com/gnubee-git/gnubee-git.github.io/raw/master/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-28.bin
Sha256sum: e2fe4cd4f70dc04372d5a7dcf1443d6895cd3e797a65c1d9ec889d79db181493

Changes: Fixed clock skew

Installation instructions @ http://gnubee.org/install_debian.html
Sources are available @ https://gogs.librecmc.org/dgazineu/GnuBee-libreCMC

closed 2017-11-29 by dgazineu