gnubee-git/GnuBee_Docs #47: enabling MIPS fpu emulator in LEDE firmware

rdslw, 2017-11-16:
Would it be possible to enable fpu emulator in firmware for gnubee?
It's described here: http://shmichen.blogspot.com/2017/08/how-to-enable-fpu-emulator-on-openwrt.html

It will probably slightly enlarge firmware size, but it will allow to use much more software on gnubee - specifically golang built binaries, nodejs and many others.

dgazineu, 2017-11-16:
> 2017/08/how-to-enable-fpu-emulator-on-openwrt.html

Hi,
It is already enabled on the compiled firmwares.

rdslw, 2017-11-16:
Hmmm, when I try to check doing "ls /sys/kernel/debug/mips/"
I see there only two files "unaligned_action        unaligned_instructions" and no sign of fpuemustats

I also have problems running golang binaries.

What is a proper way to check if fpu emulation works?

dgazineu, 2017-11-16:
What firmware file are you using? At least on
openwrt-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-03-01.bin it is
enabled for sure, as it is needed for Debian.

rdslw, 2017-11-16:
I'm using https://github.com/gnubee-git/GnuBee_Docs/blob/master/GB-PC1/firmware/GnuBee-LEDE-17.01/lede-ramips-mt7621-gb-pc1-squashfs-sysupgrade.bin as I would like to have newest patches due to many vulnerabilities in last months.

So that was my question to check if emulation is enabled in LEDE firmware generated in the link above. As I understand (correct me if I'm wrong) this is the place with newest lede based firmware for gnubee.

dgazineu, 2017-11-16:
> PC1/firmware/GnuBee-LEDE-17.01/lede-ramips-mt7621-gb-pc1-
> squashfs-sysupgrade.bin as I would like to have newest patches due to

There are firmware "lines", as Lede is indeed newer than openwrt, but the
hardware support on openwrt is better (it is the official version of kernel
from SoC vendor).
That version of Lede can have the same feature enabled, though. I'll check
it out.

rdslw, 2017-11-23:
@dgazineu any progress on it?
Can i help somehow?

dgazineu, 2017-11-23:
I have just finished an update with LibreCMC this week ( see
http://gnubee.org), and started to update LEDE. LEDE's kernel has advanced
to 4.9, and is giving me some small troubles upon Debian boot.

rdslw, 2017-11-23:
Sounds great. Waiting for the LEDE build. Id happily test if afterwards! 
I appreciate your work on this. Thank you kindly!

rdslw, 2017-12-02:
@dgazineu can I help somehow on this?
Probably first step for me would be having a LEDE build environment. Can you point me to some readme/doc how to do it?
Also if you have any other suggestions how I may help, let me know. I have gnubee1 also to test builds.

pinney, 2017-12-02:
@rdslw : 

https://wiki.openwrt.org/doc/howto/buildroot.exigence
https://wiki.openwrt.org/doc/howto/build

Just git clone the proper repo :)
[image]

dgazineu, 2017-12-02:
@rdslw My focus right now is to fine tune LibreCMC for GB-PC2 (just funded, yeah!)
If you want to fiddle with LibreCMC (recent 4.4 kernel with several bugs already worked out), I advice you to clone from https://gogs.librecmc.org/dgazineu/GnuBee-libreCMC and get started from here.

Most recent "plain" LEDE also works, but the "Debian-compatible" firmware on 4.4 kernel had some bugs and freezes on some software and on the last 4.9 the kernel got too big, rendering boot problems (also on "Debian-compatible" version).

rdslw, 2017-12-18:
I hit the wall with my tries. I'm using official lede-17.01 branch + some patches handpicked from GnuBee-lede and GnuBee-libreCMC. lede-17.01 is quite recent, so only 5 or so patches needed to by applied.

I based my .config on GB-PC1_defconfig plus my own changes. 

Nevertheless gnubee did not boot with my firmware. 

I tried then to match my .config in critical sections (kernel etc) really close to the GB-PC1_defconfig (or GB-PC1_debian_config), while there are many changes in packages (but only for more). No luck. I tried a dozen different configurations but didnt have luck here.

What may I be doing wrong?
Should I post my .config plus patch I'm applying to lede-17.01 here to check it?

Cabalist, 2018-01-24:
@rdslw  I'd be interested in reviewing what you've got if you are still trying to get that going.    I found the current state of multiple firmwares very confusing when starting out and we are working on making some more straight forward documentation.  Part of that would be configuring and building current LEDE/LibreCMC.

smurfix, 2018-02-05:
@dgazineu Any progress with enabling the FPU? It's blocking Debian installation.

rdslw, 2018-02-25:
@Cabalist Sorry for the delay. Yeah, I have managed to have current LEDE on gnubee and have build environment enabling me to pull updates from official lede repo and make my own LEDE builds. It is however  highly customsize configuration, so I have to do three things:
* create repo 
* split my patchset into 'just minimal to get official gnubee on lede' (a) and rdslw part (b)
* document this in a short file or doc
Let me know if you're still interested in this. I have a lot of things on me, so it may take a while.

Cabalist, 2018-02-26:
@rdslw yes!  I would love to have a look.  We are up and running over here but getting a good upgrade path going would be really appreciated.  👍

rdslw, 2018-06-17:
ok, so I finally did a complete up-to-date openwrt-17.01 firmware build.

you can get it at: https://github.com/rdslw/openwrt/tree/beewrt-17.01  (so it's beewrt-17.01 branch)
and then build as a normal openwrt build.

Main features:
* MIPS FPU emulation enabled
* mmc/sd works on gnubee
* proper wan/lan (eth0.2 and eth0.1) switch configuration
* reboot works properly
* newest (stable) kernel 4.4.138
* all lede-17.01 (AKA openwrt-17.01) commits up to 17 June 2018
* cherrypicked (from different repos) most gnubee important fixes

Future planned work:
* my own BEEWRT-17.01_config files in standard and full builds 
* compiled firmware binaries for download.
* I'm also considering (if anyone is interested??) extending standard openwrt preinit phase to recognize GNUBEE-ROOT and pivot to boot from it
* openwrt-18.06 build, the moment openwrt releases it

@Cabalist let me know if you have any questions.

smurfix, 2018-06-18:
Nice work, though frankly I was hoping for a newer kernel.

Support for GNUBEE_ROOT would be great.

Cabalist, 2018-06-18:
Fantastic!  I’ll check this out this week. 

  And yes GNUBEE-ROOT support would be greatly appreciated.

technik007cz, 2018-08-06:
Hi there, I tried it and it works, well done. 
These options could be added:
zram-swap (in Base system)
tmp on zram (unfortunately could not find location again)
and consider, because low system RAM, to turn on options for lxc virtualization like cgroups

rdslw, 2018-08-07:
Thanks.
I tried zram options in past, but currently they're hardcoded to allocate on mount only flashsize x2 for /tmp - which for example do not allow to do sysupgrade from image on /tmp (it's too small). 
Hence I decided to not use zram :(

technik007cz, 2018-08-07:
Ok, /tmp on zram could be left turned off. 
However swap on compressed ram device would be useful when this device is equipped with only 512MB RAM.

rdslw, 2018-08-25:
I had a problems with openwrt 17.x builds: while zram was enabled (kmod), no matter what /tmp on zram option was, openwrt always put tmp on zram. This plus small tmp space, ruled out zram completly. 
I did not have time to test it on 18 though.