gnubee-git/GnuBee_Docs #49: System clock runs too fast

daniellandau, 2017-11-26:
The clock on my Gnubee 1 is running too fast. It gets ahead ~330 seconds in 4 hours.

```sh
# uname -a
Linux gnubee-n1.gnubee 4.4.87-gnu #0 SMP Fri Nov 3 14:47:34 2017 mips GNU/Linux
# cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 8 (jessie)"
NAME="Debian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
```

Otherwise the machine is performing well so far and I'm quite happy with it.

neheb, 2017-11-26:
Either flash the bootloader to an 880Mhz version or flash new firmware

Adirelle, 2017-11-28:
Same there, using Debian Stretch 9 and latest firmware (https://github.com/gnubee-git/gnubee-git.github.io/blob/master/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-18.bin)

ntpd has to fix the time of 4 to 7 seconds every 5 minutes, which is a skew of 166 ms/s.

```
$ fgrep -C1 "Time has been changed" /var/log/syslog
Nov 28 08:09:39 exodus ntpd[1546]: Soliciting pool server 94.154.96.7
Nov 28 08:09:32 exodus systemd[864]: Time has been changed
Nov 28 08:09:32 exodus systemd[1]: Time has been changed
Nov 28 08:09:32 exodus ntpd[2849]: receive: Unexpected origin timestamp 0xddc78a33.8be21ecb does not match aorg 0000000000.00000000 from server@217.26.163.51 xmt 0xddc78a2c.c10ec338
--
Nov 28 08:14:34 exodus ntpd[1546]: Soliciting pool server 129.250.35.250
Nov 28 08:14:28 exodus systemd[864]: Time has been changed
Nov 28 08:14:28 exodus systemd[1]: Time has been changed
Nov 28 08:14:28 exodus systemd[1]: apt-daily-upgrade.timer: Adding 4min 9.997413s random time.
--
Nov 28 08:19:34 exodus ntpd[1546]: Soliciting pool server 195.154.41.195
Nov 28 08:19:27 exodus systemd[1]: Time has been changed
Nov 28 08:19:27 exodus systemd[864]: Time has been changed
Nov 28 08:19:27 exodus ntpd[2849]: receive: Unexpected origin timestamp 0xddc78c86.3051ce10 does not match aorg 0000000000.00000000 from server@178.32.216.71 xmt 0xddc78c7f.6f848620
```

dgazineu, 2017-11-29:
Updated LibreCMC firmware with clock skew fix @ http://gnubee.org/install_debian.html  (https://github.com/gnubee-git/gnubee-git.github.io/blob/master/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-28.bin)
If the problem does not get fixed with the new firmware, update the bootloader to [GB_PCx_uboot_57600-880MHz-1200RAM.bin](https://github.com/gnubee-git/GnuBee_Docs/raw/master/GB-PC1/firmware/uBoot/57600/GB_PCx_uboot_57600-880MHz-1200RAM.bin)

Please test and report back. Rgds.

closed 2017-11-29 by dgazineu

Adirelle, 2017-11-29:
The latest firmware does no good. Where can I find the instruction to flash the bootloader ? (And is there a way to check for the actual CPU/ram frequencies from Linux ?)

pinney, 2017-11-30:
@Adirelle 

If you have a serial console and tftp you can use : 

 No USB Storage found. Upgrade FW failed!

Please choose the operation: 
   1: Load system code to SDRAM via TFTP.
   2: Load system code then write to Flash via TFTP.
   3: Boot system code via Flash (default).
   4: Enter boot command line interface.
   5: Load system code then write to Flash via USB Storage.
   6: Load system code then write to Flash via Httpd.
   9: Load U-Boot code then write to Flash via TFTP.                                                                 

You chose 9
                                                                                                                       
9: System Load Boot Loader then write to Flash via TFTP. 
 Warning!! Erase Boot Loader in Flash then burn new one. Are you sure? (Y/N)
 Please Input new ones /or Ctrl-C to discard
        Input device IP (192.168.1.123) ==:192.168.1.123
        Input server IP (192.168.1.2) ==:192.168.1.2
        Input Uboot filename (uboot.bin) ==:GB_PCx_uboot_57600-880MHz-1200RAM.bin

If you want to use the web interface you can use : 

Please choose the operation: 
   1: Load system code to SDRAM via TFTP.
   2: Load system code then write to Flash via TFTP.
   3: Boot system code via Flash (default).
   4: Enter boot command line interface.
   5: Load system code then write to Flash via USB Storage.
   6: Load system code then write to Flash via Httpd.
   9: Load U-Boot code then write to Flash via TFTP.

You chose 6
                                                                                                                       0 

web load
Trying eth2

 ETH_STATE_ACTIVE!! 
Httpd Server Ip: 192.168.1.1
HTTPD READY!

Adirelle, 2017-11-30:
The upgrade went well with a TFTP server; I've got no expensive brick so far. And it seems it fixes the clock skew: NTP reports a quite stable time offset, which is under 50ms.

bhkolts, 2017-11-30:
I updated the bootloader through the web interface and all went well. I also can report less than 50 ms time offset with ntp.
So as Adirelle did, I upgraded both the firmware and the bootloader.
Thanks for all the work on this.
Barry

dgazineu, 2017-11-30:
Thanks for the feedback.

daniellandau, 2017-12-17:
Updating U-Boot through the web interface worked for me too. It took some figuring out, so I'll write up here the approximate steps:

1. Connect the blue ethernet to a switch
2. Connect computer to same switch
3. Select 6 in early boot from serial console
4. Set IP address for ethernet manually to e.g. `192.168.1.3` on computer (at least I had to do that as it wouldn't get an address automatically)
5. Open web browser to http://192.168.1.1/ and  find the U-Boot section and follow prompts

ohrensessel, 2017-12-30:
Just a short question: How can I update the bootloader, when I didn't buy the USB-to-UART adapter?

pinney, 2017-12-30:
@ohrensessel 
Using the shipped bootloader there is not a known way to trigger the webinterface.

However there is one in the works : 
https://github.com/gnubee-git/GnuBee_PCx_uboot/tree/WebLoadByButton

There are 3 header pins near the 3.5mm UART Jack allowing one to use a common USB to UART adapter.
Known working USB to UART chips are the cp2102, pl2303, ch340g, and ft232r.

ohrensessel, 2017-12-30:
Thank you for the quick reply. Updating the bootloader was successful.