gnubee-git/GnuBee_Docs #53: Install and boot Debian from /dev/md0?

kmart, 2018-01-03:
Hi,

On my GnuBee 1 I updated the firmware according to the  [Install firmware](https://github.com/gnubee-git/GnuBee_Docs/wiki/Install-firmware) guide, using the "newer image". Next I followed the [Install Debian](http://gnubee.org/install_debian.html) instructions, but then installing to `/dev/md0` (raid) device, which I had created in advance.

But when rebooting, after a successful run of the `debian-jessie-install` script, I unfortunately end up back in the firmware OS (libreCMC ver. 4.4.87-gnu) and not in Debian.

The `/dev/md0` device is still there after a reboot, and can be mounted (and do contains the installed Debian OS).

Is there an additional step that has to be done in order to boot from the `/dev/md0` device?

Thanks in advance!

br,
kmm

kmart, 2018-01-03:
Hi,

According to the [GnuBee PC 1 HW/SW](http://gnubee.org/background_info.html) page, `u-boot` should look for a partition lablled "GNUBEE-ROOT" and boot from that partition if found. When checking, the labelling looks OK:

```
root@libreCMC:(unreachable)/root# blkid /dev/md0
/dev/md0: LABEL="GNUBEE-ROOT" UUID="667633c7-2de5-4136-a1e5-9bbfdc13f31b" TYPE="ext4"
```

But if you list the labels for all the partitions, we get:

```
root@libreCMC:(unreachable)/root# lsblk -o name,label
NAME      LABEL
sda       
├─sda1    libreCMC:0
│ └─md0   GNUBEE-ROOT
└─sda2    libreCMC:1
  └─md1   
sdb       
├─sdb1    libreCMC:0
│ └─md0   GNUBEE-ROOT
└─sdb2    libreCMC:1
  └─md1   
sdc       
├─sdc1    libreCMC:0
│ └─md0   GNUBEE-ROOT
└─sdc2    libreCMC:1
  └─md1   
sdd       
├─sdd1    libreCMC:0
│ └─md0   GNUBEE-ROOT
└─sdd2    libreCMC:1
  └─md1   
sde       
├─sde1    libreCMC:0
│ └─md0   GNUBEE-ROOT
└─sde2    libreCMC:1
  └─md1   
sdf       
├─sdf1    libreCMC:0
│ └─md0   GNUBEE-ROOT
└─sdf2    libreCMC:1
  └─md1   
mtdblock0 
mtdblock1 
mtdblock2 
mtdblock3 
mtdblock4 
mtdblock5 
mtdblock6 
```

Mabe `u-boot` is not able to sort this out? Maybe putting `/boot` on a non-raid partition by itself could help?

noblepepper, 2018-01-04:
I ran into this too but haven't had time to sort it out yet. The libreCMC image needs to be modified to fix it.

When libreCMC finds the GNUBEE-ROOT partition the raid device isn't assembled yet and it tries to mount one of the devices that is part of the raid and chokes. This happens at about the [14.xxxxxx] count in the boot process.

I have switched to a small SSD with Debian on it and a RAID1 running on a couple of larger spinning disks and it works fine. You should be able to have a small partition on one drive for Debian and the rest of it as part of a raid. Since raid drives need to be partitioned the same you'll have a small partition the other drives that isn't used in the raid. OMV is easier to set up if you let it do the partitioning and raid work. It likes the shared data on a separate device from the system too.

Debian with OMV installed is using 2.3G in my setup so a 5G or 10G partition should be plenty.

Hopefully when this gets sorted out you can convert the Debian partition into a separate raid device.

Adirelle, 2018-01-04:
What are the metadata version of your /dev/md0 ? IIRC the firmware looks for GNUBEE-ROOT *after* having the kernel has tried to assemble MD arrays. However, Linux can only automatically assemble MD arrays with metadata version <1.0 (because the other requires userland tools).

kmart, 2018-01-04:
@noblepepper, @Adirelle: Thanks for helping out!

Followed the advice with installing Debian on its own partition, and now the GnuBee is at least up and running! Thanks!

But this "metadata version" thing looks interesting... Checked, and the metadata version is actually set to 1.2. As the version apparently can be set with the `--metadata` option, I think I will give it a try (and go back to the "install on a separate partition" if it don't work out).

Btw., when trying to log into the box after Debian install and reboot, using `ssh` and with the network cable attached to the blue RJ-45 connector, `ssh` just hanged. Moved the cable to the other connector and then I could log in. Noticed then that I could not access servers outside my local network from the box. Found eventually out that the reason for this was that the address 192.168.1.1 was assigned to `eth0.1` in `/etc/network/interfaces` - which incidentally is also the address of my router. Basically you then got incorrect route back when trying to log into the box with the network cable connected to the blue RJ45 connector. Fixed the problem with changing the address in  `/etc/network/interfaces` to an address outside of the 192.168.1.0/24 subnet. I guess that others that uses the same subnet will run into the same issue? Should probably be noted somewhere?

noblepepper, 2018-01-05:
I found this with a good amount of googling and @Adirelle has the answer. 

I also set the partition type to fd - linux raid autodetect (00fd for GPT I believe) before assembling the raid. All looks to be working now.

Notes:
1. You can't change from 1.2 to 0.90 so you lose anything on the raid you made with 1.2 metadata.

2. The linux raid wiki says this is not recommended, gives a horror story example of why. They recommend using initramfs instead. While we don't have initramfs, libreCMC is performing it's purpose and it should be possible to use 1.2 metadata with some work. I'll put this on my todo list but it is probably going to land on the part of the list that I never get around to doing until something forces me to.

3. I vote to close this issue unless @dgazineu wants to volunteer to take on 1.2 metadata

kmart, 2018-01-05:
OK, to close this issue for my part.

Btw., as I understood it from googling, setting the metadata version < 1.0 could mainly cause problems if you had big disks/many partitions. As I have neither, I just went and recreated the arrays with version set to 0.9. And this time booting from the `/dev/md0` partition worked!

dgazineu, 2018-01-05:
Please keep it open for a while, I will research on the 1.2 metadata
requirements

noblepepper, 2018-01-06:
@kmart - This comment in the [raid wiki](https://raid.wiki.kernel.org/index.php/Linux_Raid) is what I see as the biggest danger 
> imagine moving part of an old array onto another machine before wiping and repurposing it: reboot and watch in horror as the piece of dead array gets assembled as part of the running RAID array, ruining it

Moral of the story - always wipe a disk that was part of a raid when you remove it

dgazineu, 2018-01-06:
> Moral of the story - always wipe a disk that was part of a raid when you remove it
Agreed

noblepepper, 2018-01-27:
Until 1.2 metadata can be used you can find more detail on using a raid boot device in the [wiki](https://github.com/gnubee-git/GnuBee_Docs/wiki/RaidKernelAutoAssembly)

smurfix, 2018-02-05:
Metadata is not the problem, 0.9 is ugly as hell and does not survive changing the partition size.

The problem is that the GnuBee hook which switches to GNUBEE_ROOT runs much too early and bypasses all the nice setup that LibreCMC / LEDE does.
I have instead used an ugly hack and replaced `/sbin/procd` with this script:

```
# mv /sbin/procd /sbin/procd.real
# cat >/sbin/procd <<'EOF'
#!/bin/sh

echo "************* try to load Debian ***********************"

set -x

gnubee_pivot_root(){
  echo "Partition GNUBEE-ROOT found. Starting..." > /dev/kmsg
  cd /mnt/root
  mkdir -p rom
  pivot_root . rom
  exec chroot . /sbin/init
}

continue_boot(){
  exec /sbin/procd.real "$@"
}

gnubee_boot(){
   if test -f /etc/no-external-root ; then
     echo "/etc/no-external-root found - continue booting"
     continue_boot
   fi
   touch /etc/no-external-root
   
   #echo "/sbin/mdev" > /proc/sys/kernel/hotplug
   #mdev -s

   echo -n "Wait for GNUBEE-ROOT "
   
   # MT7621 GnuBee
   for x in 1 2 3 4 5 6 7 8 9 0 ; do
     if findfs LABEL=GNUBEE-ROOT >/dev/null 2>&1 ; then
       mount -o ro $(findfs LABEL=GNUBEE-ROOT) /mnt/root
       gnubee_pivot_root
     fi
     sleep 1
     echo -n '.'
   done
   echo "Partition GNUBEE-ROOT not found. Resuming LibreCMC boot."
   continue_boot

}

gnubee_boot
EOF
# chmod +x /sbin/procd
```

Then in Debian I will remove /rom/etc/no-external-root so that I can get back to firmware if anything ugly should happen.

smurfix, 2018-02-05:
NB: I'd hook it into the preinit script infrastructure, but frankly that's a lot of very interesting shell scriptery which I don't really have time to understand. Maybe somebody else could do that ..?

Adirelle, 2018-02-05:
> Metadata is not the problem, 0.9 is ugly as hell and does not survive changing the partition size.

It is part of the problem: the kernel cannot automagically assemble arrays with metadata superior to 1.0 (  I think it works with 1.0, but I have not tested it). Doing so requires userland tools that are not available at the time the original GnuBEE script looks for the GNUBEE-ROOT partition.

So there are basically two solutions:
1. Yours: postpone detection until all arrays are assembled by the LibreCMC/LEDE firmware.
2. Mine: use arrays that can be assembled by the kernel alone, e.g 1.0 or inferior.

neilbrown, 2018-05-13:
If you use the firmware described at https://groups.google.com/forum/#!topic/gnubee/VRNhAw1csnw
it will assemble v1.2 arrays and look for GNUBEE_ROOT on them.  If it doesn't work for you, let me know details and I'll make it work.