gnubee-git/GnuBee_Docs #54: Network failure

Adirelle, 2018-01-04:
I have a strange behavior, that I cannot easily reproduce: basically, all the network sockets suddenly stops to work.

I can still connect to the GnuBEE using the serial console and ping from or to it, but I cannot use any network services (SSH, NFS, mysql, transmission, ...) and it cannot access external services (DNS, ...). nmap still find open ports, but connecting to them times out. And sometimes I cannot even login anymore as root using the serial console. The logs does not give any clue about what is happening.

I already tried the following things to no avail:
* uninstall Samba (see #45),
* restarting the network subsystem (using systemctl restart networking),
* restarting the services,
* unplugging and plugging the network cable.

The only solution is to reboot it.

I am using Debian Stretch 9 and the following kernel (https://github.com/gnubee-git/gnubee-git.github.io/blob/master/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-28.bin IIRC) : 

```
$ cat /proc/version
Linux version 4.4.87-gnu (dgazineu@gnubee.net) (gcc version 5.4.0 (LEDE GCC 5.4.0 r64-b029924) ) #0 SMP Wed Nov 22 13:06:13 2017
```

Any idea ?

Adirelle, 2018-01-04:
Got this today on the console:

```
[ 1124.102030] ------------[ cut here ]------------
[ 1124.111268] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:306 dev_watchdog+0x258/0x2fc()
[ 1124.128111] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[ 1124.142003] Modules linked in: cryptomgr md5 nf_log_ipv6 nf_conntrack_ipv6 nf_defrag_ipv6 ip6table_filter ip6_tables nf_log_ipv4 nf_log_common xt_LOG xt_limit xt_tcpudp nf_conntrack_ipv4 nf_defrag_ipv4 xt_multiport xt_conntrack nf_conntrack iptable_filter dm_raid ip_tables x_tables
[ 1124.191811] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.4.87-gnu #0
[ 1124.204269] Stack : 00000000 00000000 806f6862 00000037 00000000 00000000 80690000 80710000
          8fc60314 8068e1c3 805bf5dc 00000001 00000000 806f367c ffffffff 00000200
          00100000 80064fc8 80690000 80710000 806928f8 806928fc 805c3f18 8fc0de04
          00000003 80062cfc ffffffff 00000200 00100000 00000000 00000006 00c0de04
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[ 1124.275019] Call Trace:
[ 1124.279915] [<80015e20>] show_stack+0x54/0x88
[ 1124.288592] [<80283d90>] dump_stack+0x84/0xbc
[ 1124.297257] [<8002c0e4>] warn_slowpath_common+0xa0/0xd0
[ 1124.307648] [<8002c140>] warn_slowpath_fmt+0x2c/0x38
[ 1124.317539] [<80453944>] dev_watchdog+0x258/0x2fc
[ 1124.326903] [<80074078>] call_timer_fn.isra.4+0x24/0x80
[ 1124.337294] [<800742d0>] run_timer_softirq+0x1fc/0x25c
[ 1124.347515] [<8002ed48>] __do_softirq+0x294/0x2e0
[ 1124.356869] [<8002f030>] irq_exit+0x78/0x94
[ 1124.365193] [<802bc66c>] plat_irq_dispatch+0xb4/0xdc
[ 1124.375062]
[ 1124.378127] ---[ end trace 8e1c5c79e868dbfc ]---
[ 1124.387390] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[ 1124.399735] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[ 1124.411730] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0838a000, max=512, ctx=40, dtx=40, fdx=39, next=40
[ 1124.432366] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0a8e6000, max=512, calc=214, drx=215
```

Adirelle, 2018-01-04:
I remember the firmware was configuring a bridge on eth0. Is this required to avoid this issue ?

neheb, 2018-01-04:
I asked @blogic (main developer?) about this. He said he's seen it too but unable to reproduce. Keep in mind the ethernet driver used in LibreCMC/LEDE/OpenWrt is mainly crappy vendor code. It's kept on life support but that's about it.

There may be plans to backport the upstream driver but I am not sure.

Adirelle, 2018-01-05:
It happened maybe once or twice two weeks ago and then nothing until yesterday. In the last 48h, it happened 4 or 5 times. Each time I have to reboot it using the serial console. I'd be happy to have a mitigation or a workaround.

neheb, 2018-01-05:
The upstream switch and Ethernet driver could be a solution. Unfortunately, that won't come for a while. I'll be trying to get those working in a few days. No promises though.

cat-tech, 2018-01-21:
I was configuring my switch to set a network address on the GnuBee and had a what the ....? moment. 
All my GnuBee MAC addresses are the same! 
From ifconfig print:
eth0      Link encap:Ethernet  HWaddr 78:a3:51:ab:b4:60
eth0.1    Link encap:Ethernet  HWaddr 78:a3:51:ab:b4:60
          inet addr:192.168.3.200  Bcast:192.168.3.255  Mask:255.255.255.0
eth0.2    Link encap:Ethernet  HWaddr 78:a3:51:ab:b4:60
          inet addr:192.168.3.198  Bcast:192.168.3.255  Mask:255.255.255.0

Note the IP addresses are different, although one is set statically. 
Currently there is nothing set in /etc/network/interfaces for the MACs. 
Could this be a default? Anyone suggest how this comes about? I will need to do some digging. Anyone got a good site for a description of how Debian goes about selecting MACs? 

For a fix see https://debian-administration.org/article/501/Working_with_MAC_addresses

Adirelle, 2018-03-29:
@cat-tech Yes, I also noticed that all MAC addresses are the same by default. I suppose it is related to the way switches work (though I am not sure). I have tried to force different MAC addresses on the interfaces but it does not prevent the ethernet stack to fail.

neilbrown, 2018-05-13:
I believe this is fixed in mainline code (drivers/staging/mt7621-eth) and in the 4.4 kernel described at https://groups.google.com/forum/#!topic/gnubee/VRNhAw1csnw
So I'm closing this issue.

nedson, 2019-02-28:
@Adirelle could you share your script for rebooting your gnubee when the network stops working? Are you running this in cron?

Adirelle, 2019-02-28:
I have an echo service running on another host (a raspberry pi, actually) and use the following script to test it : if it can ping the host but cannot have a reply from the echo service after 5 tries, it fails. And then I have something like `/usr/local/sbin/echoTest || systemctl reboot` in my crontab.

```shell
#!/bin/bash
REMOTE="${1:-192.168.0.5}"
PORT="${2:-7}"

ping -Anc1 -W1 $REMOTE >/dev/null || exit 0

for I in 1 2 3 4 5; do
        if echo test | nc -n -w5 -q6 $REMOTE $PORT | grep test >/dev/null; then
                exit 0
        fi
        sleep 2
done

exit 1
```

nedson, 2019-02-28:
What is '-q6'? I don't see that option in my version of netcat.

nedson, 2019-03-01:
Thank you for sharing your script.

nedson, 2019-03-01:
@Adirelle how often is your system rebooting? I am only running NFS on my gnubee.  I setup a samba share, but was not happy with the performance compared with NFS. I have seen a few reports that say samba is working better. As such, I'll live with this bandaid.

Since I setup the script on my system, it rebooted once after being up for 20 hours. I worry that this project may not continue. I hope it continues maturing...I wish I had more time/skill to help.

neheb, 2019-03-01:
I will note that mtk-eth recently gained DSA support for mt7621. With a few DTS changes, mtk-eth from the staging directory can go away.

neilbrown, 2019-03-02:
I've just pushed some code to gnubee/v5.0 in my gitgub (github.com/neilbrown).
It removes my7621-eth from staging and adds DSA support to the DTS.
On the Gnubee-PC2, the extra port doesn't work yet.  I think that'll require some hacking.
The other two are "lanblue" and "lanblack", but that will probably change.
The config I use is in gnubee1_defconfig.

Adirelle, 2019-03-02:
@nedson When I reboot this way, I have a script that dumps some information in a final about the kernel state. Here are the time it rebooted this way since the start of the 2019 : 

```
-rw-r--r--  1 root root 34561 janv. 12 13:27 2019-01-12T13:26:40+01:00.txt.gz
-rw-r--r--  1 root root 34430 janv. 13 03:03 2019-01-13T03:02:32+01:00.txt.gz
-rw-r--r--  1 root root 34817 janv. 14 03:21 2019-01-14T03:20:39+01:00.txt.gz
-rw-r--r--  1 root root 34974 févr.  2 03:40 2019-02-02T03:40:19+01:00.txt.gz
-rw-r--r--  1 root root 34170 févr.  3 07:12 2019-02-03T07:11:51+01:00.txt.gz
-rw-r--r--  1 root root 34181 févr.  4 03:17 2019-02-04T03:17:04+01:00.txt.gz
-rw-r--r--  1 root root 34158 févr.  8 03:06 2019-02-08T03:06:04+01:00.txt.gz
-rw-r--r--  1 root root 34078 févr.  9 03:32 2019-02-09T03:32:21+01:00.txt.gz
```

Adirelle, 2019-03-02:
@neheb what is DSA support and could it affect network stability ?

neheb, 2019-03-02:
Upstream driver for the switch. The current solution cannot be upstreamed.

Probably does not affect network stability.

The OpenWrt driver has some non upstreamed stuff that could help.

neilbrown, 2019-03-04:
Those people who still experience this network problem: what kernel are you running.  i.e. what does "uname -a" report?
As I said back in May, I think this problem is understood and fixed in later kernels.
If you are running Debian, I can easily provide you a firmware to run a newer kernel.
If you have some other user-space, and don't want to switch to Debian, what are you using?  I might be able to rebuild that firmware with the fix.

nedson, 2019-03-04:
Hello @neilbrown. I am running your kernel:

Linux gnubee-n1.gnubee 4.4.174 #124 SMP Mon Feb 11 08:58:13 AEDT 2019 mips GNU/Linux

Running a very simple setup. Basically Debian Stretch with NFS.

neilbrown, 2019-03-04:
That surprises me ... I really thought this was fixed.
Could you modify the script so that if it detects a problem, it runs "dmesg > /root/dmesg.log" or something like there, then see if there are any interesting messages at the end of dmesg.log after it reboots?

Adirelle, 2019-03-04:
It still happens on 4.15.18+ yet less frequently.  Here are the dumps made just before rebooting:
[dumps.zip](https://github.com/gnubee-git/GnuBee_Docs/files/2924643/dumps.zip)

They are generated by the following commands:

```shell
date
uptime
vmstat 1 1
netstat -ieW
netstat -aopenW
lsmod
ps -ef
dmesg -c
echo t > /proc/sysrq-trigger
dmesg
```

They do no seem to contain anything interesting. No warning nor error messages.

neilbrown, 2019-03-04:
Thanks for the dump - as you say, nothing obvious.  That narrows the search space down.
I'll try to have another look at the code sometime in the next week.

Actually... I don't think the patch ever got to the 4.15 kernel ... or 4.20.  Maybe I'll add the patch to those two kernels and get you to try again.

Adirelle, 2019-03-04:
I think I built those kernels from your repository. Does it include the patches ?

nedson, 2019-03-04:
I will be happy to test any udpated kernels as well. Haven't gotten around to dumping dmesg to a file when the failure occurs...

In the last three days, network has stopped working twice and required a reboot.

neilbrown, 2019-03-05:
I looked more carefully and my 4.15 kernel does have the patch, it just looks a bit different there.

I found another strangeness in the code.  I'm far from certain that it is the problem, but it is worth fixing.
I've pushed the patch out to my v4.4 and v4.15 branches, a build a new firmware: gnubee-pc1-4.4.174-2.bin

I suspect it won't help, but please test and let me know.

neheb, 2019-03-06:
@neilbrown mind if i sent that patch to OpenWrt?

neilbrown, 2019-03-06:
No I don't mind, but I haven't actually tested it myself.  So as long as someone confirms that it doesn't actually break anything, free free to forward it to OpenWrt.

nedson, 2019-03-06:
@neilbrown installed gnubee-pc1-4.4.174-2 last night. Thank you.

meskio, 2019-03-07:
I have the same issue with gnubee-pc1-4.4.174-2, right now in my system the network dies in around 1 hour after boot up.

meskio, 2019-03-07:
The problem was not present when I was running 4.15 from neilbrown branch, I did compile it around 10 months ago.

But 4.15 for me was giving me around 30Mbit/s over the ethernet, which I find pretty slow for a NAS. With 4.4 I get over 300 Mbit/s.

neilbrown, 2019-03-07:
> But 4.15 for me was giving me around 30Mbit/s over the ethernet

I don't think this is a code problem, more likely a config problem.  I had really slow network performance until I turned off CONFIG_SLUB_DEBUG - then my mainline kernel was as fast as 4.4.

neilbrown, 2019-03-07:
> I have the same issue with gnubee-pc1-4.4.174-2, 

Thanks for testing.  I'll need to dig deeper.
Maybe I should try to reproduce it myself.  You are getting this with mostly NFS traffic - is that right?  I'll try bashing my gnubee with NFS.

neilbrown, 2019-03-07:
It is really bizarre that people can still ping hosts, but they cannot get other traffic to work.
An easy explanation for this would be some sort of filtering somewhere that drops non-ICMP packets, but that seems extremely unlikely.
Another explanation might be performance related.  Ping sends and receives one packet per second, anything else is likely to go a lot fast.
Could someone please try a ping of some host next time their gnubee experiences there problem, and report the output.  I'm particularly interested to see the reported round-trip times, and how this compares to 'ping' times when the problem isn't happening.

meskio, 2019-03-09:
> Maybe I should try to reproduce it myself. You are getting this with mostly NFS traffic - is that right? I'll try bashing my gnubee with NFS.

Yes, I see this problem when I have continuous load over NFS.

> An easy explanation for this would be some sort of filtering somewhere that drops non-ICMP packets, but that seems extremely unlikely.

Yes, that was my first guess, did I set up some iptables or whatever? But I just roll back to my previous 4.15 kernel and the problem has disappeared. I think there is something in the 4.4 image.

> Could someone please try a ping of some host next time their gnubee experiences there problem, and report the output. I'm particularly interested to see the reported round-trip times, and how this compares to 'ping' times when the problem isn't happening.

I've pinged the gnubee without failing:
```
10 packets transmitted, 10 received, 0% packet loss, time 21ms
rtt min/avg/max/mdev = 1.208/3.505/13.729/3.699 ms
```
And after the failure:
```
10 packets transmitted, 10 received, 0% packet loss, time 238ms
rtt min/avg/max/mdev = 0.496/0.614/0.707/0.062 ms
```

Tell me if you need any more testing.

neheb, 2019-03-09:
@neilbrown NACKed? by Chinese guy. I can't figure out what he's saying. 

https://patchwork.ozlabs.org/patch/1052106/

neheb, 2019-03-09:
Actually I think I get something else...

the driver is meant to poll for a weight of 128, but the linux networking stack sets a maximum of 64.

for single core non-mt7621, a weight of 32 is specified (I lowered it to 16 mistakenly). 32 is multiplied by 4 (I guess the cores) for a total of 128. But the kernel does not allow that.

This might explain why I get consistent iperf speeds when messing with the SMP affinity...

meskio, 2019-03-09:
> > But 4.15 for me was giving me around 30Mbit/s over the ethernet
>
> I don't think this is a code problem, more likely a config problem. I had really slow network performance until I turned off CONFIG_SLUB_DEBUG - then my mainline kernel was as fas
t as 4.4.

Sure is a config problem. I don't have much experience compiling kernels and I get lost with the config.

I see CONFIG_SLUB_DEBUG was activated in my kernel, it looks like is activated by default in your 4.15 branch gnubee2 config. I have tried to recompile disabling it from the latest version in your repo's 4.15 branch, but it gives me a kernel panic:
```
[  OK  ] Created slice system-getty.slice.
         Starting Set the console keyboard layout...
[   22.760000] Installing knfsd (copyright (C) 1996 okir@monad.swb.de).
[   22.770000] Unhandled kernel unaligned access[#1]:
[   22.780000] CPU: 2 PID: 636 Comm: mount Not tainted 4.15.18+ #4
[   22[. 7 9 02020.07]9 0$0 040 ]   $:  4c 4 9 7:c 5c14c9 70c05010c00 0000 000000000000 0000 000000000000 0000
000000
[   22.790000] $ 8   : 00000008 806d6248 9bd75da4 00000000
[   22.790000] $12   : 00000000 00000000 00000400 ffffffff
[   22.790000] $16   : 9bd1fa80 c497c51c c497c51c 9947aa80
[   22.790000] $20   : 00000000 9947aa00 00000000 00000020
[   22.790000] $24   : 00000000 800083ac
[   22.790000] $28   : 9bd74000 9bd75e20 9947aa80 8010f444
[   22.790000] Hi    : 00000002
[   22.790000] Lo    : 00000000
[   22.790000] epc   : c496b354 nfsd_mount+0x14/0x48 [nfsd]
[   22.790000] ra    : 8010f444 mount_fs+0x20/0xbc
[   22.790000] Status: 1100fc03 KERNEL EXL IE
[   22.790000] Cause : 40800010 (ExcCode 04)
[   22.790000] BadVA : 6e756f81
[   22.790000] PrId  : 0001992f (MIPS 1004Kc)
[   22.790000] Modules linked in: nfsd lockd grace sunrpc ip_tables x_tables autofs4 mmc_block mtk_sd mmc_core usb_storage xhci_mtk xhci_hcd usbcore usb_common nls_base
[   22.790000] Process mount (pid: 636, threadinfo=4a42fc08, task=a87635de, tls=77bdb490)
[   22.790000] Stack : 9947aa80 80ff0000 80ff1040 00000020 9947aa80 8012c328 99888f38 8010f444
[   22.790000]         00000004 00000000 ffffffed c497c51c 00000000 9947aa80 9bd1fa80 c497c51c
[   22.790000]         00000000 8012de0c 00000004 9947aa00 00000000 00000002 ffffffed 00000020
[   22.790000]         00000000 c497c51c 00000000 8012f974 00000000 00000000 00000005 801074cc
[   22.790000]         00000000 9a0620f8 00000008 9b8bacc0 00000005 9947aa00 9bd79a90 9b8badd0
[   22.790000]         ...
[   22.790000] Call Trace:
[   22.790000] [<94415c87>] nfsd_mount+0x14/0x48 [nfsd]
[   22.790000] [<8d73d3c2>] mount_fs+0x20/0xbc
[   22.790000] [<194a7258>] vfs_kern_mount+0x6c/0x134
[   22.790000] [<bdb94a7e>] do_mount+0x664/0xb6c
[   22.790000] [<9dfa7766>] SyS_mount+0xd0/0xe4
[   22.790000] [<29577356>] syscall_common+0x34/0x58
[   22.790000] Code: afbf001c  8f820000  8c4203ac <8c470014> 3c02c497  2442b388  afa20014  8ce20030  afa20010
[   22.790000]
[   22.790000] ---[ end trace c54959b2cfd8d046 ]---
[   22.790000] Kernel panic - not syncing: Fatal exception
```

 I have rolled back into my previous self-compile kernel (with slow network, at least it works).

Adirelle, 2019-03-10:
> I found another strangeness in the code. I'm far from certain that it is the problem, but it is worth fixing.
> I've pushed the patch out to my v4.4 and v4.15 branches, a build a new firmware: gnubee-pc1-4.4.174-2.bin

Just recompiled your new v4.15 with these changes. I will let you known if it changes something.

neilbrown, 2019-03-10:
> @neilbrown NACKed? by Chinese guy. I can't figure out what he's saying.

The things said about napi are simply wrong.  As long as fe_poll() returns "budget" - which it doesn't if there might be more work to do - napi will keep calling fe_poll().  Interrupts are completely irrelevant until fe_poll() returns something less that budget.  At that point we must be sure that and opportunity to do more work is signaled by an interrupt.
If we acknowledge the interrupt *after* testing if there is work to do, then new work could come in between the test and the acknowledgement and so we might never see the interrupt.

Enabling the interrupt (which is quite different from acknowledging it) isn't so sensitive to timing.  The interrupt should be disabled before scheduling napi, and re-enabled before returning a value less than 'budget' from fe_poll() (which effectively de-schedules napi).  The code gets this right.

I don't think the "weight" number is at all interesting.  It ends up being the "budget" which is passed to fe_poll().  If you have multiple network interfaces, then giving the faster one a large "weight" might result in more fairness, I doubt you would noticing unless there was a large disparity in the performance or the weight values.

neilbrown, 2019-03-10:
> I have tried to recompile disabling it from the latest version in your repo's 4.15 branch, but it gives me a kernel panic:

This is almost certainly because you are using old modules with the new kernel.  The mkinitramfs script doesn't cope will with changing the kernel but not changing the version.
If you set CONFIG_LOCALVERSION_AUTO and rebuild, it should fix this.
I've pushed out a change to gnubee[12]_defconfig to remove SLUB_DEBUG and add LOCALVERSION_AUTO.

neilbrown, 2019-03-10:
@meskio thanks for the ping results.  It is interesting that they are slower when the system is working.  Presumably it is very busy?
The important detail is that even when not working, the ping times are good, less than 1ms.  This confirms that interrupts are working. 
Could you try to ping with a larger packet size, e.g. "-s 1400" and "-s 4096".  If either of these consistently fail, try to find the large size that consistently works.

I would be nice to turn on internal debug tracing, but it looks like that can only be done at boot time  - or module load time - so it might be best to try other things first.

Can you run "tcpdump" to collect network traffic too and from the gnubee. e.g.
  tcpdump -s 0 -w /tmp/trace.pcap -i $interfacename host $IPADDRESS &
then while that is running, try to connect to the gnubee.  After it times out, interrupt the tcpdump and attach the trace.pcap - or put it somewhere that I can download.  I'm wondering if there are partial replies, or corrupt replies or something like that.
So far, I haven't been able to reproduce any problem myself (running 4.4.174).

meskio, 2019-03-11:
> @meskio thanks for the ping results. It is interesting that they are slower when the system is working. Presumably it is very busy?

Yes, it might be busy. I have another computer that is using the disks over NFS.

> Could you try to ping with a larger packet size, e.g. "-s 1400" and "-s 4096". If either of these consistently fail, try to find the large size that consistently works.

It works fine, I tried to increment it and it works up to 65507, the maximum allowed (is that a packet size of icmp?). And I keep getting normal times:
```
10 packets transmitted, 10 received, 0% packet loss, time 24ms
rtt min/avg/max/mdev = 4.038/4.334/4.726/0.208 ms
```

> Can you run "tcpdump" to collect network traffic too and from the gnubee. e.g.
> tcpdump -s 0 -w /tmp/trace.pcap -i $interfacename host $IPADDRESS &
> then while that is running, try to connect to the [...]

https://share.riseup.net/#EAXsiaecoFdguQCOcHg1PQ
Here you see an attempt to ssh the gnubee that hangs and a curl to port 9090 (where there should be transmission answering) that returns with a 'Connection refused'.

> So far, I haven't been able to reproduce any problem myself (running 4.4.174).

Weird, maybe is me that I have more load in my NFS or is it easier to hit on the gnubee2. For me it takes few hours to reproduce it.

meskio, 2019-03-11:
> This is almost certainly because you are using old modules with the new kernel. The mkinitramfs script doesn't cope will with changing the kernel but not changing the version.
> If you set [...]

I'm running 4.15 with the default gnubee2_defconfig and so far so good. Thanks.

neilbrown, 2019-03-11:
Thanks for the further testing.  I think we can now rule out both missing interrupts and problems with buffer space.
The network trace shows nothing for port 9090, but it does show and failed attempt to connect to port 9091 - the gnubee says there is nothing to connect to.
The 'ssh' part of the trace is more interesting.  I can see the 'SYN' sent to the gnubee, and the "SYN/ACK" reply, but no ACK back to the gnubee as would be expected.
The suggests that the SYN/ACK from the gnubee is not being accepted.
The packets looks fine, though I cannot be sure of the checksum - checksum testing is often done on the network card these days, so what is seen on the host may not be the same as was on the wire.
However the checksum is the only place where I can think that a problem could be.
You can turn of tcp checksum offload on the gnubee with the command
 
> `ethtool  -K eth0.1 tx off`

(assuming `eth0.1` if the relevant network interface).
You can see what offload is enabled with `ethtool -k eth0.1`.
If you, or someone still running the 4.4 kernel and experiencing the problem, could try to disable tx checksum offload and see if it makes a difference, that would be helpful.

neheb, 2019-03-11:
eth0.1 is a VLAN interface in OpenWrt. The integer ones are the real interfaces.

neilbrown, 2019-03-11:
Yes, eth0.1 is a vlan interface, but it appears that it is still possible to disable tx checksumming on it.  So I think it is worth a try.

neilbrown, 2019-03-12:
.... but yes, it would be safest to disable tx checksumming on both  eth0 and eth0.1

xvybihal, 2019-03-15:
I am not bringing any new hard data, but I just want to report, that the network failure still exists in 4.15.18-163850-g7e529f40779d. I was watching movie via NFS and it just locked out the network. Had to reboot to get it working again. Using untouched gnubee1_defconfig from Neilbrowns gnubee/v4.15 branch.

meskio, 2019-03-20:
I haven't seeing the problem with the gnubee/v4.15 branch in a gnubee2. I'm running that kernel since more than a week without problems and heavy use of NFS.

I will not be able to do any tests on the gnubee for few weeks, sorry.

meskio, 2020-04-27:
I still see this problem with the latest 5.6.3 in my gnubee2. Using NFS it doesn't last more than 2 days until the network stops working, sometimes is less than one hour. the only message I see in dmesg/kern.log is:
```
[46004.346781] rpc-srv/tcp: nfsd: sent only 44680 when sending 65668 bytes - shutting down socket
```

I've being running for the last year my own compiled 4.15.18 from your repo. This one did fail as well sometimes, but it usually lasted more than a month until it fails. I'll try to stick with 5.6.3 using samba instead of nfs, let's see if it works more stable.

Tell me if I can do something to help debugging this. Thank you.

neilbrown, 2020-04-27:
That "rpc-srv/tcp: nfsd: sent only 44680..." messages is, I think, caused by running out of memory.  Do you have a swap partition configured?  If not, does adding one help.
Alternately (or as well) trying changing "newsock->sk->sk_sndtimeo = HZ*30;" in net/sunrpc/svcsock.c to a larger number - maybe much larger (HZ*60*60*24).  That will cause nfsd to wait for memory rather than giving an error.
When you say "the network stops working", is it all network traffic, or just NFS?

meskio, 2020-04-28:
> That "rpc-srv/tcp: nfsd: sent only 44680..." messages is, I think, caused by running out of memory. Do you have a swap partition configured? If not, does adding one help.

I have a swap of 2GB, I could increase it, but I almost never see it full. I can try to monitore it closely.

> Alternately (or as well) trying changing "newsock->sk->sk_sndtimeo = HZ_30;" in net/sunrpc/svcsock.c to a larger number - maybe much larger (HZ_60_60_24). That will cause nfsd to wait for memory rather than giving an error.

I'll try to compile a kernel with that change, I don't even remember how I compiled it last time. I was using your prebuild image from http://neil.brown.name/gnubee/

> When you say "the network stops working", is it all network traffic, or just NFS?

The same problem we discussed before. Ping keeps working, but I can't do any kind of TCP connection (I tried ssh and samba) until I restart the gnubee.

neilbrown, 2020-04-30:
Hmm.... don't bother with more swap - the memory exhaustion is probably just the wmem limit (/proc/sys/net/ipv4/tcp_wmem) because the TCP stream has stalled.
Did you try disabling checksum offload?  "ethtool -K eth0 tx off" should do it.

If it happens again with checksum offload disabled, I'd appreciate a 'tcpdump' from some other machine that is trying to connect to the gnubee - we did that before but it wouldn't hurt to get a fresh one to make sure the same thing is still happening, and maybe to get more clues.

neilbrown, 2020-04-30:
The other thing I was going to suggest is to see if NFS/UDP works.
ie. "mount -o udp server:/path /mountpoint"
If that works, it will suggest a TCP problem.

neheb, 2020-04-30:
I will mention ksmbd as an alternative to NFS for these devices: https://github.com/cifsd-team/cifsd

A PC1 user privately told me he gets ~47MB/s ksmbd vs ~25MB/s NFS. I don't use NFS so I had no answer.

As far as network failures go, we recently switched to the upstream ethernet driver in OpenWrt for mt7621. This was done as the old OpenWrt specific driver had several issue that went unfixed.

The biggest one has to do with ethernet flow control and/or pause frames. Here's a description of the problem: http://lists.infradead.org/pipermail/openwrt-devel/2020-February/021742.html

neilbrown, 2020-04-30:
Thanks - that email is interesting. I suspect that "pause" frames are the cause of the long standing problem with the u-boot network control.
As you probably know, if you leave a gnubee at the u-boot prompt with a network port connected to a switch, the switch will stop working after a few minutes.  Presumably the gnubee starts sending pause frames for some reason.  It would be nice to fix that aspect of the u-boot.

I doubt the pause frames are causing the present problem.  However for testing it can be disabled with
ethtool -A ethblue autoneg off
ethtool -A ethblack autoneg off

neheb, 2020-04-30:
In the linked email, he disabled pause frames for all ports. Currently, I believe pause frames are only disabled for the CPU port.

tsileo, 2020-05-01:
Hey, for the record, I also always had this issue too. I recently upgraded the firmware to `gnubee-5.6.3-gbpc2.bin` and Debian Strech (and it did not fix the issue).

I am only running minidlna and deluged on the server, and it rarely last more than hours before the network starts freezing.

If I remember correctly, before it would not even respond to ping, but with my recent upgrade, pinging the server still works when the "network gets freezed" (I cannot SSH into it/current session freezes and the DLNA server is not working anymore).

Is there anything I can do to help troubleshoot the issue? 

Thanks!

neilbrown, 2020-05-01:
> Is there anything I can do to help troubleshoot the issue?

Yes.
1/ run immediately after boot "ethtool -K eth0 tx off" and report whether or not that makes any difference.
2/ While the machine is not working, attempt to connect to it from another machine while running "tcpdump -s 0 -w /tmp/gnubee.pcap host IP-OF-GNUBEE".  Then interrupt the tcpdump and put the "/tmp/gnubee.pcap" somewhere that I can download it, or email it to me (neil@brown.name).
3/ While the machine is not working try accessing some service that uses UDP.  NFS is probably easiest - don't worry if you don't have it configured and don't find it straight forward to set up.

[9 more replies]