gnubee-git/GnuBee_Docs #55: GnuBee PC2 Deluxe Kit shipped and still no documentation for PC2 available ?

blue101010, 2018-01-09:
GnuBee PC2 Deluxe Kit shipped and still no  documentation for PC2 available?
Please upload ALL PC2 documents somewhere ASAP.

Why https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PC2 is missing

Without any documentation on PC2 this is seriously impacting and nothing like open-source ?

Thank you
Franck

noblepepper, 2018-01-17:
> is seriously impacting

AFAI there are very few schematic changes and no changes to the source code so I don't understand how this is impacting much of anything.

> nothing like open-source

The source is available on github so it is **totally open-source**. You probably mean open-hardware which is a different concept.

blue101010, 2018-01-17:
Hi,

For PC2 The PCB layout is different, the hardware is different and there
are undocumented sources differences.
There is a good website on PC1 version : http://gnubee.org/internals.html
with a PDF on Assembly for PC1 :
http://gnubee.org/GB-PC1/Assembly/Assembly.pdf
However there is strictly no published document on PC2 (layout ? size ?
sources differences ?)

So this is disappointing I wished to have bought only a PC1 beccause as of
now PC2 without the same website documentations for PC2 that however exist
for PC1 is officially no way "open"
I do not understand why PC1 doc was published and where is the difficulty
to create a branch with the documents for PC2 ?

Thanks

blue101010, 2018-01-17:
Closing due to https://github.com/gnubee-git/GnuBee_Docs/tree/master/GB-PCx branch rename and new availability of https://github.com/gnubee-git/GnuBee_Docs/blob/master/GB-PCx/Documents/GB-PC2%20V1.0.pdf

closed 2018-01-17 by blue101010