gnubee-git/GnuBee_Docs #56: recommended network settings?

cat-tech, 2018-01-12:
Hi people, 

I am having issues with finding the right settings for the network interface. What do you recommend, or have set up on your system? 

I have found that I need to have ipv6 set to disabled to have the GnuBee not hang during boot. 

Currently have (this was set by OMV):
root@gnubee-pc1:~# cat /etc/network/interfaces
# The loopback network interface
auto lo
iface lo inet loopback

# eth0 network interface
auto eth0
allow-hotplug eth0
iface eth0 inet dhcp
    dns-search HAME
iface eth0 inet6 manual
    pre-down ip -6 addr flush dev $IFACE

# eth0.1 network interface
auto eth0.1
allow-hotplug eth0.1
iface eth0.1 inet static
    address 192.168.10.1
    netmask 255.255.255.0
    dns-nameservers 192.168.10.1
    dns-search HAME
iface eth0.1 inet6 manual
    pre-down ip -6 addr flush dev $IFACE

# eth0.2 network interface
auto eth0.2
allow-hotplug eth0.2
iface eth0.2 inet dhcp
    dns-search HAME
iface eth0.2 inet6 manual
    pre-down ip -6 addr flush dev $IFACE
   
The web interface settings:
[image]

FWIW: 
Using 2 x SSDs, partitioned with  Debian GNU/Linux 8 (jessie)! on partion on drive in slot 1. 
 uname -a
Linux gnubee-pc1 4.4.87-gnu #0 SMP Wed Nov 22 13:06:13 2017 mips GNU/Linux
openmediavault 3.0.65 (Erasmus) gnubee-pc1 ttyS0
initial boot is into:
[    0.000000] Linux version 4.4.87-gnu (dgazineu@gnubee.net) (gcc version 5.4.0 (LEDE GCC 5.4.0 r64-b029924) ) #0 SMP Wed Nov 22 13:06:13 2017

And when it hangs -it is not with the above settings, but if I re-enable ipv6, it would - it hangs at:
[   28.226057] systemd-journald[554]: Received request to flush runtime journal from PID 1
[[32m  OK  [0m] Started Trigger Flushing of Journal to Persistent Storage.
         Starting LSB: Set console font and keymap...
         Starting Network Time Synchronization...
         Starting Update UTMP about System Boot/Shutdown...
[[32m  OK  [0m] Started Network Time Synchronization.
[[32m  OK  [0m] Reached target System Time Synchronized.
[[32m  OK  [0m] Started Update UTMP about System Boot/Shutdown.
[[32m  OK  [0m] Started LSB: Set console font and keymap.
[[0m[31m*     [0m] A start job is running for LSB: Raise network interf...11s / no limit)
As you can see, about 29 seconds into boot sequence. And yes, that last line lasts forever...

While I have dabbled in *nix since Minix days, I would not count myself as fluent, or maybe even competent. Any help would be appreciated.

Adirelle, 2018-01-12:
I believe eth0 should not be configured at IP level (e.g. set it to manual).

My /etc/network/interfaces looks like this (I do not use eth0.1):

```
auto lo
iface lo inet loopback

auto eth0
iface eth0 inet manual

auto eth0.2
iface eth0.2 inet static
    address 192.168.0.11/24
    gateway 192.168.0.254

# Site-local IPv6 address
iface eth0.2 inet6 static
    address fd15:83c4:f5b8:3e3b::11/64
    # Does this do something at all ?
    scope site 

# Global IPv6 address
iface eth0.2 inet6 static
    address public-prefix-assigned-by-my-ISP::11/64
```

But as I said in #54, I have random, weird failures (IP, ICMP: ok, TCP & UDP: dead) so maybe I am doing something wrong.

cat-tech, 2018-01-12:
Hi Adirelle, 

thanks for sharing your interfaces file. A useful point of comparison. 

Are you hand editing this? OMV won't let me save eth0 as static without an address and mask. 
After removing the Domain reference I get (showing for eth0 only) - saving from OMV:
' #' eth0 network interface
auto eth0
allow-hotplug eth0
iface eth0 inet dhcp
iface eth0 inet6 manual
    pre-down ip -6 addr flush dev $IFACE

Still significantly more than you have. Could I set eth0 as you suggest from within OMV? How would I do that?

I have used eth0, eth0.1 and eth0.2 only because that was how it was set in Debian before OMV was installed.  

And my 2c on your file, could you be getting a conflict between the 2 inet6 addresses for eth0.2?

Adirelle, 2018-01-12:
> Are you hand editing this?

Yes.

> And my 2c on your file, could you be getting a conflict between the 2 inet6 addresses for eth0.2?

It's fine to have several IPv6 addresses on the same interface; I think that it is the way it is intended to work, as there are different classes of addresses (link-local, site-local, global) with different purposes.