gnubee-git/GnuBee_Docs #57: Fix typos on webpage?

Cabalist, 2018-01-16:
Hey I am noticing a lot of typos on the webpage.  Is the source on github for editing?

neilbrown, 2018-05-13:
I'm sorry with was left unanswered for so long, particularly when the answer is so easy (once you know).
The pages for https://gnubee.org/ are built from https://github.com/ngiger/GnuBee_Docs/tree/gh-pages
I've submitted a pull request with some fixes that you can view at  https://neilbrown.github.io/GnuBee_Docs/
If the typo you found isn't fixed there, you could make your own pull request, or describe it here and I'll submit a fix.

neilbrown, 2018-05-20:
The gnubee.org pages has just been updated and should have most typos fixed.
Other updates can be submitted to the gh-pages-gnubee.org branch of this project where that can be reviewed by other interested parties, then submitted for inclusion.

closed 2018-05-20 by neilbrown