gnubee-git/GnuBee_Docs #58: Broken debian install instructions on http://gnubee.org/install_debian.html

noblepepper, 2018-01-17:
Commit d98ef8abd59ec9f1790b0699c0ee7693be05ff09 breaks the instructions for installing debian on the website. 

Changing 
``` wget --no-check-certificate https://raw.githubusercontent.com/gnubee-git/GnuBee_Docs/master/GB-PC1/scripts/jessie_3.10.14/debian-jessie-install```
to 
``` wget --no-check-certificate https://raw.githubusercontent.com/gnubee-git/GnuBee_Docs/master/GB-PCx/scripts/jessie_3.10.14/debian-jessie-install```
 fixes it

meskio, 2018-04-30:
Also the sha256 from the openwrt-... image is missing the first `8a` chars.

remraz, 2018-12-03:
### Some feedback from my successful installation : 
* debian-jessie-install wouldn't detect a already formatted partition made on top of a logical volume. When it finally did, the script failed to carry on.
* At the end, osm-initsystem was seemingly not found. The following did the trick : https://forum.openmediavault.org/index.php/Thread/17341-Problems-after-omv-initsystem/?postID=137108#post137108 
* I also had to had no-check-certificate to the wget inside _debian-modules-install_. Shall it get a PR ?