gnubee-git/GnuBee_Docs #59: Walkthrough - not a bug, but may help some

cat-tech, 2018-01-18:
Walkthrough for installing Open Media Vault on GnuBee  

Caveat Emptor: this is, as best I can recollect, the process I ended up using to install OMV on GnuBee hardware. It is offered in the hope that some part will help others. 

Highly desirable it is to have:
a PC that can take the boot drive for the GnuBee. Once there is an identifiable GNUBEE-ROOT you will not get a LUCI command line.   
a serial cable to monitor the boot process. 

## Assembly
I recommend placing insulating washers between the mounting brackets and the PCB. There are tracks running under the mounting brackets, and these may short either when first assembled, or later. These are not to prevent continuity between the PCB ground and case, but serve 2 purposes: physical separation, and to not create any short circuits themselves. 

## serial (ttyS0) adapter
The serial adapter is required to be a 3.3 volt type. Connections are: 
[image]

If you are not sure, better to buy one of the pre-made ones. Good value at $8. 

## Selection of boot location
The boot drive, it appears, cannot be on part of a RAID array. If you use the uSD card as boot, then use the WRT version to flash GnuBee. If you use a separate 2.5" drive, then both WRT and Libre flashes should work. See documentation. 

It may be possible to use the boot drive to form part of a RAID array, but this will require the array to be formed manually. I don't know, above my pay scale... 

## Debian
Flash firmware and install Debian to your chosen target as per documentation. http://gnubee.org/install_debian.html Ensure there is sufficient space for installation, it appears the script is silent if any part fails. 
One exception is that I partitioned my boot drive Thus:
Device        Start       End   Sectors  Size Type
/dev/sda1      2048  41945087  41943040   20G Linux filesystem
/dev/sda2  41945088  46139391   4194304    2G Linux swap
/dev/sda3  46139392 234455006 188315615 89.8G Linux filesystem

At this point also if you have partitioned as I have:
mkfs -t <your chosen FS type> /dev/sda3 
Use blkid to find the UUID. Use UUID not PARTUUID. See mounting other drives below. 

Check Debian has picked up the swap partition:
swapon -s 
root@gnubee-pc1:# swapon -s
Filename                                Type            Size    Used    Priority
/dev/sda2                               partition       2097148 0       -1
root@gnubee-pc1:/~#

If not: 
mkswap /dev/sda2
swapon /dev/sda2
To have this available as swap after reboots, add to /etc/fstab using vi or other text editor. If you are not comfortable with vi, perhaps edit fstab by moving the drive to another machine.   eg 
/dev/sda2       none    swap    defaults        0       2

Reboot. 

## Install OMV
Install OMV as per documentation. http://gnubee.org/install_omv.html I suggest you use SSH rather than the serial monitor for this, the selection screens are more difficult on the serial (ttyS0) link. 
The last instruction in the OMV script is omv-systeminit. This triggers some selection screens. I accepted the defaults, apart from the locale, which I recommend is set to en-US-UTF8. You may wish to add others as well.  

Log into OMV through your browser. 

The network settings need to be there, or when the configuration is saved, the interfaces file has only the loopback active. And you will be locked out if you do not have a serial (ttyS0) connection. To repair, the boot drive can also be physically moved to another machine, and the /etc/network/interfaces file reconfigured.  

Select System, Network, Interfaces tab. 
The settings I found to work:
interface: eth0.1 or eth0.2
IP4
	set to DHCP or Static as suits. Include all necessary fields. 

IP6
		I have not got this to work, and set it to disabled. If this is set to anything else, it hangs during boot. Repair is only possible by putting the boot drive in another machine, and editing  /etc/network/interfaces. 

Select Storage, Physical Disks. 
	Select each of the disks to use in your RAID in order, and Wipe each of them. Do not do this for your boot drive.   

Select Storage, RAID Management
	select "+Create" to create new array. Your newly Wiped disks should be listed. Be patient, it may take 10 or 20 seconds. 
In my case I used 2 x 500GB drives mirrored, and it took just under 1 hour to organise itself. 
[image]

Wait for this step to complete before formatting. 
[image]

Go to a command terminal, and format the new array. 
First, check what it is:
root@gnubee-pc1:~# ls -lrt /dev/md*
brw-rw---- 1 root disk 9, 0 Jan 16 02:42 /dev/md0

Then eg 
mkfs -t ext4 /dev/md0
		*** note: I think BTRFS is preferred for SSDs, I need to check. *** It would be nice if formatting could be done within OMV, but not yet.
		
Back in the web interface, the new RAID should now be available to mount. 
Select Storage, File Systems. 
	select /dev/mdo and then mount. 
	OMV will create a mount point on /media, using the UUID, and an entry in /etc/fstab. The entry for your md array will show 'Yes' under mounted, and 'Online' under Status.  

This is what my fstab looks like with 2 OMV mount points, one is /dev/sda3, the other is /dev/md0 (2 disk array):	
root@gnubee-pc1:~# cat /etc/fstab
(#} UNCONFIGURED FSTAB FOR BASE SYSTEM
LABEL=GNUBEE-ROOT /  ext4  noatime,errors=remount-ro 0  1
tmpfs           /tmp            tmpfs   defaults        0       0
/dev/sda2       none    swap    defaults        0       2

(#) >>> [openmediavault]
UUID=ec6e00e6-981c-464b-aab3-02f376ab6af9 /media/ec6e00e6-981c-464b-aab3-02f376ab6af9 ext4 defaults,nofail,user_xattr,noexec,usrjquota=aquota.user,grpjquota=aquota.group,jqfmt=vfsv0,discard,acl 0 2
UUID=8991be53-9979-4296-adc1-99c18425d475 /media/8991be53-9979-4296-adc1-99c18425d475 ext4 defaults,nofail,user_xattr,noexec,usrjquota=aquota.user,grpjquota=aquota.group,jqfmt=vfsv0,acl 0 2
(#) <<< [openmediavault]
root@gnubee-pc1:~#

Select Acess Rights Management, Shared Folders
	select "+Add" to create a new share. 
	your RAID should be visible as a Device. 
You will also need to create Groups and Users to give access to a share. I'll leave this to you to sort out. 

## RAID initialisation
One suggestion if your RAID fails to initialise properly. All drives for a RAID MUST have a partition type of msdos. This can be achieved with wipedisk from within OMV. To do this manually:

Use fdisk or parted. 
eg fdisk -l #(to list all disks)
then eg
fdisk /dev/sdb
	o 	#(to create msdos partition)
	w 	#(to write partition and exit fdisk)
As always, reboot, then hopefully the disks with msdos partitions will now show in the Create window. 

	
## Mounting other drives	
Mounting other drives. For OMV to unmount them, it seems they need to be mounted by UUID. 
First, find the UUID
root@gnubee-pc1:~# blkid
/dev/mtdblock5: TYPE="squashfs"
/dev/sda1: LABEL="GNUBEE-ROOT" UUID="e94dc4a6-cc6e-45d3-a5a7-7dab7668a56b" TYPE="ext4" PARTUUID="f150185e-4754-4e59-9de0-861e4f2f356d"
/dev/sda2: UUID="494043e4-b31a-4d0d-9d19-5a28224799e5" TYPE="swap" PARTUUID="c2343b18-d4a9-4067-a2a7-3ee410bf1c6d"
/dev/sda3: LABEL="rootdata" UUID="ec6e00e6-981c-464b-aab3-02f376ab6af9" TYPE="ext4" PARTUUID="a9215476-1e57-4b0f-98ee-8f7acfcdc3f1"
 
mkdir /media/sda3
eg mount UUID=ec6e00e6-981c-464b-aab3-02f376ab6af9 /media/sda3 defaults 0 2
			*** if you do this then use Storage, File Systems to unmount and remount drive to create an OMV style mount point. 
or better (because it works in with OMV's way of doing things:
mkdir /media/ec6e00e6-981c-464b-aab3-02f376ab6af9
eg mount UUID=ec6e00e6-981c-464b-aab3-02f376ab6af9 /media/ec6e00e6-981c-464b-aab3-02f376ab6af9 defaults 0 2    

## Clam 
Clamav takes about 70% of available memory - pity. 

## Time servers
I am in Australia, so have set System, Date & Time, Time servers to:
tic.ntp.telstra.net,toc.ntp.telstra.net,au.pool.ntp.org,2.au.pool.ntp.org
What is best for you depends on your ISP. Check the ping times to different servers, and put the quickest responses first. 

## network interfaces file
My original network interfaces file:
root@gnubee-pc1:~# cat /etc/network/interfaces
auto lo
iface lo inet loopback

auto eth0.1
iface eth0.1 inet static
    address 192.168.1.1
    netmask 255.255.255.0

auto eth0.2
iface eth0.2 inet dhcp

root@gnubee-pc1:~#

## Power consumption
With 3 SSDs, and idle, my GnuBee draws between 500 and 600 mA, about 6 watts. I'm happy with that, for what should be a unit capable of very good throughput.

cat-tech, 2018-01-18:
No response necessary.

closed 2018-01-18 by cat-tech

pinney, 2018-01-18:
Nice Walkthough Thanks!
A bit of electrical tape may serve just as well between the brackets and PCB.