gnubee-git/GnuBee_Docs #60: Can't login in after updating firmware @d98ef8a

Cabalist, 2018-01-18:
After updating the firmware on the gb-pc2 to `librecmc-ramips-mt7621-gb-pc2-squashfs-sysupgrade.bin` I am unable to access the login page or any web interface.   192.168.10.1 is serving a lua script instead of the login page. 

The lua file contents are:

```lua
#!/usr/bin/lua
require "luci.cacheloader"
require "luci.sgi.cgi"
luci.dispatcher.indexcache = "/tmp/luci-indexcache"
luci.sgi.cgi.run()
```
Running this in a lua interpreter while ssh'd into the PC2 gets me the HTML I am expecting but I can't seem to get the webpage to load.

Where is the firmware to downgrade to the version that shipped with this?  It was working fine before.

Cabalist, 2018-01-18:
Are the pc1 and pc2 firmwares interchangeable?    Could I install the LEDE pc1 system over the existing LibreCMC?

pinney, 2018-01-18:
Ye.s The PC2 will run PC1 firmware.

Cabalist, 2018-01-18:
Cool.  We'll flash the LEDE firmware and report back.

Cabalist, 2018-01-18:
Nope.  :/  got this error message:

```
Device gb-pc2 not supported by this image
Supported devices: gb-pc1
Image check 'fwtool_check_image' failed.
```

I used `lede-ramips-mt7621-gb-pc1-full-sysupgrade.bin`

neheb, 2018-01-18:
@Cabalist need to use --force.

Cabalist, 2018-01-19:
@pinney This did not work.   There are 3 USB ports and none of them gave the blink code that was expected from the docs.  Should they all work the same?  

The blink code I got was Bottom slow.  Top slow. Bottom Solid.  Top fast. and then it can be ssh'd into and the previous firmware is present. 

@neheb I was a little hesitant to try that but I can.

pinney, 2018-01-19:
@Cabalist Any USB port should work.
The USB 'stick' must be formatted with the 'msdos' filesystem ... FAT32 should work but exFAT will not.
The firmware must be in the root directory AND named gnubee.bin

Cabalist, 2018-01-19:
@pinney I believe so.  MBR partitioned, FAT32,  gnubee.bin is the only thing on the drive and is just a renamed `lede-ramips-mt7621-gb-pc1-full-sysupgrade.bin`.  

Would hooking up a serial terminal provide any insight?

pinney, 2018-01-19:
@Cabalist Anytime I have a unit near my desktop ... I plug in a USB to UART :)

Cabalist, 2018-01-19:
Here is what I get:

```
U-Boot 1.1.3 (Aug  7 2017 - 16:23:41)

Board: MediaTek APSoC DRAM: 512 MB

Config XHCI 40M PLL
MediaTek SPI flash driver, SPI clock: 45MHz
spi device id: 1 2 19 4d
find flash: S25FL256S
*** Warning - bad CRC, using default environment

============================================
MediaTek U-Boot Version: 5.0.1.0-6
--------------------------------------------
ASIC MT7621A DualCore (MAC to MT7530 Mode)
DRAM_CONF_FROM: Auto-Detection
DRAM_TYPE: DDR3
DRAM bus: 16 bit
Xtal Mode=3 OCP Ratio=1/3
Flash component: SPI Flash
Date:Aug  7 2017  Time:16:23:41
============================================
icache: sets:256, ways:4, linesz:32, total:32768
dcache: sets:256, ways:4, linesz:32, total:32768

 #### The CPU freq = 900 MHZ ####
 estimate memory size = 512 Mbytes

 Reset MT7530
set LAN/WAN WLLLL
(Re)start USB...
USB0:   mtk-xhci: init hccr be1c0000 and hcor be1c0020 hc_length 32
Register 300010f NbrPorts 3
Starting the controller
USB XHCI 0.96
scanning bus 0 for devices... 3 USB Device(s) found
       scanning bus for storage devices... 1 Storage Device(s) found

** Unable to use usb 0:1 for fatload **

 Upgrade FW from USB storage failed!

Please choose the operation:
   1: Load system code to SDRAM via TFTP.
   2: Load system code then write to Flash via TFTP.
   3: Boot system code via Flash (default).
   4: Enter boot command line interface.
   5: Load system code then write to Flash via USB Storage.
   6: Load system code then write to Flash via Httpd.
   9: Load U-Boot code then write to Flash via TFTP.                                                                                                                                                            0

3: System Boot system code via Flash.
```

Looks like it IS seeing the USB drive but something is keeping it from working.   I've tried reformatting the drive.  Would there be an issue going from libreCMC -> LEDE in this way?  

I also see that I have a CRC error almost immediately.   This would make me think that the previous flash using the webadmin did not go well...

Is there a copy of the firmware that shipped with this device?

Cabalist, 2018-01-19:
Reformatted the USB drive to be sure and the message changed to this:

```
scanning bus 0 for devices... 3 USB Device(s) found
       scanning bus for storage devices... 1 Storage Device(s) found
Device NOT ready
   Request Sense returned 00 00 00

** Unable to use usb 0:1 for fatload **

 Upgrade FW from USB storage failed!
```

Moved ports and it is flashing right now.  :)  Weird.  I suspect my USB drive.  Looking good so far.

Cabalist, 2018-01-19:
Cool.  I'm good to go.   LEDE changes the IP to `192.168.1.1` and my device model is listed as *GB-PC1* in the overview but everything else seems to be there.   Thanks everybody!

neheb, 2018-01-19:
Which kernel version does that firmware use?

Cabalist, 2018-01-19:
4.4.87

neheb, 2018-01-19:
ok. currently kernel 4.9 is broken with disk drives due to some CONFIG option causing breakage. I removed a bunch of them and compiled working firmware but lost the config. Trying to diagnose which exact CONFIG it is.

Cabalist, 2018-01-19:
Gotcha.

I'm impressed by the technical documentation for this project but I think some end-user documentation would really help.   Is the wiki the best place to submit some 'noobie gnubee' guides?

noblepepper, 2018-01-27:
> Is the wiki the best place to submit some 'noobie gnubee' guides?

That's where I've been putting stuff. Since the list on home page was getting pretty long I added a "[more howtos](https://github.com/gnubee-git/GnuBee_Docs/wiki/HowTos)" page. If it gets too long we'll need to make categories. That's a good problem to have ;-)

neilbrown, 2018-05-13:
I think this issue has been resolved, so I'm marking it as 'closed'.

closed 2018-05-13 by neilbrown