gnubee-git/GnuBee_Docs #61: GB2 slot 1 has insufficient clearance

rillian, 2018-01-21:
Hi. Got my GB2 this week and put it together. Very exciting.

Something I didn't see mentioned elsewhere is that there doesn't seem to be enough clearance for a full-height (23mm) 3.5" drive in slot 1. The top edge of the drive case overlaps with the two 12V headers (for case fans, I guess) and can't be seated properly on the connector. It's fine for an SATA flash disk or other smaller drive.

Cabalist, 2018-01-22:
We also noticed this issue.

davidak, 2018-01-30:
Same. See this picture:

[image]

pinney, 2018-01-30:
Low Tech Fix : 

Remove Plastic (I used hemostats)

[image]

Install common breadboard jumper wires

[image]

This should allow one to use most any common 12v fan.

neilbrown, 2018-05-13:
It would be great if these photos and details were placed in the wiki - would someone like to do that?
Then we could close this issue and still have the information easily available.