gnubee-git/GnuBee_Docs #62: SD card not accessible from 4.4.87 firmware

bneumeier, 2018-01-24:
Hi,

The firmware images:
librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-18.bin
librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-28.bin

(and possibly others) do not have support for SD cards built in. That means it's impossible to use the SD card as the debian root filesystem of a PC1 or PC2 with those firmware images. Since the PC2 deluxe kit includes an imaged SD card, it would be really nice if the most up-to-date firmware image allowed that SD card to be used!

neilbrown, 2018-01-31:
This can be fixed by rebuilding the firmware without the CONFIG_HIGHMEM kernel config option.  
Hopefully the people who build the firmware images will see this and give it a try.

dgazineu, 2018-01-31:
I will, thanks for the update!

smurfix, 2018-02-05:
Thank you, rebuilding from the current GnuBee libreCMC repository fixed that problem.
However, I now have a different problem: everything I run from my Debian partition (using chroot, obviously) now results in an ugly "Illegal instruction" error. Any ideas?
```
root@libreCMC:~# mount /dev/mmcblk0 /mnt
root@libreCMC:~# ls /mnt
bin         etc         lost+found  opt         root        srv         usr
boot        home        media       proc        run         sys         var
dev         lib         mnt         rom         sbin        tmp
root@libreCMC:~# chroot /mnt /bin/bash
Illegal instruction
root@libreCMC:~#
```

neilbrown, 2018-02-05:
> now results in an ugly "Illegal instruction" error. Any ideas?

You need to enable CONFIG_MIPS_FPU_EMULATOR when compiling the kernel.
Mainline Linux has a Floating Point Unit emulator that cannot be disabled. openwrt and related distros add a patch to disable it to save a bit of code space, and provide a c library which never uses any FPU instructions.  Glibc in debian does use FPU instructions and so gets SIGILL when it tries to use one.

smurfix, 2018-02-05:
@neilbrown Didn't work last time I tried, but that's probably a missing dependency. I'll run "make clean" and try again.

In any case, that should be done by the LEDE configuration machinery …

neheb, 2018-02-05:
@neilbrown CONFIG_HIGHMEM enables the second 256MB. It's not 2GB.

neilbrown, 2018-02-05:
> CONFIG_HIGHMEM enables the second 256MB. It's not 2GB.

My kernels compiled without CONFIG_HIGHMEM report:
> [    0.000000]  memory: 1c000000 @ 00000000 (usable)
> [    0.000000]  memory: 04000000 @ 20000000 (usable)
and 
> [    0.000000] Memory: 439488K/458752K available (4487K kernel code, 217K rwdata, 836K rodata, 9232K init, 226K bss, 19264K reserved, 0K cma-reserved)

A kernel with CONFIG_HIGHMEM reports
> [    0.000000] Memory: 504092K/524288K available (4451K kernel code, 213K rwdata, 832K rodata, 9148K init, 250K bss, 20196K reserved, 0K cma-reserved, 65536K highmem)

So it does seem that the last 64Meg (not 256M) is unavailable without CONFIG_HIGHMEM.
That's rather interesting.  You should be able to address up to 2GB of RAM completely without CONFIG_HIGHMEM.  I'm sure you can on x86.  Maybe there is something funny about the MIPS MMU....  I'll have a look if I find time.
Thanks.

neheb, 2018-02-05:
That's bizarre. Here are my results:

Before:

cat /proc/meminfo | head
MemTotal:         514292 kB
MemFree:           69460 kB
MemAvailable:      52040 kB
Buffers:             628 kB
Cached:             4288 kB
SwapCached:            0 kB
Active:            14180 kB
Inactive:           2524 kB
Active(anon):      12120 kB
Inactive(anon):      608 kB

[    0.000000] Memory: 514096K/524288K available (3617K kernel code, 199K rwdata, 840K rodata, 196K init, 252K bss, 10192K reserved, 0K cma-reserved, 262144K highmem)

After:

cat /proc/meminfo | head
MemTotal:         254760 kB
MemFree:           44728 kB
MemAvailable:     183168 kB
Buffers:           12720 kB
Cached:           162928 kB
SwapCached:            0 kB
Active:            22600 kB
Inactive:         157168 kB
Active(anon):       4208 kB
Inactive(anon):      496 kB

[    0.000000] Memory: 254556K/262144K available (3616K kernel code, 196K rwdata, 836K rodata, 204K init, 227K bss, 7588K reserved, 0K cma-reserved)

I'm on OpenWrt's 4.4 kernel as OpenWrt's 4.9 kernel returns bad data when reading from /dev/sdX. Some DMA mapping issue probably.

neilbrown, 2018-02-05:
OK ... I don't yet understand why HIGHMEM is needed, but if you want all 512MB of RAM and still to have a working uSD card, then you need to re-enable CONFIG_HIGHMEM and add the line
`    mmc_dev(mmc)->dma_mask = NULL;`
to msdc_drv_probe() in `drivers/mmc/host/mtk-mmc/sd.c`, somewhere after "`mmc = mmc_alloc_host(...)`".
That effectively tells the block layer that the driver cannot handled buffers in highmem, and that they must be copied to/from lowmem for IO.
I think this is a hack rather than a fix, but it is the best I can do without spending several hours untangling the code.
(Note that I've only tested this in 4.15 - not 4.4. I suspect it will still work though)

neilbrown, 2018-02-06:
> That's bizarre. Here are my results:

Yes - bizarre. What do you get after "Determined physical RAM map:" in dmesg?
It might be a difference between 4,.4 and 4.15 which is what I'm mostly playing with (though the network doesn't work yet, so I wouldn't recommend it for production).

neheb, 2018-02-06:
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 1c000000 @ 00000000 (usable)
[    0.000000]  memory: 04000000 @ 20000000 (usable)

The same before and after

pinney, 2018-02-06:
Quite OLD ... but possibly relevant :   
https://dev.openwrt.org/changeset/45892
https://dev.openwrt.org/changeset/45889

neilbrown, 2018-02-06:
Ahhh.. According to http://linux-mm.org/HighMemory a 32bit MIPS processor can only directly map 512M of RAM.  It needs 64M for IO address space, so only 448M of RAM in the direct mapping.  The extra 64M needs to be mapped on a page-by-page basis using TLBs.
So the mmc/sd driver needs to be taught to DMA directly from highmem (which probably isn't hard if you know how).  Until then, my hack which tells the block layer (actually it tells the mmc layer to tell the block layer) to avoid highmem is a good simple solution.
https://github.com/neilbrown/linux/commit/dc97800015e66b29a4bb067b355cc2e2b6eb72cb

neilbrown, 2018-02-06:
> arch/mips/include/asm/mach-ralink/mt7621.h:#define MT7621_DDR2_SIZE_MAX 256

That value is only used for the device-tree file doesn't specify memory.  The code hunts from _SIZE_MIN up to _SIZE_MAX looking for the end of memory.

The reason you only see 256M of low-mem on your 4.4.87 kernel is the patch 
`target/linux/generic/patches-4.4/307-mips_highmem_offset.patch`
It reduces HIGHMEM_START from 512M down to 256M.

neheb, 2018-02-06:
Thank you for investigating. It seems that patch is for reducing the size of the kernel. Appropriate for OpenWrt I guess.

While that small change that disables highmem support in the driver is a bit of a hack, it would probably get accepted given how little attention actually gets paid to the driver.

neheb, 2018-02-18:
Small update on the SD card driver, a user on the LEDE forums has posted one that MediaTek developed for their ARM devices. He stripped the ARM specific stuff. From a quick read of the code, it seems this is compatible with the interface found in the GnuBee.

Link: https://gist.github.com/jonpry/a712d0efa2165080878f800ba82236dd

neilbrown, 2018-05-13:
This issue is fixed in the 4.4 firmware described here: https://groups.google.com/forum/#!topic/gnubee/VRNhAw1csnw
I don't think there is more to say here, so I'm closing this issue.

closed 2018-05-13 by neilbrown