gnubee-git/GnuBee_Docs #63: Unresponsive after firmware upgrade

nohkumado, 2018-01-24:
Hello
I just got the gnubee2 a few days ago, assembled it, started it and had the bad surprise of having a device which could do nothing with either the disks nor the sd card i bought it with and that i had to do a firmware upgrade (which i hate doing precicely because it allways fails).....

Well i tried a lot of different images (froma usb key) but my device now does strictly nothing.....
on the mailing list i was told that instead of the useless sd card i should have taken an adapter cable, which usage seems very complicated to me, do really again have to go through all the complicated steps between kickstarter strangeness and our local customs office here to get such a cable?To do whar with it?

Or what can i do?

nohkumado, 2018-01-24:
> You can use any USB to UART adapter. Like this one
> <https://www.amazon.com/CP2102-UART-6PIN-Serial-Converter/dp/B00CD264HG>
>
> On the bottom of the Gnubee2 are three pins labeled GND, TX, RX. Connect
> the GND pin of the serial adapter to the GND pin on the board. Connect the
> TX pin to the RX pin and the RX to the TX pin.
>
> *(TX stands for Transfer and RX stands for Receive. You connect them
> opposite of the labels so that the USB can Transfer to a pin that Receives
> and vice versa)*
>
> Then in a terminal use this command to connect to your device:
>
> On my mac:
>
> screen /dev/cu.SLAB_USBtoUART 57600
>
> On a linux machine:
>
> screen /dev/ttyUSB0 57600
>
> After that you should see a CLI that is completely removed from any network
> requirements and you can watch the whole system boot up.
>
> ok, thanks i will try this out
>
> It has been invaluable to me and I have no idea how someone would set up
> this device without
>
> well yeah, i agree, if instead of hinting at a usable out of the box thing
> they had sayd that without upgrading firmwares and using hardware monitors
> you can't get this working, and worse, hinting that by buying the supplied
> SD card with a system this would be a waltz, well i would have indeed
> refrained from buying one....
>
> yeah, this thing should be bundled with the gnubee by default, since
> without it your are lost...
>
> it. The bundled SD card simply didn't work us and we've had to go through a
> number of firmwares to find one that booted the correct thing.
>
> do you have a suggestion of one that could work? i tried all the ones given
> in the wiki.... and none worked....
>
> anyway thanks for the hints
> Bruno

Adirelle, 2018-01-24:
> It can seem very intimidating at first but is pretty simple and is a skill that translates to a lot of other devices.

I should be "highly recommended", at least, or said otherwise: "buy one if you do not already have one".

Cabalist, 2018-01-24:
@nohkumado 
We are using a USB flash drive.  A low profile 16GB one has worked for us so far.  I'll see if I can get a walkthrough up shortly.  I *think* this is enough space but we haven't put the NAS through it's paces yet.  

There is nothing stopping you from using a SATA disk as the root as well.  You could in theory have the debian root ON the RAID itself but documentation there is pretty sparse.  

Don't get me wrong the documentation on this project is extensive!  But right now almost none of it is End-User documentation.  Once you get past the firmware and into a Debian system it gets to be a lot more straightforward.  Although the OMV that is installed in the instructions is broken and not up to date.   Others have fixed this but it isn't in the documentation you have to go digging through Closed Issues.  

Other drawbacks I have learned about since I started:
- ZFS is not able to be used on this device with so little RAM.
- ClamAV cannot run on this device with so little RAM.
- Because of drive configuration and minimal mounting hardware vibration may decrease the lifespan of the drives.
- Confusion about LEDE/LibreCMC/OpenWRT and the various capabilities for each.

There is a lot of good to say about this project.  And a lot of places where someone who isn't necessarily a core dev could come in and improve it without a lot of technical knowledge. 

I'm still a believer!

noblepepper, 2018-01-27:
> Others have fixed this but it isn't in the documentation

It is [now](https://github.com/gnubee-git/GnuBee_Docs/wiki/BuildOpenMediaVault)

You can build openmediavault on the GnuBee itself!

neilbrown, 2018-05-13:
I think this issue has been addressed, so I'll close it.

closed 2018-05-13 by neilbrown

ColinNg, 2018-07-07:
For ZFS, @Cabalist did you try lowering the kernel and arc sizes? Apparently a ZFS dev says 1GB RAM is fine for even 1 exabyte of storage (and explains that the 1GB per 1TB recommendations arise from people not knowing that the ARC is just a cache, and it will evict pages as needed). 

https://linustechtips.com/main/topic/738402-zfs-memory-requirements/

I don't have the hardware myself, but if ZFS does work on it, it makes for a *very* appealing build.

Adirelle, 2018-07-07:
ZFS has been written for 64-bit CPU and kernel. GnuBEE CPU is a 32-bit one so there is no warranty it works well.