gnubee-git/GnuBee_Docs #64: Can't install debian firmware - bad magic number error

innolee, 2018-01-27:
I just received my new GB-PC2.  It seems to work fine with the default libreCMC firmware, but I want to get Debian running.

I tried to install the debian specific firmware using a USB stick and renaming the file to gnubee.bin as described in the instructions, but it doesn't work.  It gives a 'Bad Magic Number' error in the serial console after reading the file: 

```
The uscanning bus 0 for devices... 3 USB Device(s) found
       scanning bus for storage devices... 1 Storage Device(s) found
reading gnubee.bin

.....................................................................
......................................................................
......................................................................
......................................................................
......................................................

17039524 bytes read
## Checking image at 8e000000 ...
Bad Magic Number,0DF6F587
Check image error!
```

I have tried several different firmware files (from https://github.com/gnubee-git/gnubee-git.github.io/tree/master/debian) and sha256sum does not indicate any problems.

When I tried upgrading the firmware through the web interface I get the similar error 'The uploaded image file does not contain a supported format. Make sure that you choose the generic image format for your platform.'

Where am I going wrong?  Am I trying to flash the wrong firmware?

Thanks.

noblepepper, 2018-01-27:
Well, das u-boot thinks you are flashing the wrong firmware...

Which file are you trying? [This one](https://github.com/gnubee-git/gnubee-git.github.io/raw/master/debian/librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-28.bin)?

If the sha256sum comes up right you should have the correct image (with the correct magic) so double check:
1. You copied the correct file to the usb stick
2. The sha256sum is correct on the usb stick
3. You "safely removed" the usb stick after writing it

The magic number is the first 4 bytes of the file and should be:
`#define IH_MAGIC	0x27051956	/* Image Magic Number		*/`
You can check this with vbindiff if you are on a linux box

innolee, 2018-01-28:
Problem solved - the double checks you suggested didn't help, but it worked when I tried a different USB stick.

Thanks for your help.

closed 2018-01-28 by innolee