gnubee-git/GnuBee_Docs #65: No IP given after firmware update

jcerdan, 2018-02-07:
Hi, 
I've just installed this firmware openwrt-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-03-01.bin
waited about 10 minutes and Gnubee was doing nothing... So I rebooted it.
It looks like it boots and after some seconds lights in ethernet are on.
But when I plug the ethernet cable on the black or blue ports, I'm not given any IP.
How can I solve this situation?
Regards

jcerdan, 2018-02-07:
I have connected an USB 2 UART and this is what I get in output

`Flash component: SPI Flash
Date:Jul  9 2017  Time:14:14:05
============================================ 
icache: sets:256, ways:4, linesz:32, total:32768
dcache: sets:256, ways:4, linesz:32, total:32768 

 #### The CPU freq = 900 MHZ #### 
 estimate memory size = 512 Mbytes

 Reset MT7530
set LAN/WAN WLLLL
(Re)start USB...
USB0:   mtk-xhci: init hccr be1c0000 and hcor be1c0020 hc_length 32
Register 300010f NbrPorts 3
Starting the controller
USB XHCI 0.96
scanning bus 0 for devices... 2 USB Device(s) found
       scanning bus for storage devices... 0 Storage Device(s) found

 No USB Storage found. Upgrade FW failed!

Please choose the operation: 
   1: Load system code to SDRAM via TFTP.
   2: Load system code then write to Flash via TFTP.
   3: Boot system code via Flash (default).
   4: Enter boot command line interface.
   5: Load system code then write to Flash via USB Storage.
   6: Load system code then write to Flash via Httpd.
   9: Load U-Boot code then write to Flash via TFTP.

You chose 3
                                                                                                                                                                       0 

   
3: System Boot system code via Flash.
## Checking image at bc050000 ...
   Image Name:   OpenWrt Linux-3.10.14-p112871
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2251293 Bytes =  2.1 MB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK
No initrd
## Transferring control to Linux (at address 80001000) ...
## Giving linux memsize in MB, 512

Starting kernel ...

LINUX started...
ramsize = 512
rambase not set, set to default (0x00000000)
MEMORY DESCRIPTOR dump:
[0,80672250]: base<00000000> size<1c000000> type<Free RAM memory>
[0,8067225c]: base<20000000> size<04000000> type<Free RAM memory>
[    0.000000] Linux version 3.10.14 (dgazineu@olimpo) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 unknown) ) #1 SMP Wed Mar 1 10:20:40 BRT 2017
[    0.000000] 
[    0.000000]  The CPU feqenuce set to 900 MHz
[    0.000000] GCMP present
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] Software DMA cache coherency
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 1c000000 @ 00000000 (usable)
[    0.000000]  memory: 04000000 @ 20000000 (usable)
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x1fffffff]
[    0.000000]   HighMem  [mem 0x20000000-0x23ffffff]
[    0.000000] Movable zone start for each node
[... 422 lines cut ...]
[   24.212000] 
[   24.212000] phy_qrx_ring = 0x1b194000, qrx_ring = 0xbb194000
[   24.228000] 
[   24.228000] phy_rx_ring0 = 0x1b6c0000, rx_ring0 = 0xbb6c0000
[   24.256000] change HW-TRAP to 0x17ccf
[   24.264000] set LAN/WAN LLLLW
[   24.276000] GMAC1_MAC_ADRH -- : 0x000078a3
[   24.284000] GMAC1_MAC_ADRL -- : 0x51abb4a0
[   24.292000] eth1: ===> VirtualIF_open
[   24.300000] CDMA_CSG_CFG = 81000000
[   24.304000] GDMA1_FWD_CFG = 20710000
[   24.312000] GDMA2_FWD_CFG = 20710000
[   24.320000] device eth0 entered promiscuous mode
[   24.332000] device eth0.1 entered promiscuous mode
[   24.344000] br-lan: port 1(eth0.1) entered forwarding state
[   24.356000] br-lan: port 1(eth0.1) entered forwarding state
procd: - init complete -
[   26.360000] br-lan: port 1(eth0.1) entered forwarding state
`

Looks like firmware failed! 
the system does not wait for me to choose and starts with 3rd option.

What can I do ?
How to reinstall the firmware from there?

jcerdan, 2018-02-07:
ok, found how to reinstall the firmware. Now I get into the busybox GNUBee console:

`
 Please press Enter to activate this console.

BusyBox v1.22.1 (2017-03-01 10:11:29 BRT) built-in shell (ash)
Enter 'help' for a list of built-in commands.
<logo>

=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
`
But when I plug an ethernet into the black network port, I still don't get an IP.

I have used this firmware:
openwrt-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-03-01.bin

Shouldn't I, at this point, have an IP and could access the web interface?
I don't know i I should try another firmware. 
I would apreciate some help if someone has an idea of what's going wrong... (and what needs to be done, of course...)

Regards

jcerdan, 2018-02-07:
Ok, I don't get anything.

I have installed this firmware: librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-28.bin
and I get IP and can get into the libreCMC web interface. that's good, of course, but according to the doc at http://gnubee.org/install_debian.html I should install the previous firmware in order to install Debian on SD Card.

Does the kernel I have installed support Debian on HD and Debian on SD Card?

Regards,

pinney, 2018-02-07:
[ 26.360000] br-lan: port 1(eth0.1) entered forwarding state
This indicates the network is up.

Look for libreCMC on 192.168.10.1
Look for everything else on 192.168.1.1

and yes the firmware you have loaded is looking for Debian :
findfs: unable to resolve 'LABEL=GNUBE[ 15.444000] Partition GNUBEE-ROOT not found. Resuming recovery boot.

jcerdan, 2018-02-08:
Hi Pinney,

Yes thank you, I could finally access the web interface with librecmc-ramips-mt7621-gb-pc1-squashfs-sysupgrade_2017-11-28.bin.
Is this firmware able to start Debian on an SD Card?

regards

pinney, 2018-02-08:
@jcerdan 
Please see this : 
http://gnubee.org/install_debian.html

jcerdan, 2018-02-09:
Hi @pinney 
ok thanks, I finally reinstalled old firmware and managed to install OMV on an SD. ;)
Thanks again & Regards,

neilbrown, 2018-05-13:
This issue has been resolved, so I'm closing it.

closed 2018-05-13 by neilbrown