gnubee-git/GnuBee_Docs #66: iotop doesn't work with kernel 4.4.87

cmm, 2018-02-09:
(I didn't try the older kernel, so don't know if it works there)

iotop has this to say:
```
Could not run iotop as some of the requirements are not met:
- Linux >= 2.6.20 with
  - I/O accounting support (CONFIG_TASKSTATS, CONFIG_TASK_DELAY_ACCT, CONFIG_TASK_IO_ACCOUNTING)

```

neilbrown, 2018-05-13:
I've fixed this for the kernels built with https://github.com/neilbrown/gnubee-tools -  see https://groups.google.com/forum/#!topic/gnubee/VRNhAw1csnw
If you particularly need some other firmware, say which and I might be able to look at it.

cmm, 2018-05-13:
Thank you Neil, this is very much appreciated!
Is "flashcp -v gnubee.bin /dev/mtd3" expected to work with the "legacy" 4.4 kernel?  I remember you fixing flash writing for the mainline kernel, but no idea if 4.4 does or does not come with the same bugs...

neilbrown, 2018-05-13:
Yes, flashcp should work on all the older kernels.  It in just the 4.15
kernels before I discovered the problem that it doesn't work for.

NeilBrown