gnubee-git/GnuBee_Docs #67: kmod-fs-nfsd's nfsd.ko crashes on Unknown symbol __stack_chk_guard

kabo, 2018-02-17:
I've installed `kmod-fs-nfsd` and `nfs-kernel-server`. `nfsd` is enabled and tries to autostart on boot. But I get this from `dmesg`:
```
[   48.786469] nfsd: Unknown symbol __stack_chk_guard (err 0)
[   48.798483] nfsd: Unknown symbol __stack_chk_fail (err 0)
[   48.843266] nfsd: Unknown symbol __stack_chk_guard (err 0)
[   48.855385] nfsd: Unknown symbol __stack_chk_fail (err 0)
[   48.901988] nfsd: Unknown symbol __stack_chk_guard (err 0)
[   48.914177] nfsd: Unknown symbol __stack_chk_fail (err 0)
[   48.958916] kmodloader: 1 module could not be probed
[   48.969275] kmodloader: - nfsd - 0
```

```
# insmod nfsd.ko
failed to insert /lib/modules/4.4.108-gnu/nfsd.ko
```

```
# uname -a
Linux libreNAS 4.4.108-gnu #0 SMP Mon Jan 1 02:50:55 2018 mips GNU/Linux
```

Should the kernel or nfsd.ko be compiled with some other options, like `-fno-stack-protector` or `-fstack-protector-all`?

kabo, 2018-02-17:
I'm building from libreCMC v1.4.2

neilbrown, 2018-05-02:
Did you resolve this?
It looks like you were building modules in mid February, and trying to load them into a kernel that was compiled in early January.  Maybe they simply weren't compatible.  Flashing firmware with the new kernel should allow the new modules to be loaded.

neilbrown, 2018-05-13:
I don't think there is anything we can help with here, so I'll close this issue. If you still have a problem, please open another issue with details.

closed 2018-05-13 by neilbrown